espressif/esp-hosted #581: Next genertation: module error: Unknown error 517

peti634, 2025-03-16:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Hi,

I'm not sure, that I had a mistake or it is a bug.

### Actual behavior (suspected bug)

OS: Raspberry pi OS (64 bit) - 2024 -11-19 (fresh install)
HW: Rpi 3+

### Error logs or terminal output

```shell
make[1]: Entering directory '/usr/src/linux-headers-6.6.74+rpt-rpi-v8'
  CC [M]  /home/peti634/esp-hosted/esp_hosted_ng/host/spi/esp_spi.o
  CC [M]  /home/peti634/esp-hosted/esp_hosted_ng/host/esp_bt.o
  CC [M]  /home/peti634/esp-hosted/esp_hosted_ng/host/main.o
  CC [M]  /home/peti634/esp-hosted/esp_hosted_ng/host/esp_cmd.o
  CC [M]  /home/peti634/esp-hosted/esp_hosted_ng/host/esp_utils.o
  CC [M]  /home/peti634/esp-hosted/esp_hosted_ng/host/esp_cfg80211.o
  CC [M]  /home/peti634/esp-hosted/esp_hosted_ng/host/esp_stats.o
  CC [M]  /home/peti634/esp-hosted/esp_hosted_ng/host/esp_debugfs.o
  CC [M]  /home/peti634/esp-hosted/esp_hosted_ng/host/esp_log.o
  LD [M]  /home/peti634/esp-hosted/esp_hosted_ng/host/esp32_spi.o
  MODPOST /home/peti634/esp-hosted/esp_hosted_ng/host/Module.symvers
  CC [M]  /home/peti634/esp-hosted/esp_hosted_ng/host/esp32_spi.mod.o
  LD [M]  /home/peti634/esp-hosted/esp_hosted_ng/host/esp32_spi.ko
make[1]: Leaving directory '/usr/src/linux-headers-6.6.74+rpt-rpi-v8'
insmod: ERROR: could not insert module esp32_spi.ko: Unknown error 517
```

### Steps to reproduce the behavior

esp-hosted/esp_hosted_ng/host $ ->
./rpi_init.sh spi

### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Raspberry pi OS

### Shell

ZSH

mantriyogesh, 2025-03-17:
Can you please attach the textual logs from ESP and host?

1. ESP: 
- is this release flashed or master branch manually flashed?
- Something else let know the git commit uses
- `idf.py flash monitor` log from start

2. Host
- is this release flashed or master branch manually flashed?
- Something else let know the git commit uses
- `sudo dmesg` from the system bootup up at RPi

peti634, 2025-03-17:
1. I think it's not ESP related issue

2.
It's a "git clone master" code that I compiled.

There is dmesg log:
`[  271.338918] esp32_spi: loading out-of-tree module taints kernel.
[  271.546233] esp32_spi: spi_dev_init: Using SPI MODE 2
[  271.550512] esp32_spi: spi_dev_init: ESP32 peripheral is registered to SPI bus [0],chip select [0], SPI Clock [10]
[  271.550548] esp32_spi: spi_dev_init: Failed to obtain GPIO for Handshake pin, err:-517
[  272.390082] esp32_spi: spi_init: Failed Init SPI device
`

I could resolve the problem, but Im not sure how.
I will rewrite a new RPi OS to other SD Card.

The new issue:
[  731.612479] esp32_spi: Unknown symbol hci_free_dev (err -2)
[  731.612599] esp32_spi: Unknown symbol hci_alloc_dev_priv (err -2)
[  731.613365] esp32_spi: Unknown symbol bt_err (err -2)
[  731.613475] esp32_spi: Unknown symbol hci_unregister_dev (err -2)
[  731.613582] esp32_spi: Unknown symbol hci_recv_frame (err -2)
[  731.613900] esp32_spi: Unknown symbol hci_register_dev (err -2)

The "bluetooth" and other things have already installed.
Can I skip the bluetooth in the build?

mantriyogesh, 2025-03-17:
Please use rpi_init.sh as said in the documentation.

This kind of issues would only be experienced if the documentation is not followed.

For the -517 error, we would need the logs specified in the https://github.com/espressif/esp-hosted/issues/581#issuecomment-2727779762

unless logs provided, we cannot understand the issue.

peti634, 2025-03-17:
I followed this documentation:
https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/setup.md

Steps:
```sudo apt update
sudo apt install raspberrypi-kernel-headers
ls /lib/modules/$(uname -r)/build/ (OK!)
sudo apt install git raspi-gpio bluetooth bluez bluez-tools rfkill bluez-firmware pi-bluetooth
git clone https://github.com/espressif/esp-hosted.git
```

`sudo nano /boot/firmware/config.txt` (/boot/config.txt outdated?)

Add 4 lines:
```
[all]
dtparam=spi=on
dtoverlay=disable-bt
core_freq=250
core_freq_min=250

```

reboot!

```
cd esp_hosted/esp_hosted_ng/host/
bash rpi_init.sh spi
```

The full build output:
```
Building for spi protocol
rm: cannot remove 'spidev_disabler.dtbo': No such file or directory
bluetooth module inserted
rm -rf *.o sdio/*.o spi/*.o *.ko
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/6.6.51+rpt-rpi-v8/build M=/home/peti634/esp-hosted/esp_hosted_ng/host clean
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[1]: Entering directory '/usr/src/linux-headers-6.6.51+rpt-rpi-v8'
make[1]: Leaving directory '/usr/src/linux-headers-6.6.51+rpt-rpi-v8'
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/6.6.51+rpt-rpi-v8/build M=/home/peti634/esp-hosted/esp_hosted_ng/host modules
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[1]: Entering directory '/usr/src/linux-headers-6.6.51+rpt-rpi-v8'
  CC [M]  /home/peti634/esp-hosted/esp_hosted_ng/host/spi/esp_spi.o
  CC [M]  /home/peti634/esp-hosted/esp_hosted_ng/host/esp_bt.o
  CC [M]  /home/peti634/esp-hosted/esp_hosted_ng/host/main.o
  CC [M]  /home/peti634/esp-hosted/esp_hosted_ng/host/esp_cmd.o
  CC [M]  /home/peti634/esp-hosted/esp_hosted_ng/host/esp_utils.o
  CC [M]  /home/peti634/esp-hosted/esp_hosted_ng/host/esp_cfg80211.o
  CC [M]  /home/peti634/esp-hosted/esp_hosted_ng/host/esp_stats.o
  CC [M]  /home/peti634/esp-hosted/esp_hosted_ng/host/esp_debugfs.o
  CC [M]  /home/peti634/esp-hosted/esp_hosted_ng/host/esp_log.o
  LD [M]  /home/peti634/esp-hosted/esp_hosted_ng/host/esp32_spi.o
  MODPOST /home/peti634/esp-hosted/esp_hosted_ng/host/Module.symvers
  CC [M]  /home/peti634/esp-hosted/esp_hosted_ng/host/esp32_spi.mod.o
  LD [M]  /home/peti634/esp-hosted/esp_hosted_ng/host/esp32_spi.ko
make[1]: Leaving directory '/usr/src/linux-headers-6.6.51+rpt-rpi-v8'
insmod: ERROR: could not insert module esp32_spi.ko: Unknown error 517
```

dmesg:

```
[  140.811413] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/spi@7e204000/spidev@0/status
[  141.123047] Bluetooth: Core ver 2.22
[  141.123355] NET: Registered PF_BLUETOOTH protocol family
[  141.123368] Bluetooth: HCI device and connection manager initialized
[  141.123404] Bluetooth: HCI socket layer initialized
[  141.123423] Bluetooth: L2CAP socket layer initialized
[  141.123459] Bluetooth: SCO socket layer initialized
[  142.685930] hwmon hwmon1: Undervoltage detected!
[  146.717970] hwmon hwmon1: Voltage normalised
[  195.517175] esp32_spi: loading out-of-tree module taints kernel.
[  195.726065] esp32_spi: spi_dev_init: Using SPI MODE 2
[  195.726678] esp32_spi: spi_dev_init: ESP32 peripheral is registered to SPI bus [0],chip select [0], SPI Clock [10]
[  195.726698] esp32_spi: spi_dev_init: Failed to obtain GPIO for Handshake pin, err:-517
[  196.573869] esp32_spi: spi_init: Failed Init SPI device
```

Or do you need full dmesg?

mantriyogesh, 2025-03-17:
From the logs, I can see the issue clearly:

```
Failed to obtain GPIO for Handshake pin, err:-517
```

You are on the Raspberry Pi which is recent one (>bookwork version). Can you please check [this](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/Troubleshoot.md#42-reset-pin-and-spi-gpio-numbering) post?

Basically, every 

NEW_GPIO = OLD_GPIO + 512

Changes:
1. https://github.com/espressif/esp-hosted/blob/513bdfa93f5dda86d828767e5e55def718443911/esp_hosted_ng/host/rpi_init.sh#L23

from
```
RPI_RESETPIN=6
```

to 
```
RPI_RESETPIN=518
```

2. 
https://github.com/espressif/esp-hosted/blob/513bdfa93f5dda86d828767e5e55def718443911/esp_hosted_ng/host/spi/esp_spi.h#L12

to 

```
#define HANDSHAKE_PIN          534 /* (22+512) */
```

3. 
https://github.com/espressif/esp-hosted/blob/513bdfa93f5dda86d828767e5e55def718443911/esp_hosted_ng/host/spi/esp_spi.h#L14

to 

```
#define SPI_DATA_READY_PIN      539 /*(27+512)*/

peti634, 2025-03-17:
Thanks, it works.

Some additional observations:

1. The officially available downloadable release is only version 1.0.2, while the Linux HOST is currently looking for version 1.0.3.
Would it be possible to upload the newer version?

2. After compiling (rpi_init.sh spi), I get the following error:
SIOCSIFFLAGS: Operation not possible due to RF-kill

`sudo rfkill list`
output:

```
0: phy0: Wireless LAN
	Soft blocked: yes
	Hard blocked: no
2: phy2: Wireless LAN
	Soft blocked: no
	Hard blocked: no
3: hci0: Bluetooth
	Soft blocked: yes
	Hard blocked: no
```

`ip link show`
output:

```
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP mode DEFAULT group default qlen 1000
    link/ether b8:27:eb:93:75:46 brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/ether b8:27:eb:c6:20:13 brd ff:ff:ff:ff:ff:ff
4: wlan1: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/ether 3c:71:bf:88:a8:18 brd ff:ff:ff:ff:ff:ff
```

However, I am able to enable wlan1:
`sudo ifconfig wlan1 up`
After this, it's okay!

I read the documentation more times but I didn't find any thing to refer to this.

mantriyogesh, 2025-03-18:
Thanks for the explanation.

We will add your observations in troubleshoot document.

Regarding version, we will spin the version but for now, it is better if you use master on both sides. Ensure you use exact same commit at both sides. Else in some corner cases it may just fail..

closed 2025-03-18 by mantriyogesh