espressif/esp-hosted #582: hosted-fg(bt/ble control)

alexsvn, 2025-03-17:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

When using hosted-fg project on MCU, MCU cannot integrate bluez, and BT protocol stack is on ESP32 side, how to use BT/BLE; is there a control path (slave_control to control Bluetooth similar to WiFi?

### Use cases

Na

mantriyogesh, 2025-03-17:
The control interface is only required for general ['control' path APIs](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/common/ctrl_apis.md). There is just `enable_bt(void)` or `disable_bt(void)` supported to high level control of Bluetooth to be enabled or disabled. 
https://github.com/espressif/esp-hosted/blob/513bdfa93f5dda86d828767e5e55def718443911/esp_hosted_fg/host/linux/host_control/c_support/test_utils.c#L1206-L1236

Other than that, the Bluetooth doesn't need such RPC, as the **standard HCI protocol**/interface exists in between Bluetooth stack and the Bluetooth controller.

We rely on the standard HCI interface.
There are two types of HCI implementation alternative to each other.
1. Standard HCI 
Need dedicated transport like Bluetooth. Cannot multiplex with other traffic types, as these messages are standard.
- Cannot be shared with main transport, for example RPC and wifi running over SPI (or SDIO). Standard HCI would run on additional explicit transport like UART.
- Transparent
- Extra GPIOs , extra processing when both transports are active
- Could be used also as fault-tolerant , for example, spi is idled or stopped etc, still uart can run with lessor power etc

2. vHCI or Hosted HCI
The standard HCI messages are just wrapped with Hosted header to re-use same transport , where other traffic types are being used.

For Please check the detailed design for a lot of clarity:
https://github.com/espressif/esp-hosted-mcu/blob/main/docs/bluetooth_design.md

Although repo is different than main hosted for linux, the concept and design is the same.

mantriyogesh, 2025-03-17:
I am little confused by your question.

Can you please clarify if you want to use bluetooth stack on MCU or Linux?

If Linux, current repo ESP-Hosted-FG is fine.

But if MCU then refrain from using current repo. reasone being, we have dedicated repo, https://github.com/espressif/esp-hosted-mcu for catering MCU case.
In this repo, examples are catered to handle MC cases, with port definition provided as ESP chipset (ESP as slave and ESP as host).
https://github.com/espressif/esp-hosted-mcu is also implemented by non ESP as MCU as example , stm32 (non public)

alexsvn, 2025-03-17:
Use hosted-fg wifi network card mode on Linux platform; BT stack is not implemented in Linux, only control, enable esp32 BT stack bluedroid
Looking at the code of host_control/c_support/, I can use the message mechanism here to add Linux control ESP32 BT/BLE

mantriyogesh, 2025-03-17:
# Problem understanding 
If I understand correctly, you want to maintain the bluedroid stack at ESP only and control the stack using serial command.

Basically stack+controller on ESP itself. Just high level control from Linux.

## Short answer
So, If you you develop a simple bluedroid (bt/ble) example, which you just wish to trigger kind of 'start' and 'stop' control from Linux.

Hmm.. from user perspective, Yes! This is definitely do-able, with primitive controls directly from Linux.

## Extra details on implementation

### Ways to implement
You have multiple options in this:
A. Just use existing APIs - Linux user space RPCand ESP slave_control.c RPC implementation can be ammended to give extra meaning to existing APIs.
B. Add new RPC - You can actually add your own new rpc functions too, by referring any simple sample API implementation from Linux to ESP and from ESP to host.

### Some details of RPC

You can use RPC request-response when host want to get/set something at slave. 
The rpc request may be synchronous or asynchronous.

#### RPC Request - Response
- Synchronous request
is something that host current thread would block till the acknowledgement/nack response.

- Asynchronous request
is something that host current thread will just submit the request with callback registered. Host would not wait for response. When response returned, callback would be invoked.

Note: Ensure that you do not send synchronous request in the response handler itself (as synchronous would block the response handler i.e. rx  thread). If you wish to send request2 in request1 response handler, use asynchronous way.

However such limitation does not exist if you handle responses in your own thread.

Although rpc requests are thread safe, the slave handle all requests sequentially. So to simplify it is recommended to do all rpc requests from single user thread at Linux.

#### RPC Events 
Also you can use RPC events , when slave wants to sent something interesting happened to host asynchronously. async only.

The callbacks are called from rx thread again. So something small, non blocking operation can be done in the callback functions. Do not block the callback function by sending sync request or any blocking operation.

## Final remarks

You can adjust the code as per your use case. Idf has a lot of examples to use bluedroid or nimble with stack and controller on same ESP chip. You can form your simple example first and provide small configurable items from Linux as your bt/ble example needs.

Although, for something complex, we recommend using HCI as would be tremendously easier. If you have space crunch, you can also have a look at nimble stack on Linux (BLE only).

If I think from public repo perspective, every stack has very different APIs. So standard HCI or hosted HCI  for public repo makes more sense.

alexsvn, 2025-04-09:
Closes #582

closed 2025-04-09 by alexsvn