espressif/esp-hosted #583: Segfault Esp32c3 Firmware

BrelJordan, 2025-03-21:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

No Segmentation fault

### Actual behavior (suspected bug)

I have esphosted ng on the host and the corresponding firmware v1.0.3.0.10. I connect to an AP and performed an iperf test. I lost the interface shortly after and the kernel module on the host detected a boot up event. I could also successfully reconnect. But repeating the test still fails.

### Error logs or terminal output

```shell
I (50549) FW_MAIN: Get Tx power command
I (50618) FW_MAIN: Get Tx power command
I (50669) FW_MAIN: Get Tx power command
I (50889) FW_MAIN: Get Tx power command
I (51079) FW_MAIN: Get Tx power command
I (51229) FW_MAIN: Get Tx power command
I (51279) FW_MAIN: Get Tx power command
I (51309) FW_MAIN: Get Tx power command
I (51379) FW_MAIN: Get Tx power command
I (51428) FW_MAIN: Get Tx power command
I (51519) FW_MAIN: Get Tx power command
I (51599) FW_MAIN: Get Tx power command
I (51679) FW_MAIN: Scan request
I (51729) FW_MAIN: Set Tx power command
I (51778) FW_MAIN: Set Tx power command
I (54139) FW_MAIN: Auth request
I (54340) FW_CMD: Connecting to MC-Tech-Net, channel: 6 [3]
I (54340) FW_CMD: AUTH Commit

I (54345) wifi:new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1, snd_ch_cfg:0x0
I (54347) wifi:state: init -> auth (0xb0)
I (54349) FW_CMD: Unregistered event: 43

I (54352) FW_CMD: handle_wpa_sta_rx_mgmt:731 Auth[11] recvd

I (54408) FW_MAIN: Assoc request
I (54408) wifi:state: auth -> assoc (0x0)
I (54428) FW_CMD: handle_wpa_sta_rx_mgmt:739 ASSOC Resp[1] recvd

I (54428) FW_CMD: STA connect event [channel 6]

I (54429) BSSID: 0x3fca963e   34 56 fe c7 34 23                                 |4V..4#|
I (54436) wifi:state: assoc -> run (0x10)
I (54468) FW_MAIN: Get Tx power command
I (54529) FW_MAIN: Get Tx power command
I (54568) FW_MAIN: Add key request
I (54568) FW_CMD: set_key_internal:1685

I (54568) FW_CMD: Setting PTK algo=3 index=0
I (54628) FW_MAIN: Add key request
I (54628) FW_CMD: set_key_internal:1685

I (54628) FW_CMD: Setting GTK [1]

I (54628) wifi:connected with MC-Tech-Net, aid = 1, channel 6, BW20, bssid = 34:56:fe:c7:34:23
I (54635) wifi:security: WPA2-PSK, phy: bgn, rssi: -65
I (54640) wifi:pm start, type: 1

I (54643) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
[... 180 lines cut ...]
3. run an iperf test test

### Project release version

v1.0.3.0.10

### System architecture

ARM 64-bit (Apple M1/M2, Raspberry Pi 4/5)

### Operating system

Linux

### Operating system version

openwrt 24

### Shell

sh

### Additional context

The esp is connected via spi to the host.

BrelJordan, 2025-03-21:
I used this command : 
iperf3 -c [iperf server] --bidir -t 20

It seems the paket block sizes are to big for the esp, by reducing the default BS from 8000 bytes to 100 the error does not occur:

iperf3 -c saeber.de --bidir -t 20 --set-mss 100

mantriyogesh, 2025-03-24:
@kapilkedawat PTAL