espressif/esp-hosted #584: There were some problems trying to compile esp32c3 firmware in esp_hosted_ng

YousYu, 2025-03-30:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

Hello, I tried to compile (**esp-hosted\esp_hosted_ng\esp\esp_driver\network_adapter**) path esp32c3 firmware, encountered some problems in the process, did not find the relevant solution.Only the target and spi modes are set.

**The compilation error message is as follows:**

[1/1] C:\Windows\system32\cmd.exe /C "cd /D D:\Embedded\esp-hosted\esp_hosted_ng\esp\esp_driver\network_adapter\build\bootloader\esp-idf\esptool_py && D:\Espressif\IDF5.3.1\python_env\idf5.4_py3.13_env\Scripts\python.exe D:/Embedded/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 D:/Embedded/esp-hosted/esp_hosted_ng/esp/esp_driver/network_adapter/build/bootloader/bootloader.bin"
Bootloader binary size 0x5150 bytes. 0x2eb0 bytes (36%) free.
[4/6] Linking CXX executable network_adapter.elf
FAILED: network_adapter.elf 
C:\Windows\system32\cmd.exe /C "cd . && D:\Espressif\IDF5.3.1\tools\riscv32-esp-elf\esp-14.2.0_20241119\riscv32-esp-elf\bin\riscv32-esp-elf-g++.exe -march=rv32imc_zicsr_zifencei -nostartfiles -march=rv32imc_zicsr_zifencei --specs=nosys.specs    -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32C3=0 -Wl,--Map=D:/Embedded/esp-hosted/esp_hosted_ng/esp/esp_driver/network_adapter/build/network_adapter.map -Wl,--no-warn-rwx-segments -Wl,--orphan-handling=warn -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T rom.api.ld -T esp32c3.peripherals.ld -T esp32c3.rom.ld -T esp32c3.rom.api.ld -T esp32c3.rom.bt_funcs.ld -T esp32c3.rom.libgcc.ld -T esp32c3.rom.version.ld -T esp32c3.rom.eco3.ld -T esp32c3.rom.eco3_bt_funcs.ld -T esp32c3.rom.newlib.ld -T memory.ld -T sections.ld @CMakeFiles\network_adapter.elf.rsp -o network_adapter.elf && cd ."
D:/Espressif/IDF5.3.1/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/main/libmain.a(cmd.c.obj): in function `mgmt_txcb':
D:/Embedded/esp-hosted/esp_hosted_ng/esp/esp_driver/network_adapter/main/cmd.c:1894:(.text.mgmt_txcb+0x18): undefined reference to `esp_wifi_get_eb_data'
D:/Espressif/IDF5.3.1/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld.exe: D:/Embedded/esp-hosted/esp_hosted_ng/esp/esp_driver/network_adapter/main/cmd.c:1900:(.text.mgmt_txcb+0x24): undefined reference to `esp_wifi_get_eb_data_len'
D:/Espressif/IDF5.3.1/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/main/libmain.a(cmd.c.obj): in function `add_station_node_ap':
D:/Embedded/esp-hosted/esp_hosted_ng/esp/esp_driver/network_adapter/main/cmd.c:2098:(.text.add_station_node_ap+0x17e): undefined reference to `ieee80211_add_node'
D:/Espressif/IDF5.3.1/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/main/libmain.a(cmd.c.obj): in function `process_disconnect':
D:/Embedded/esp-hosted/esp_hosted_ng/esp/esp_driver/network_adapter/main/cmd.c:1282:(.text.process_disconnect+0xa0): undefined reference to `ieee80211_delete_node'
D:/Espressif/IDF5.3.1/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/main/libmain.a(cmd.c.obj): in function `process_auth_request':
D:/Embedded/esp-hosted/esp_hosted_ng/esp/esp_driver/network_adapter/main/cmd.c:1322:(.text.process_auth_request+0x18e): undefined reference to `esp_wifi_send_auth_internal'
D:/Espressif/IDF5.3.1/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/main/libmain.a(cmd.c.obj): in function `process_assoc_request':
D:/Embedded/esp-hosted/esp_hosted_ng/esp/esp_driver/network_adapter/main/cmd.c:1399:(.text.process_assoc_request+0x1a): undefined reference to `esp_wifi_send_assoc_internal'
D:/Espressif/IDF5.3.1/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/main/libmain.a(cmd.c.obj): in function `process_mgmt_tx':
D:/Embedded/esp-hosted/esp_hosted_ng/esp/esp_driver/network_adapter/main/cmd.c:2056:(.text.process_mgmt_tx+0x88): undefined reference to `ieee80211_send_mgmt_internal'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

mantriyogesh, 2025-03-30:
Have u u followed the building steps here: https://github.com/espressif/esp-hosted/tree/master/esp_hosted_ng/esp/esp_driver

YousYu, 2025-03-30:
Thank you for your reply. I will try to follow the steps again.

closed 2025-04-03 by YousYu