espressif/esp-hosted #586: softAP mode issue

helpfile, 2025-04-02:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

The host can work in AP mode

### Actual behavior (suspected bug)

I (59748) wifi:mode : softAP (54:32:04:0b:e0:90)
I (59752) wifi:Total power save buffer number: 20
I (59756) wifi:Init max length of beacon: 752/752
E (59761) wifi:ieee80211_output.c 5202

### Error logs or terminal output

```shell
linux dmesg：
[   63.560665] esp32c6: wait_and_decode_cmd_resp: Command[0x16] timed out
[   63.570047] esp32c6: cmd_set_mode: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22
[   63.579136] esp32c6: esp_cfg80211_change_iface: wdev iftype=2, ret=-22
[   63.586524] esp32c6: update_mac_filter: 1: 01:00:5e:00:00:01
[   63.586549] esp32c6: update_mac_filter: Setting Multicast list
[   63.586860] esp32c6: esp_cfg80211_change_iface: current iface type=0 new iface type=1
[   63.597832] cmd_set_mode(1966)
[   63.601167] esp32c6: esp_cmd_work: Processing Command [0xE]
[   63.640391] esp32c6: esp_cmd_work: Busy in another cmd execution
[   68.580365] esp32c6: wait_and_decode_cmd_resp: Command[0xE] timed out
[   68.587150] esp32c6: cmd_set_mcast_mac_list: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22
[   68.600366] esp32c6: wait_and_decode_cmd_resp: Command[0x16] timed out
[   68.607159] esp32c6: reset_cmd_node: recycling command still in cmd queue
[   68.607176] esp32c6: cmd_set_mode: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22
[   68.616266] esp32c6: esp_cfg80211_change_iface: wdev iftype=2, ret=-22
[   68.623325] esp32c6: update_mac_filter: Setting Multicast list
[   68.660401] esp32c6: esp_cmd_work: Processing Command [0xE]
[   73.620380] esp32c6: wait_and_decode_cmd_resp: Command[0xE] timed out
[   73.627203] esp32c6: cmd_set_mcast_mac_list: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22
[   73.639808] esp32c6: update_mac_filter: 1: 01:00:5e:00:00:01
[   73.639831] esp32c6: update_mac_filter: Setting Multicast list
[   73.670388] esp32c6: esp_cmd_work: Processing Command [0xE]
[   78.630488] esp32c6: wait_and_decode_cmd_resp: Command[0xE] timed out
[   78.640013] esp32c6: cmd_set_mcast_mac_list: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22

======================================================
ESP32C6 log：

ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0xf (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd44
load:0x4086e610,len:0x2de4
load:0x40875728,len:0x17dc
entry 0x4086c410
I (23) boot: ESP-IDF HEAD-HASH-NOTFOUND 2nd stage bootloader
I (24) boot: compile time Jan 16 2025 16:18:45
I (24) boot: chip revision: v0.0
I (27) boot.esp32c6: SPI Speed      : 80MHz
I (32) boot.esp32c6: SPI Mode       : DIO
I (37) boot.esp32c6: SPI Flash Size : 4MB
I (42) boot: Enabling RNG early entropy source...
I (47) boot: Partition Table:
[... 248 lines cut ...]
[   73.639808] esp32c6: update_mac_filter: 1: 01:00:5e:00:00:01
[   73.639831] esp32c6: update_mac_filter: Setting Multicast list
[   73.670388] esp32c6: esp_cmd_work: Processing Command [0xE]
[   78.630488] esp32c6: wait_and_decode_cmd_resp: Command[0xE] timed out
[   78.640013] esp32c6: cmd_set_mcast_mac_list: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22

### Project release version

NG-1.0.3

### System architecture

other (details in Additional context)

### Operating system

Linux

### Operating system version

linux 3.10

### Shell

ZSH

mantriyogesh, 2025-04-02:
The Wifi watchdog is concerning.

The watchdog is similar to https://github.com/espressif/esp-hosted/issues/573. But the resolution there is not going to help in this issue.

@kapilkedawat Can you please take a look?

kapilkedawat, 2025-04-02:
@Shreyas0-7 please check.

mantriyogesh, 2025-04-02:
@helpfile , ensure that same esp-hosted git commit is used at host and slave at any point.
you can simply go to latest esp-hosted commit both sides and flash both sides.