espressif/esp-hosted #590: hosted-fg:After turning on PSRAM, SDIO cannot send data, and it prompts that it is not DMA memory

alexsvn, 2025-04-07:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

I (2493) NETWORK_ADAPTER: Start Data Path
E (3119) sdio_slave: sdio_slave_send_queue(596): buffer to send should be DMA capable and 32-bit aligned
E (3119) SDIO_SLAVE: sdio slave tx error, ret : 0x102

mantriyogesh, 2025-04-07:
Hello @alexsvn ,

This looks like a bug, but we need more confirmation, if you are okay with it:

1. Can you please attach the full text from bootup for esp?
2. Please attach sdkconfig used, which is easier for us to test with.
3. Is the host Linux or MCU?
4. Code base used at both places:
- Are you using same code (exact same git commit) at slave and host?
- Are you using latest master?
- If not, please try first on latest master.
- Please help stating git commit both sides.

alexsvn, 2025-04-07:
This library is also used by Linux
esp-hosted commitid
commit 513bdfa93f5dda86d828767e5e55def718443911 (HEAD -> master, origin/master, origin/HEAD)
Merge: 823762c 9936f17
Author: Yogesh Mantri <yogesh.mantri@espressif.com>
Date:   Wed Mar 12 11:43:32 2025 +0800

alexsvn, 2025-04-09:
Dear @mantriyogesh,
The issue is consistently reproducible by enabling PSRAM without any code modifications.
Steps to reproduce:
1. Clean build environment: `idf.py fullclean`
2. Enable PSRAM in menuconfig:
(Top) → Component config → ESP PSRAM
[*] Support for external, SPI-connected RAM
        SPI RAM config  --->
3. Flash the boar
Observed behavior:
I (2493) NETWORK_ADAPTER: Start Data Path
E (3119) sdio_slave: sdio_slave_send_queue(596): buffer to send should be DMA capable and 32-bit aligned
E (3119) SDIO_SLAVE: sdio slave tx error, ret : 0x102

SohKamYung-Espressif, 2025-04-09:
@alexsvn Just a check: are you using the ESP32 as the slave?

alexsvn, 2025-04-09:
@SohKamYung-Espressif
are you using the ESP32 as the slave? Yes

SohKamYung-Espressif, 2025-04-09:
@alexsvn Can you send your `sdkconfig`, with PSRAM enabled?

alexsvn, 2025-04-09:
@SohKamYung-Espressif 
<details>
  <summary>sdkconfig, with PSRAM enabled</summary>

#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) 5.3.2 Project Configuration
#
CONFIG_SOC_BROWNOUT_RESET_SUPPORTED="Not determined"
CONFIG_SOC_TWAI_BRP_DIV_SUPPORTED="Not determined"
CONFIG_SOC_DPORT_WORKAROUND="Not determined"
CONFIG_SOC_CAPS_ECO_VER_MAX=301
CONFIG_SOC_ADC_SUPPORTED=y
CONFIG_SOC_DAC_SUPPORTED=y
CONFIG_SOC_UART_SUPPORTED=y
CONFIG_SOC_MCPWM_SUPPORTED=y
CONFIG_SOC_GPTIMER_SUPPORTED=y
CONFIG_SOC_SDMMC_HOST_SUPPORTED=y
CONFIG_SOC_BT_SUPPORTED=y
CONFIG_SOC_PCNT_SUPPORTED=y
CONFIG_SOC_PHY_SUPPORTED=y
CONFIG_SOC_WIFI_SUPPORTED=y
CONFIG_SOC_SDIO_SLAVE_SUPPORTED=y
CONFIG_SOC_TWAI_SUPPORTED=y
CONFIG_SOC_EFUSE_SUPPORTED=y
CONFIG_SOC_EMAC_SUPPORTED=y
CONFIG_SOC_ULP_SUPPORTED=y
CONFIG_SOC_CCOMP_TIMER_SUPPORTED=y
CONFIG_SOC_RTC_FAST_MEM_SUPPORTED=y
CONFIG_SOC_RTC_SLOW_MEM_SUPPORTED=y
CONFIG_SOC_RTC_MEM_SUPPORTED=y
CONFIG_SOC_I2S_SUPPORTED=y
CONFIG_SOC_RMT_SUPPORTED=y
CONFIG_SOC_SDM_SUPPORTED=y
CONFIG_SOC_GPSPI_SUPPORTED=y
CONFIG_SOC_LEDC_SUPPORTED=y
CONFIG_SOC_I2C_SUPPORTED=y
CONFIG_SOC_SUPPORT_COEXISTENCE=y
CONFIG_SOC_AES_SUPPORTED=y
CONFIG_SOC_MPI_SUPPORTED=y
CONFIG_SOC_SHA_SUPPORTED=y
CONFIG_SOC_FLASH_ENC_SUPPORTED=y
CONFIG_SOC_SECURE_BOOT_SUPPORTED=y
CONFIG_SOC_TOUCH_SENSOR_SUPPORTED=y
CONFIG_SOC_BOD_SUPPORTED=y
CONFIG_SOC_ULP_FSM_SUPPORTED=y
CONFIG_SOC_CLK_TREE_SUPPORTED=y
CONFIG_SOC_MPU_SUPPORTED=y
CONFIG_SOC_WDT_SUPPORTED=y
CONFIG_SOC_SPI_FLASH_SUPPORTED=y
CONFIG_SOC_RNG_SUPPORTED=y
CONFIG_SOC_LIGHT_SLEEP_SUPPORTED=y
CONFIG_SOC_DEEP_SLEEP_SUPPORTED=y
CONFIG_SOC_LP_PERIPH_SHARE_INTERRUPT=y
CONFIG_SOC_PM_SUPPORTED=y
CONFIG_SOC_DPORT_WORKAROUND_DIS_INTERRUPT_LVL=5
CONFIG_SOC_XTAL_SUPPORT_26M=y
CONFIG_SOC_XTAL_SUPPORT_40M=y
CONFIG_SOC_XTAL_SUPPORT_AUTO_DETECT=y
CONFIG_SOC_ADC_RTC_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_DIG_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_DMA_SUPPORTED=y
CONFIG_SOC_ADC_PERIPH_NUM=2
CONFIG_SOC_ADC_MAX_CHANNEL_NUM=10
CONFIG_SOC_ADC_ATTEN_NUM=4
CONFIG_SOC_ADC_DIGI_CONTROLLER_NUM=2
CONFIG_SOC_ADC_PATT_LEN_MAX=16
CONFIG_SOC_ADC_DIGI_MIN_BITWIDTH=9
CONFIG_SOC_ADC_DIGI_MAX_BITWIDTH=12
CONFIG_SOC_ADC_DIGI_RESULT_BYTES=2
[... 2167 lines cut ...]
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_TIME_SYSCALL_USE_RTC_HRT=y
CONFIG_ESP32_TIME_SYSCALL_USE_RTC_FRC1=y
# CONFIG_ESP32_TIME_SYSCALL_USE_RTC is not set
# CONFIG_ESP32_TIME_SYSCALL_USE_HRT is not set
# CONFIG_ESP32_TIME_SYSCALL_USE_FRC1 is not set
# CONFIG_ESP32_TIME_SYSCALL_USE_NONE is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
# CONFIG_ESP32_ULP_COPROC_ENABLED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
# End of deprecated options

</details>

closed 2025-04-09 by espressif-bot

SohKamYung-Espressif, 2025-04-09:
@alexsvn https://github.com/espressif/esp-hosted/commit/337c8fa1703cb39cc14aa6722237a508c7ef91b6 should fix the issue.

alexsvn, 2025-04-09:
@SohKamYung-Espressif  Thank you for the prompt resolution! The PSRAM initialization issue is now confirmed fixed after applying your patch. Closing as resolved. Closes #588