espressif/esp-hosted #592: How can ESP32c6 be awakened by slave sdio after entering light sleep?

helpfile, 2025-04-11:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

SDIO can wake up light sleep

### Actual behavior (suspected bug)

We are now using the hosted mode of esp32c6, and using slave sdio to communicate with other master controllers.
Please tell me how to wake up the esp32c6 after it enters light-sleep by slave sdio?
﻿
I read in the specification that sdio can wake up light-sleep.
However, there is only one definition in the entire hosted code, and there is no other related code.
The hosted code is the latest version cloned from the official website last week, and the idf version is 5.4

### Error logs or terminal output

```shell
lucid@ub24:/dev/shm/esp-hosted$ grep -r "PMU_SDIO_WAKEUP_EN"  ./
./esp_hosted_fg/esp/esp_driver/esp-idf/components/esp_hw_support/port/esp32c6/private_include/pmu_bit_defs.h:#define PMU_SDIO_WAKEUP_EN          BIT(8)
./esp_hosted_fg/esp/esp_driver/esp-idf/components/esp_hw_support/port/esp32p4/private_include/pmu_bit_defs.h:#define PMU_SDIO_WAKEUP_EN              BIT(0)
./esp_hosted_ng/esp/esp_driver/esp-idf/components/esp_hw_support/port/esp32c6/private_include/pmu_bit_defs.h:#define PMU_SDIO_WAKEUP_EN          BIT(8)
./esp_hosted_ng/esp/esp_driver/esp-idf/components/esp_hw_support/port/esp32p4/private_include/pmu_bit_defs.h:#define PMU_SDIO_WAKEUP_EN              BIT(0)
lucid@ub24:/dev/shm/esp-hosted$
```

### Steps to reproduce the behavior

Start up

### Project release version

NG-1.0.3.0.12

### System architecture

other (details in Additional context)

### Operating system

Linux

### Operating system version

esp32c6 sdk

### Shell

ZSH

mantriyogesh, 2025-04-11:
i did not understand correctly. Can you please state the use case?

the light sleep is supported through esp-idf. if you wish you can enable on your own.

hosted driver yet not integrated with light sleep. but it should work in the same way esp-idf light sleep examples would work.

if you wish to put host into deep sleep or linux power save, it is supported out of the box, [here](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/host_sleep.md). However check your linux supports deep sleep or not first. some linux support , whereas some don't.

helpfile, 2025-04-12:
We use esp32c6 as a wifi/ble modules to another linux host using sdio interface,  the esp32c6 runs esp_hosted_ng firmware.

The problem is that when esp32c6 goes to ligth-sleep, it can't wakeup by sdio communication issued by the linux host.

From the source code of hosted project, we can find some API to wakeup esp32c6 from light-sleep status such as :  
esp_sleep_enable_gpio_wakeup()
esp_sleep_enable_timer_wakeup()
esp_sleep_enable_uart_wakeup()

but we can't find similar API for sdio  to wakeup light-sleep.

Please provide some demo that sdio can wakeup the esp32c6 from light-sleep status .

mantriyogesh, 2025-04-12:
This is DIY and for now out of scope of ESP-Hosted, anyway just to help, can we please answer @SohKamYung-Espressif , what we can suggest?

SohKamYung-Espressif, 2025-04-14:
@helpfile You can look at https://github.com/espressif/esp-idf/tree/master/examples/system/light_sleep for an example on how to use GPIO for light sleep.

I have tested this on the ESP32-C6 as slave by adding code to go to light sleep after a while, then pressing the BOOT button to wake up the ESP32-C6 and do SDIO communications (perform a scan). 

**Caution:** enabling light sleep will power down Wi-Fi and Bluetooth https://docs.espressif.com/projects/esp-idf/en/latest/esp32c6/api-reference/system/sleep_modes.html#wi-fi-bluetooth-and-sleep-modes 

If you need to keep Bluetooth or Wi-Fi alive during light sleep, consider auto light sleep mode instead https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/power_management.html#configuration

SohKamYung-Espressif, 2025-04-15:
@helpfile I can confirm that at the moment, slave SDIO is not available as a wake-up source on the ESP32-C6.

You can try using GPIO wakeup https://github.com/espressif/esp-hosted/issues/592#issuecomment-2800385367 or use auto light sleep instead.

Do you have further questions on this topic, or can it be considered closed for now?