espressif/esp-hosted #593: The solution of using ESP32-C6 with SDIO as HOST_NG for I.MX93

xiatian-xt, 2025-04-13:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

**device tree**
&usdhc3 {
	pinctrl-names = "default", "state_100mhz", "state_200mhz";
	pinctrl-0 = <&pinctrl_usdhc3>;
	pinctrl-1 = <&pinctrl_usdhc3>;
	pinctrl-2 = <&pinctrl_usdhc3>;
	mmc-pwrseq = <&usdhc3_pwrseq>;
	bus-width = <4>;
	broken-cd;
	keep-power-in-suspend;
	status = "okay";
};

**dmesg log**
[    9.629125] systemd-journald[172]: Received SIGUSR2 signal from PID 1, as request to rotate journal, rotating.
[    9.645727] systemd-journald[172]: Vacuuming done, freed 0B of archived journals from /run/log/journal/8f1435.
[   18.661352] platform imx93-lpm: deferred probe pending
[   46.640645] esp32_hosted: esp_reset: Triggering ESP reset.
[   57.495476] esp32_hosted: esp_reset: Triggering ESP reset.
[   69.640427] esp32_hosted: esp_reset: Triggering ESP reset.
[   92.243182] esp32_hosted: esp_reset: Triggering ESP reset.

The device tree information and dmesg information are as above. It seems that the SDIO is not functioning. I captured the signals with a logic analyzer and found that there is no clock signal. Therefore, I suspect that it is still an issue with my configuration. I'd like to inquire if there are any similar cases. Thank you!

### Use cases

We are currently in the planning phase of designing an electroencephalogram (EEG) acquisition device that uses Wi-Fi to transmit EEG signals.

xiatian-xt, 2025-04-13:
root@atk-imx93:/usr/output/scripts# dmesg |  grep mmc
[    0.000000] Kernel command line: root=/dev/mmcblk0p2 rootwait rw console=ttyLP0,115200
[    1.825077] mmc2: SDHCI controller on 428b0000.mmc [428b0000.mmc] using ADMA
[    1.829471] mmc0: SDHCI controller on 42850000.mmc [42850000.mmc] using ADMA
[    1.947540] mmc0: new HS400 Enhanced strobe MMC card at address 0001
[    1.957963] mmcblk0: mmc0:0001 88A43A 14.6 GiB 
[    1.968411]  mmcblk0: p1 p2
[    1.983346] mmcblk0boot0: mmc0:0001 88A43A 4.00 MiB 
[    1.993683] mmcblk0boot1: mmc0:0001 88A43A 4.00 MiB 
[    2.003594] mmcblk0rpmb: mmc0:0001 88A43A 4.00 MiB, chardev (234:0)
[    2.338501] sdhci-esdhc-imx 42860000.mmc: Got CD GPIO
[    2.386552] mmc1: SDHCI controller on 42860000.mmc [42860000.mmc] using ADMA
[    2.399806] EXT4-fs (mmcblk0p2): mounting ext3 file system using the ext4 subsystem
[    2.423636] EXT4-fs (mmcblk0p2): recovery complete
[    2.428596] EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Quota mode: none.
[    3.954622] EXT4-fs (mmcblk0p2): re-mounted. Quota mode: none.

SohKamYung-Espressif, 2025-04-14:
@kapilkedawat Can you take a look to see if the device tree looks correct?

kapilkedawat, 2025-04-14:
Hi @xiatian-xt could you paste the output of `ls /sys/class/mmc_host/mmc2/` ? 

Another user was using IMX93, may be we can get the working DT from [there](https://github.com/espressif/esp-hosted/issues/549)

mantriyogesh, 2025-04-24:
In addition, @xiatian-xt,  can you please capture a photo of your setup? Also please check if you have followed the [Porting Guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md)? It has briefed about the expected external pull-up registered and expected lines.

Once you have PCB setup with the external pull-ups with expected lines and values, it should work. With wired setup, SDIO cannot function properly because of signal integrity issues. JFYI, we use 51KOhm in our setup.