espressif/esp-hosted #594: What issue will let the esp32c3 hand shake pin always reset?

LuciaferKelvin, 2025-04-15:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

Firmware version: Release v0.4
Control base: MCU-based
Issue: The handshake pin remains in a reset state（low level）, so the MCU controller cannot send any messages to the ESP32, including control and network messages. After detecting the communication failure, the MCU resets the power of the ESP, and then the system returns to normal.

SohKamYung-Espressif, 2025-04-15:
Hi. To make the issue clearer can you provide the following information:

- are you running ESP-Hosted FG or NG?
- when you say "Release v0.4", this is the ESP-Hosted version? If yes, this is an older version. Is it possible for you to try with the latest version?

Can you provide the debug log output from the ESP32 and your host when the problem occurs?

LuciaferKelvin, 2025-04-15:
> Hi. To make the issue clearer can you provide the following information:
> 
> * are you running ESP-Hosted FG or NG?
> * when you say "Release v0.4", this is the ESP-Hosted version? If yes, this is an older version. Is it possible for you to try with the latest version?
> Yes.The product is already widely deployed at client sites, and the firmware version cannot be updated.This product has been maintained for over two years. Previously, we consulted on other issues and received suggestions to update the firmware version. However, the current firmware does not support a universal command to retrieve the version information. Additionally, there were communication protocol changes after version 0.4, which caused compatibility issues—devices running the updated firmware cannot communicate using the old commands, and clients with products already in use cannot switch to the new commands. Furthermore, using old commands with the new firmware causes the ESP to crash, and using new commands with the old firmware also leads to crashes. These issues have been verified through testing, so updating the firmware version is currently not a feasible option.
> Can you provide the debug log output from the ESP32 and your host when the problem occurs?
> The issue occurred on the customer's product, so it is not possible to obtain logs from the ESP32. We can only analyze the ESP32's status through the logs from the main controller.

SohKamYung-Espressif, 2025-04-15:
Is your product using ESP-Hosted FG or NG?

LuciaferKelvin, 2025-04-15:
The communication method used for control is SPI.

LuciaferKelvin, 2025-04-15:
> Is your product using ESP-Hosted FG or NG?

Using the FG case

SohKamYung-Espressif, 2025-04-15:
Do you have commit ID of the ESP-Hosted source code you are using?

LuciaferKelvin, 2025-04-15:
[image]That tag

SohKamYung-Espressif, 2025-04-15:
> The on-site situation is as follows:
The Wi-Fi signal is very weak, around -80 dBm, but the Wi-Fi connection remains stable. Under normal conditions, the ESP32 does not get disconnected from the hotspot.
There is a high volume of network traffic, with broadcast messages occurring approximately every 20 ms.

Just to understand the situation more:
- is the ESP32 receiving or generating the high volume of network traffic?

LuciaferKelvin, 2025-04-15:
Both of then.
I have a hypothesis: if the ESP32 experiences poor communication with the router for a short period of time, causing a backlog of packets in its internal buffer, and it's trying to send those to the hotspot — is it possible that the ESP's internal buffer becomes full, and as a result, it stops accepting new data from the MCU?

LuciaferKelvin, 2025-04-15:
I think there might be something processed incorrectly here:
         1.From my tests, the ESP detects the heartbeat from the main controller. When the main controller doesn't send any messages to the ESP32 for some time, the ESP closes the STA and AP.
         2.If the ESP32 automatically closes the STA and AP, will it release all the network data packets?
         3.If point 2 is true, will the ESP drop all network packets when it hasn’t connected to a hotspot or opened the SoftAP?
         4.If point 3 is true, then this issue shouldn’t happen. There must be some reason that causes it not to drop all the packets.

    Checking Wi-Fi signal strength is not very practical, because based on my tests, signal strength doesn't fully represent the communication quality. In some cases, the signal strength is poor, but communication works fine. In other cases, the signal strength is strong, yet the ESP frequently disconnects from the router, and data transmission is often blocked.

LuciaferKelvin, 2025-04-15:
> > From my tests, the ESP detects the heartbeat from the main controller
> 
> I looked at the protobuf spec for release/v0.4 https://github.com/espressif/esp-hosted/blob/release/v0.4/common/proto/esp_hosted_config.proto , and I don't see a protobuf msg type for heartbeat.
> 
> Was the heartbeat added after release/v0.4?

It is not the heartbeat command. What I mean is that the ESP detects communication with the controller.
When I tested the SoftAP, if I only enabled it but didn’t send any messages (neither controller messages nor network messages), the ESP would close the SoftAP shortly afterward.
I haven’t checked the code, but the test results show this behavior.

SohKamYung-Espressif, 2025-04-15:
> When I tested the SoftAP, if I only enabled it but didn’t send any messages (neither controller messages nor network messages), the ESP would close the SoftAP shortly afterward.

I see.

Once you start the SoftAP, it should remain up.

When this happens, can you check to see if a `softap stop handler stop` message on the ESP console? This should come from the `softap_event_handler` https://github.com/espressif/esp-hosted/blob/release/v0.4/esp/esp_driver/network_adapter/main/slave_commands.c#L129

LuciaferKelvin, 2025-04-15:
That’s not the main point. The purpose of bringing this up is to confirm whether the ESP32 itself has a mechanism to monitor communication with the main controller, and if the communication is abnormal, it will shut down the previously enabled SoftAP and STA functions.

If such a mechanism exists, then the current issue shouldn’t be happening.

The key question is: why, after the buffer is full, the ESP is unable to release all the buffered data?
What is the current handling mechanism?
Does the ESP cache all the network data and attempt to resend it after reconnecting to Wi-Fi?
Or, if a communication failure with the main controller is detected, does it release all the buffered data when closing the STA connection?

LuciaferKelvin, 2025-04-15:
All the issues we are discussing are centered around this current phenomenon. I only discovered through testing that the ESP might have such a mechanism, but I cannot confirm it with certainty.
And if this mechanism does exist, what exactly is the behavior after it is triggered?

SohKamYung-Espressif, 2025-04-15:
Release/v0.4 was done in 2022. Since then there have been many changes to ESP-Hosted. So, it is also not possible for me to say at the moment where the issue lies.

One question: which version of ESP-IDF are your using to build the fw for the ESP32?

@kapilkedawat Could you take a look and see if there may be an explanation for the behaviour?

SohKamYung-Espressif, 2025-04-15:
You mean the ESP firmware binary was provided to you?

I can try to compile the release/v0.4 code and see if I can reproduce some of the issues you mentioned.

LuciaferKelvin, 2025-04-15:
> You mean the ESP firmware binary was provided to you?
> 
> I can try to compile the release/v0.4 code and see if I can reproduce some of the issues you mentioned.

sure

SohKamYung-Espressif, 2025-04-16:
@LuciaferKelvin Do you have access to the ESP32 console log from boot-up? If yes, please share it here.

It would be helpful as it contains information like the version of ESP-IDF used to build the firmware. It may also give clues as to why you are encountering the issues as the console log may print error messages when the error condition happens.

SohKamYung-Espressif, 2025-04-17:
I have built fw for a ESP32C3 based on ESP-Hosted slave release/v0.4. I am using a RaspPi 4 as the host as I don't have a STM32, but that should not matter.

Here is my debug log output for reference:

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xe (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x16b8
load:0x403cc710,len:0x944
load:0x403ce710,len:0x2dfc
entry 0x403cc710
I (30) boot: ESP-IDF v4.4.4 2nd stage bootloader
I (30) boot: compile time 15:35:15
I (30) boot: chip revision: v0.3
I (32) boot.esp32c3: SPI Speed      : 80MHz
I (37) boot.esp32c3: SPI Mode       : DIO
I (42) boot.esp32c3: SPI Flash Size : 4MB
I (46) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (70) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (78) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  3 factory          factory app      00 00 00010000 00100000
I (92) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (100) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (107) boot: End of partition table
I (112) boot: Defaulting to factory image
I (116) esp_image: segment 0: paddr=00010020 vaddr=3c0a0020 size=1c9c8h (117192) map
I (144) esp_image: segment 1: paddr=0002c9f0 vaddr=3fc93800 size=03400h ( 13312) load
I (146) esp_image: segment 2: paddr=0002fdf8 vaddr=40380000 size=00220h (   544) load
I (150) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=9609ch (614556) map
I (256) esp_image: segment 4: paddr=000c60c4 vaddr=40380220 size=1341ch ( 78876) load
I (277) boot: Loaded app from partition at offset 0x10000
I (277) boot: Disabling RNG early entropy source...
I (288) cpu_start: Pro cpu up.
I (297) cpu_start: Pro cpu start user code
I (297) cpu_start: cpu freq: 160000000
I (297) cpu_start: Application information:
I (300) cpu_start: Project name:     network_adapter
I (306) cpu_start: App version:      release/v0.4
I (311) cpu_start: Compile time:     Apr 16 2025 15:35:12
I (317) cpu_start: ELF file SHA256:  008bc22ef7c804cc...
I (323) cpu_start: ESP-IDF:          v4.4.4
I (328) heap_init: Initializing. RAM available for dynamic allocation:
I (335) heap_init: At 3FC9BBC0 len 00040B50 (258 KiB): DRAM
I (342) heap_init: At 3FCDC710 len 00002950 (10 KiB): STACK/DRAM
I (348) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (355) spi_flash: detected chip: generic
I (359) spi_flash: flash io: dio
I (364) sleep: Configure to isolate all GPIO pins in sleep state
I (370) sleep: Enable automatic switching of GPIO sleep configuration
I (377) coexist: coexist rom version 9387209
I (382) cpu_start: Starting scheduler.
I (387) NETWORK_ADAPTER: *********************************************************************
I (387) NETWORK_ADAPTER:                 ESP-Hosted Firmware version :: 0.4
I (397) NETWORK_ADAPTER:                 Transport used :: SPI
I (407) NETWORK_ADAPTER: *********************************************************************
I (417) NETWORK_ADAPTER: Supported features are:
I (427) NETWORK_ADAPTER: - WLAN over SPI
I (427) ESP_BT: - BT/BLE
I (437) ESP_BT:    - HCI Over SPI
I (437) ESP_BT:    - BLE only
I (447) BTDM_INIT: BT controller compile version [02bfb7f]
I (447) phy_init: phy_version 950,11a46e9,Oct 21 2022,08:56:12
I (487) system_api: Base MAC address is not set
I (487) system_api: read default base MAC address from EFUSE
[... 30 lines cut ...]
I (12197) slave_commands: station d6:cc:a7:4d:99:46 join, AID=1
I (16747) slave_commands: station d6:cc:a7:4d:99:46 leave, AID=1
I (540347) slave_commands: station d6:cc:a7:4d:99:46 join, AID=1
I (543837) slave_commands: station d6:cc:a7:4d:99:46 leave, AID=1
```

I can join and leave the softap at the start of the test and after about 9 minutes. During this time, the softap was not automatically closed.

Similarly, I started the ESP32C3 as a sta and joined a router.

```
I (46197) slave_commands: station mode set
I (48097) slave_commands: Connected to AP
I (48097) slave_commands: connected to ap SSID:'TestRouter', password:'TestRouterPassword'
I (48117) slave_commands: Get station mac address
I (48117) slave_commands: mac [7c:df:a1:da:d7:80]
```

The host controller was idle. But in this case, the router may be sending broadcast packet to the ESP32C3, so it may not be idle.

Again, the behaviour was expected. The connection was not automatically closed.

I will try to test throughput behaviour with a weak Wi-Fi signal next.

If you wish to test the fw I built, email me at sohkamyung<at>espressif<dot>com and I can send the `network_adapter.bin` file.

LuciaferKelvin, 2025-04-17:
Can you make sure that your main controller is not sending any data to the ESP32?
In my test, I completely stopped all data transmission to and from the ESP32 after enabling the SoftAP.
That means it neither receives data from the ESP32 nor sends any data to it.

SohKamYung-Espressif, 2025-04-21:
> Can you make sure that your main controller is not sending any data to the ESP32?
> In my test, I completely stopped all data transmission to and from the ESP32 after enabling the SoftAP.
> That means it neither receives data from the ESP32 nor sends any data to it.

Yes. I have verified that after I start the SoftAP, the main controller is not sending any data to the ESP32-C3 by connecting a logic analyser to the signals. See the attached image.

[image]

The burst of traffic at the start is ESP-Hosted initialisation and starting the SoftAP. The burst at +60 secs is me checking that the SoftAP is operating by joining and then leaving the SoftAP. After that, there is no SPI traffic for over 150 secs.

During that time, the Handshake signal remains Active.

LuciaferKelvin, 2025-04-21:
> Yes. I have verified that after I start the SoftAP, the main controller is not sending any data to the ESP32-C3 by connecting a logic analyser to the signals. See the attached image.
>
> [image]
>
> The [...]

Seems I am wrong, There have a question I should go to confirm why the softAp in my esp32 will auto close.

SohKamYung-Espressif, 2025-04-21:
> Seems I am wrong, There have a question I should go to confirm why the softAp in my esp32 will auto close.

Do you have the ESP32-C3 console output when this happens?

LuciaferKelvin, 2025-04-21:
I haven't yet. I will try to replicate it.
What happens is: when I try to connect the ESP32 to a hotspot, the SoftAP temporarily shuts down and then reopens.

SohKamYung-Espressif, 2025-04-21:
Could you also try to capture the SPI transactions using a logic analyzer? It is possible a SPI condition may have occurred that cause the Handshake signal to always remain inactive.

LuciaferKelvin, 2025-04-21:
This might take quite a long time, as the issue was only reproduced at the customer's site, and I don't have access to the location.
I've already spent some time trying to replicate it, but so far, I haven't been able to reproduce the same behavior.

LuciaferKelvin, 2025-04-24:
In my local environment, once the signal strength drops to that level, the Wi-Fi can't connect at all.

mantriyogesh, 2025-04-25:
Hello @LuciaferKelvin ,

I tried to fetch up some issues raised, #594 #175 #268 from you.
From the history, I can interpret that you are using MCU at host (Not Linux) and older release/0.4. Actually, this (current) repo is deprecated for microcontroller code base, as we have dedicated https://github.com/espressif/esp-hosted-mcu repo for MCU cases. The current repo is only supported for Linux FG and Linux NG.

However, we understand that you might have deployed the code with you already.

There were multiple points of confusion for me.

> when I try to connect the ESP32 to a hotspot, the SoftAP temporarily shuts down and then reopens.

Is this the only case when the issue happens?

## Confusion 1 : Wifi mode causing issues?

If so, for now, I am suspecting the `esp_wifi_set_mode()` on the fly causing these issues (unsure, but a suspect) on older IDF.

### Suggestion 1: Don't change the wifi mode on the fly

As I understand your case, you need station and softap both on the go. The most simplistic way would be not doing any mode changes on the fly. Just set the mode as: 

```
esp_err_t ensure_apsta_mode(void)
{
    wifi_mode_t current_mode;
    esp_err_t err = esp_wifi_get_mode(&current_mode);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "Failed to get current WiFi mode: %s", esp_err_to_name(err));
        return err;
    }

    if (current_mode != WIFI_MODE_APSTA) {
        err = esp_wifi_set_mode(WIFI_MODE_APSTA);
        if (err != ESP_OK) {
            ESP_LOGE(TAG, "Failed to set WiFi mode to APSTA: %s", esp_err_to_name(err));
            return err;
        }
        ESP_LOGI(TAG, "WiFi mode set to WIFI_MODE_APSTA");
    } else {
        ESP_LOGI(TAG, "WiFi mode is already WIFI_MODE_APSTA");
    }

    return ESP_OK;
}
```

Now remove every `esp_wifi_set_mode()` in https://github.com/espressif/esp-hosted/blob/release/v0.4/esp/esp_driver/network_adapter/main/slave_commands.c

And just call up above function always to set this mode, irrespective what you use, be it station or softap.

```c
esp_err_t err = ensure_apsta_mode();
if (err != ESP_OK) {
    // Handle error
}
```
### Reasoning 1
Why this change?
Once in WIFI_MODE_APSTA, you can control what’s active using:
esp_wifi_start() – Starts both AP and STA interfaces
esp_wifi_connect() – Initiates STA connection
esp_wifi_disconnect() – Stops STA connection
You can bring up/down softAP dynamically with esp_wifi_set_config(WIFI_IF_AP, ...) and/or esp_wifi_stop() if you want to fully turn off Wi-Fi.

This would at least **rule out** __break-before-make__ Wi-Fi case

## Confusion 2:  Handshake goes and stays low after some time.

For  handshake, the low is transient state, while the ESP is pushing next SPI transaction . The stable state of handshake is high. Low to high triggers interrupt to host (This is registered as ISR at host, to start up new spi transaction)

This is just for your information. We would suggest you **not** to change the SPI design code, as it becomes difficult later to understand how your code is different than with us.

Can you please attach the complete tar ball of esp-hosted code (including slave and host code) here using https://wetransfer.com/ ? This would give us a fair comparison point, for code understanding and issue reproduce and debugging.

LuciaferKelvin, 2025-04-25:
Hi @LuciaferKelvin ,

Thanks for your response.

The SoftAP disappearance issue that you are focusing on is not the key problem I'm raising here.
The actual issue I am experiencing is that the handshake pin remains low permanently after the ESP32 has been running for a while.

As I understand it, the handshake pin should be set high to notify the host that the ESP32 is idle and ready to receive the next command or network packet.
However, in my case, the pin gets stuck in the low state, which prevents the host from continuing communication with the ESP32.

The environment and behavior are similar to what's described in this comment:
👉 https://github.com/espressif/esp-hosted/issues/594#issuecomment-2827083393

To clarify again:
The SoftAP disappearing is not the issue in my case.
If you can confirm that no such mechanism exists, that's fine.What I wanted to point out is that I suspected whether the ESP32 might have some kind of heartbeat or idle detection mechanism — for example, when the host does not send any commands for a long time, the ESP32 may actively enter an idle state.

If such a mechanism does exist, then the handshake pin staying low wouldn't be an issue, because the ESP32 would be intentionally idling.
But if no such mechanism exists, that’s okay too — I just assumed there might be one based on the observed behavior.

That said, this is not the main issue, and I’m not too concerned about it at this point.
Previously, the reason I encountered the issue was that one of my task threads was getting blocked, which caused the host to stop sending any network commands or data to the ESP32 for some time.

In any case, this is not the root problem I’m trying to address here.

Thanks!

LuciaferKelvin, 2025-05-23:
Could you help me determine how much total memory this ESP32 firmware allocates for handling command reception and network data frames?

mantriyogesh, 2025-05-23:
Hello @LuciaferKelvin ,

There are a many ways to look at this. Wi-Fi , Bluetooth consumptions etc and ESP-Hosted consumption.
Even in ESP-Hosted-MCU, we have allocation for queues and tasks for handling the network frames (wifi, bluetooth), spi driver, control packets. But it all depends on the repo you use and may vary across commits, as recent changes went in for lowering the consumption and make it more controllable through the mempool.

Can you please let us know the repo and commit you use?
If it ESP-Hosted-MCU, I suggest moving this issue in that repo (which I can easily do upon confirmation)

LuciaferKelvin, 2025-05-23:
> Hello [@LuciaferKelvin](https://github.com/LuciaferKelvin) ,
>
> There are a many ways to look at this. Wi-Fi , Bluetooth consumptions etc and ESP-Hosted consumption. Even in ESP-Hosted-MCU, we have [...]

Can I find this information from the logs?

LuciaferKelvin, 2025-05-23:
ESP-Hosted Firmware version :: 0.4    the frimware verion

LuciaferKelvin, 2025-05-23:
the idf version:[0;32mI (30) boot: ESP-IDF v4.4.4-235-gfee50c208a 2nd stage bootloader[0m****

LuciaferKelvin, 2025-05-23:
[0;32mI (394) NETWORK_ADAPTER: *********************************************************************[0m
[0;32mI (394) NETWORK_ADAPTER:                 ESP-Hosted Firmware version :: 0.4                        [0m
[0;32mI (404) NETWORK_ADAPTER:                 Transport used :: SPI                           [0m
[0;32mI (414) NETWORK_ADAPTER: *********************************************************************[0m
[0;32mI (424) NETWORK_ADAPTER: Supported features are:[0m
[0;32mI (434) NETWORK_ADAPTER: - WLAN over SPI[0m

mantriyogesh, 2025-05-23:
This is little older release.

However, for easiest way, you can embed and call this macro
```
#define MEM_DUMP(s) \
    printf("%s free:%lu min-free:%lu lfb-dma:%u lfb-def:%u lfb-8bit:%u\n", s, \
                  esp_get_free_heap_size(), esp_get_minimum_free_heap_size(), \
                  heap_caps_get_largest_free_block(MALLOC_CAP_DMA),\
                  heap_caps_get_largest_free_block(MALLOC_CAP_DEFAULT),\
                  heap_caps_get_largest_free_block(MALLOC_CAP_8BIT))
```

at start ([here](https://github.com/espressif/esp-hosted/blob/74e707b604280ecb2eafa60ba5ed6993ce57b457/esp/esp_driver/network_adapter/main/app_main.c#L675))

periodically at
https://github.com/espressif/esp-hosted/blob/74e707b604280ecb2eafa60ba5ed6993ce57b457/esp/esp_driver/network_adapter/main/app_main.c#L757
```
--      ESP_LOGI(TAG,"Initial set up done");
--  }
++ ESP_LOGI(TAG,"Initial set up done");
++ while (1) {
++     MEM_DUMP("periodic");
++ }
```

`current - start` would lead to correct use.

How to optimise:
1. in latest commits, we have integrated mempool, which can be integrated, which would ensure there is no or minimal transient 'heap' allocations. instead always rely on mempool pre allocated memoiry on bootup

2. Hosted one time cost (doesn't change)
  - 2.1 app_main.c - https://github.com/espressif/esp-hosted/blob/release/v0.4/esp/esp_driver/network_adapter/main/app_main.c#L739-L749
  - 2.2 spi - https://github.com/espressif/esp-hosted/blob/release/v0.4/esp/esp_driver/network_adapter/main/spi_slave_api.c#L508-L517
  - 2.3 spi - https://github.com/espressif/esp-hosted/blob/74e707b604280ecb2eafa60ba5ed6993ce57b457/esp/esp_driver/network_adapter/main/spi_slave_api.c#L96 * 1600
  - 2.4 protocomm - https://github.com/espressif/esp-hosted/blob/74e707b604280ecb2eafa60ba5ed6993ce57b457/esp/esp_driver/network_adapter/main/protocomm_pserial.c#L263
  - 2.5 protocomm - https://github.com/espressif/esp-hosted/blob/74e707b604280ecb2eafa60ba5ed6993ce57b457/esp/esp_driver/network_adapter/main/protocomm_pserial.c#L259

3. Dynamic (on-off dependoing upon current load)
  - `heap_caps_malloc` for Tx and Rx buffers of 1600 towards lower spi driver and `malloc` of `sizeof(spi_slave_transaction_t)` 
  - The peaks depend upon https://github.com/espressif/esp-hosted/blob/74e707b604280ecb2eafa60ba5ed6993ce57b457/esp/esp_driver/network_adapter/main/spi_slave_api.c#L95-L110
  - slave_control.c - The control packet on receival , are malloced for request and responses, as per the proto buffer size. But they should be small and transient

Major allocations can be optimised in app_main.c and spi_slave_api.c for 

Impatient to lower memory use?
1. Use latest master. We were not very conscious earlier releases.
2. No I want to use release/0.4 only.
 - Quick release of memory would happen if you remove the 'PRIORITY_QUEUE' implementation and let all buffers fall in same bucket.
 - Tx queues and Rx queus could be optimized. Please be cautious, a lot lower queue size would end up lower throughput
3. Don't use bluetooth? Disable from Kconfig altogether.
4. You can do optimizations as stated in https://docs.espressif.com/projects/esp-idf/en/v4.4/esp32c3/api-guides/performance/ram-usage.html which is more of system way of lowering iram use. For example, moving freertos functions to flash.

Please let us know any further detail you wish or the intention behind the question, to answer which could be potential beneficial for you.

Again, thanks for being with us. If anything you still wish us to customise, please let us know. Within our possible support limits, we would be happy to do.

LuciaferKelvin, 2025-05-23:
Thank you. I will check it.

closed 2025-05-23 by LuciaferKelvin