espressif/esp-hosted #595: ESP32-WROOM-32E-N4 failed to load driver on Linux core board with SDIO interface

MaJianHao123, 2025-04-15:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

CPU：RK3588
ESP32 firmware：ESP-Hosted-NG_release_v1.0.2\esp32\spi_only
interface：SDIO
The firmware is burned successfully, but an error occurs when loading the driver。

[image]

Using the same hardware, the driver can be successfully loaded on TI's AM6254

What might be the cause?

There is an error in the log report after upgrading the firmware. Is this the reason?

[image]

SohKamYung-Espressif, 2025-04-15:
For the RK3588 board, did you add the SDIO pullups required and followed the guidelines for SDIO https://docs.espressif.com/projects/esp-idf/en/latest/esp32c6/api-reference/peripherals/sd_pullup_requirements.html ?

Did you change the partition information for the ESP32?

MaJianHao123, 2025-04-15:
The SDIO connection is connected according to the documentation。
Here is schematic

[image]

How to change the partition information for the ESP32?

SohKamYung-Espressif, 2025-04-16:
@kapilkedawat Do you know what may be the cause of the `invalid magic byte` error reported by `esp_image`?

SohKamYung-Espressif, 2025-04-16:
@MaJianHao123 The schematic should be okay, since you got the ESP32 working with TI's AM6254.

Upon loading the driver, the kernel tried to probe for the SDIO card but failed.

There can by many issues with getting SDIO to work, related to how the SDMMC signals are routed:
- are the signals of equal length
- do the signals have an adequate ground plane

If possible, try with a lower SDIO CLK frequency first, or switch to one-bit SDIO mode on the host. You may also want to tap the SDMMC signals (start with the CLK and CMD signals) with an oscilloscope to see if they are noisy or not.

You can use your working AM6254 as a reference.

MaJianHao123, 2025-04-16:
> [@kapilkedawat](https://github.com/kapilkedawat) Do you know what may be the cause of the error reported by ?`invalid magic byte``esp_image`

I solved this firmware problem by replacing the firmware provided by Espressif technicians, but the problem with SDIO communication has not been solved yet.

kapilkedawat, 2025-04-16:
@MaJianHao123 could you please try the latest firmware and host? Also before checking for module working, we need to make sure SDIO communication is up and running, once latest code is present at both host and firmware, please try to run raw throughput first and share us the logs on both firmware and host side.

MaJianHao123, 2025-04-17:
Using the latest firmware from the github，now the one-bit SDIO mode of SDIO can communicate normally, but the 4-bit SDIO cannot

SohKamYung-Espressif, 2025-04-17:
> Using the latest firmware from the github，now the one-bit SDIO mode of SDIO can communicate normally, but the 4-bit SDIO cannot

This points to an issue with the SDMMC signals. Please check your PCB routing, and use of pull-ups on the SDMMC signals.

What CLK frequency are you using? 40 MHz? Try a lower CLK, like 10 or 20 MHz.

mantriyogesh, 2025-04-24:
Have you tried 
```
broken-cd;
```
in @MaJianHao123 ?