espressif/esp-hosted #596: SDIO 4-Bit mode not working / DAT2 pulled down by ESP32

olell, 2025-04-21:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

I'm currently porting esp-hosted to a custom board using a Rockchip RV1106G3 as hosting SoC and an ESP32-WROOM-32. In 1-Bit mode the SDIO communication between the ESP32 and the SoC works fine except the low data throughput. Currently I'm trying to switch to 4-Bit mode and encountered a problem with the DAT2 line.

The connections on the board are kept short, below 30mm, referencing to a solid ground plane and very similar length. I will attach a schematic diagram. According to the documentation I added 51kΩ pull up resistors on CMD & DAT0..3. 

I detected the following behavior: All lines are working correctly in 4-Bit mode except DAT2, there I can measure a voltage of around 1.5V which stays permanently. Therefore I guess the initialization of the driver fails. I put both chips in reset state to test the influence of those chips on the line:

| Chip in Reset | Voltage on DAT2 |
|--------------|-----------------:|
| ESP32            | 3.3V                    |
| None              | 1.5V                     |
| RV1106G3     | 0V                        |

from that I concluded, that the ESP32 is pulling the line down with a similar resistor as the SoC is pulling it up. My conclusion is that it might have something to do with the MTDI bootstrapping pin.

But I burned the efuses of the ESP32 so that the MTDI strapping pin is disabled:
```
Vdd fuses:
XPD_SDIO_REG (BLOCK0)                              read for XPD_SDIO_REG                              = True R/W (0b1)
XPD_SDIO_TIEH (BLOCK0)                             If XPD_SDIO_FORCE & XPD_SDIO_REG                   = 3.3V R/W (0b1)
XPD_SDIO_FORCE (BLOCK0)                            Ignore MTDI pin (GPIO12) for VDD_SDIO on reset     = True R/W (0b1)
```

Is there anything in the firmware or configuration that I might have overseen or can you maybe help me figure out why the ESP32 is pulling that line down. Or is it maybe something completely different. I'm beginning to despair because I can't find the error :/

`# dmesg | grep -C4 esp32`
```
[    5.713461] g_cdc gadget.0: CDC Composite Gadget, version: King Kamehameha Day 2008
[    5.713476] g_cdc gadget.0: g_cdc ready
[    5.728879] g_cdc gadget.0: suspend
[    5.857192] usb0: eth_open
[    5.966190] esp32_sdio: loading out-of-tree module taints kernel.
[    7.232466] gs_open: ttyGS0 (b05e7824,e6a6bf49)
[    7.241472] gs_close: ttyGS0 (b05e7824,e6a6bf49) ...
[    7.241550] gs_close: ttyGS0 (b05e7824,e6a6bf49) done!
[    7.243320] gs_open: ttyGS0 (2c3e93e5,99650811)
--
[    8.691486] mmc2: queuing unknown CIS tuple 0x01 [d9 01 ff] (3 bytes)
[    8.699221] mmc2: queuing unknown CIS tuple 0x1a [01 01 00 02 07] (5 bytes)
[    8.702663] mmc2: queuing unknown CIS tuple 0x1b [c1 41 30 30 ff ff ff ff] (8 bytes)
[    8.705858] mmc2: new SDIO card at address 0001
[    8.706685] esp32_sdio: esp_probe: ESP network device detected
[    8.707328] esp32_sdio: probe of mmc2:0001:1 failed with error -84
[    8.707811] esp32_sdio: probe of mmc2:0001:2 failed with error -22
[    9.844665] g_cdc gadget.0: packet filter 0e
[    9.844720] g_cdc gadget.0: ecm req21.43 v000e i0000 l0
[    9.903971] g_cdc gadget.0: packet filter 0e
[    9.904017] g_cdc gadget.0: ecm req21.43 v000e i0000 l0
```

`# cat /sys/kernel/debug/mmc2/ios`
```
clock:          25000000 Hz
vdd:            21 (3.3 ~ 3.4 V)
bus mode:       2 (push-pull)
chip select:    0 (don't care)
power mode:     2 (on)
bus width:      2 (4 bits)
timing spec:    0 (legacy)
signal voltage: 0 (3.30 V)
driver type:    0 (driver type B)
```

[... 169 lines cut ...]
I (631) wifi:Init management short buffer num: 32
I (635) wifi:Init dynamic tx buffer num: 64
I (639) wifi:Init static rx buffer size: 1600
I (644) wifi:Init static rx buffer num: 16
I (647) wifi:Init dynamic rx buffer num: 40
I (652) wifi_init: rx ba win: 32
I (654) wifi_init: accept mbox: 6
I (657) wifi_init: tcpip mbox: 32
I (660) wifi_init: udp mbox: 6
I (663) wifi_init: tcp mbox: 6
I (666) wifi_init: tcp tx win: 5760
I (669) wifi_init: tcp rx win: 5760
I (672) wifi_init: tcp mss: 1440
I (675) wifi_init: WiFi IRAM OP enabled
I (679) wifi_init: WiFi RX IRAM OP enabled
I (683) BTDM_INIT: BT controller compile version [20b459d]
I (689) BTDM_INIT: Bluetooth MAC: e0:e2:e6:d6:42:1e
I (695) phy_init: phy_version 4840,02e0d70,Sep  2 2024,19:39:07
I (1075) FW_MAIN: ESP Bluetooth MAC addr: e0-e2-e6-d6-42-1e

I (1075) FW_SDIO_SLAVE: Using SDIO interface
I (2077) gpio: GPIO[22]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
```
There is nothing more is happening after this line was printed
</details>

SohKamYung-Espressif, 2025-04-22:
@olell  I believe you have burned the efuses following this instructions, https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sd_pullup_requirements.html#strapping-conflicts-dat2 ?

In your table, when the RV1106G3 is in reset, is the ESP32 running? And the voltage you see on DAT2 is 0V? This does not look correct, as the ESP32 should not be driving DAT2.

What is the voltage when both chips are in reset state? It should be 3.3V.

@kapilkedawat Can you take a look?

olell, 2025-04-22:
Hello,

Yes I burned the efuses according to the instructions you've linked.

Currently I'm not at the workplace and don't have access to the board, but I am pretty sure that the DAT2 line has 3.3V if both chips are in reset state. I will measure it later when I have access to it.

olell, 2025-04-22:
I just measured a few more voltages and states. This is my result. To check if there is something wrong with the hardware I also wrote a little script which "blinks" this particular pin to see if it does that and that works quite fine it blinks between 0 and 3.3V.

Here are the results of my measurements:

|                  | No external pull-up | External pull-up 47kΩ |
| ---------------- | ------------------- | --------------------- |
| No chip in reset | 3.30V               | 3.3V                  |
| RV1106 in reset  | 1.72V               | 2.16V                 |
| ESP32 in reset   | 3.30V               | 3.3V                  |
| Both in reset    | 0V                  | 1.32V                 |

In the setting where both are in reset, if i release the reset state of the ESP32 it shortly drops down to 0.7V until the firmware has started. Maybe the boot loader does something with this pin?

Repeated using pre compiled esp-hosted firmware. Linux host unchanged:

|                  | External pull-up 47kΩ |
| ---------------- | --------------------- |
| No chip in reset | 3.3V                  |
| RV1106 in reset  | 1.32V                 |
| ESP32 in reset   | 3.3V                  |
| Both in reset    | 1.32V                 |

Difference between those firmware: probably internal pull-up resistor enabled and not

Then I cutted the PCB trace, pull up resistor is on ESP32 side:

| External pull-up 47kΩ | ESP32 Side | SoC Side |
| --------------------- | ---------- | -------- |
| IC in reset           | 3.3V       | 0V       |
| IC not in reset       | 3.3V       | 3.3V     |
| IC booting            | 1V         | N/A      |

SohKamYung-Espressif, 2025-04-23:
Can you provide information on the specific ESP32 module you are using?

There is a list of ESP32 related SDIO compatibility issues which is Soc or Module dependent https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sd_pullup_requirements.html#overview-of-compatibility

mantriyogesh, 2025-04-24:
Hello @olell ,

I have gone through the thread again, below are some understanding and observations:
1. I also understand that you are unsure of external pull-up value. >10KOhm is okay. We use 51KOhm. In case you wish to get our schematic, please feel free to mail me.
 
2. You already have done the following:
```
components/esptool_py/esptool/espefuse.py set_flash_voltage 3.3V
```
and external pull-ups on CMD, DAT0, DAT1, DAT2 , DAT3 with >=10KOhm.

3. Are you able to flash the esp32-wroom-32 reliably? if so, no need to care for 'download mode not working'

4. In normal working conditions, you observe that you get 3.3V on all the SDIO pins.

If you reached point (3), everything is fine, except one thing, which is SDIO interference.

5. From logs at host,
```
[    8.707328] esp32_sdio: probe of mmc2:0001:1 failed with error -84
```

> #define	[EILSEQ](https://elixir.bootlin.com/u-boot/v2021.01-rc3/source/include/linux/errno.h#L92)		84	/* Illegal byte sequence */

- Possible causes:

  - 5.1 Incorrect SDIO pull-up resistors (esp. on CMD and DAT lines).

  - 5.2 Clock too high or too low.

  - 5.3 Signal integrity problems.

  - 5.4 Mismatch in SDIO firmware settings on the ESP32 side. ([Timing tuning](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md#242-sdio) explained in porting guide)

  - 5.5 SoC is sufficiently powered and 3.3V on SDIO lines is consistent (in normal SDIO operation)

6. From logs
```
There is nothing more is happening after this line was printed
```

Here ESP SDIO slave is waiting to get first communication from host to slave, which would print at ESP:
https://github.com/espressif/esp-hosted/blob/d9e607d3d7edef6636b78d75807c3331c073ba30/esp_hosted_ng/esp/esp_driver/network_adapter/main/app_main.c#L611

which is triggered from host esp driver at https://github.com/espressif/esp-hosted/blob/d9e607d3d7edef6636b78d75807c3331c073ba30/esp_hosted_ng/host/sdio/esp_sdio.c#L762

6.1 From the logs, I can see the device was tried to probe. Which exact code leg did the Linux driver started to return -84 would be the key 
https://github.com/espressif/esp-hosted/blob/d9e607d3d7edef6636b78d75807c3331c073ba30/esp_hosted_fg/host/linux/host_driver/esp32/sdio/esp_sdio.c#L722

nichonaugle, 2025-07-28:
@olell Can you share your dts, I am currently working to implement the same thing except with an ESP32-C6 over SDIO to the RV1106G3. I have seen a couple others struggling with HOST to SLAVE communication when using this chipset. The more data I have the sooner I can provide a solution. I have another thread up where I got SPI to work, however it was at a very low throughput. Thanks

olell, 2025-07-28:
@nichonaugle 

Sure, my device tree and the rest of my buildroot setup is in this repo. The bus-width is currently set to 1, as described with 4 bits it doesn't work.

https://github.com/olell/rv1106-buildroot/blob/platform/soundcube-digiboard/board/RV1106/rv1106-custom.dts

nichonaugle, 2025-07-28:
@olell I took a look at your DTS. You should remove no-sdio as that could be part of your problem. Combined with the no-mmc, it causes the host to recognize the device as an SD card which it is not.

[image]

I just got a connection over SDIO between the C6 and luckfox pico ultra this morning, I will followup with more updates once the DTS is finalized. Below is my current **NOT FINAL** dts: 

[image]

Another note: If you are using the luckfox sdk to build the OS, please disable all other functions on the GPIOS you are using for SDIO. I noticed their examples have a tendency to overlap with custom DTS configurations and initialize gpios as they load their kernel modules in via init.d. This causes strange things to happen and errors that are hard to diagnose.

nichonaugle, 2025-07-28:
I further stripped down the parameters and found that the RV1106 would not work at all when interfacing with the esp32 over 4 bit mode. changing to 1-bit worked: 

[image]

However, I am facing communication issues from Host -> ESP, with raw_tp_tests only giving 0 kbits/sec. ESP-> Host testing yields 33249 kbits/sec. Given that I had a similar issue on a different RV1106 processor running esp-hosted over SPI, I am inclined to believe it is a configuration error on the RV1106 buildroot side