espressif/esp-hosted #597: ESP32-P4 EVK C6 - Not enough heap memory

simoberny, 2025-04-22:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Hi, I'm currently working with the ESP32-P4 EVK using the esp-hosted solution in softAP mode.
On the P4, I have a running web server with an image streaming endpoint (similar to the old esp-who example).

### Actual behavior (suspected bug)

The same code works flawlessly on my old ESP32-S3, but on the P4 I'm encountering the following issue:
[image]

The only way to get it working reliably is to lower the JPEG quality to 25. This leads me to believe it's related to the communication buffer size or heap constraints between the P4 (host) and C6 (Wi-Fi chip). It does not seems to be deterministic. Increasing the time between the frame send seems also to help. 
I’ve followed the specific configuration steps for the EVK on the docs. 

Any hint or tweaks on how to get rid of this issue?

Thank you,
Best regards

### Error logs or terminal output

```shell
E (73247) dma_utils: esp_dma_capable_malloc(181): Not enough heap memory

assert failed: transport_drv_ap_tx transport_drv.c:256 (copy_buff)
Core  0 register dump:
MEPC    : 0x4ff00dea  RA      : 0x4ff0aa9c  SP      : 0x4ff2b9a0  GP      : 0x4ff15600  
--- 0x4ff00dea: panic_abort at C:/Users/Simone.Bernabe/esp/v5.4.1/esp-idf/components/esp_system/panic.c:468
--- 0x4ff0aa9c: flash_read at C:/Users/Simone.Bernabe/esp/v5.4.1/esp-idf/components/esp_system/port/image_process.c:77

TP      : 0x4ff2bc30  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x00000001  S1      : 0x00000085  A0      : 0x4ff2b9e8  A1      : 0x4ff16ef1
A2      : 0x00000001  A3      : 0x4ff1b000  A4      : 0x00000001  A5      : 0x4ff1b000
A6      : 0x00000008  A7      : 0x76757473  S2      : 0x4ff2b9e8  S3      : 0x4ff2b9f2
S4      : 0x4ff16ef0  S5      : 0x4ff2b9e8  S6      : 0x4ff3aaa8  S7      : 0x00000001
S8      : 0x00000006  S9      : 0x4ff41014  S10     : 0x4ff1b000  S11     : 0x4ff3aa50
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00011880  MTVEC   : 0x4ff00003  MCAUSE  : 0x00000002  MTVAL   : 0x00000000
--- 0x4ff00003: _vector_table at ??:?

MHARTID : 0x00000000

Stack memory:
4ff2b9a0: 0x6e6d6c6b 0x6a696867 0x66656463 0x4ff13c1a 0x00363532 0x0000009e 0x00000000 0x00000000
--- 0x4ff13c1a: esp_ptr_in_drom at C:/Users/Simone.Bernabe/esp/v5.4.1/esp-idf/components/esp_hw_support/include/esp_memory_utils.h:335
---  (inlined by) __assert_func at C:/Users/Simone.Bernabe/esp/v5.4.1/esp-idf/components/newlib/assert.c:62

4ff2b9c0: 0x00000000 0x4ff1756c 0x4818f1c4 0x4ff17550 0x48165413 0x4ff1757c 0x4ff2b9b0 0x4ff17580
4ff2b9e0: 0x4816544c 0x4ff16ef0 0x65737361 0x66207472 0x656c6961 0x74203a64 0x736e6172 0x74726f70
4ff2ba00: 0x7672645f 0x5f70615f 0x74207874 0x736e6172 0x74726f70 0x7672645f 0x323a632e 0x28203635
4ff2ba20: 0x79706f63 0x6675625f 0x4f002966 0x4ff01206 0x00000001 0x00000008 0x00000680 0x4ff2ba54
--- 0x4ff01206: heap_caps_aligned_alloc_base at C:/Users/Simone.Bernabe/esp/v5.4.1/esp-idf/components/heap/heap_caps_base.c:158

4ff2ba40: 0x000005d6 0x00000000 0x00000008 0x4810185a 0x00000008 0x00011e1f 0x48180d7c 0x4819cf44
--- 0x4810185a: esp_dma_capable_malloc at C:/Users/Simone.Bernabe/esp/v5.4.1/esp-idf/components/esp_hw_support/dma/esp_dma_utils.c:181 (discriminator 5)

4ff2ba60: 0x000000b5 0x00000551 0x00001000 0x0000009e 0x000005d6 0x4ff4b306 0x00000001 0x00000040
4ff2ba80: 0x000005d6 0x4ff2cdfc 0x00000001 0x4ff1b000 0x00000000 0x00000600 0x4ff23094 0x480133a6
--- 0x480133a6: mempool_alloc at C:/Users/Simone.Bernabe/workspace/ZCam/managed_components/espressif__esp_hosted/host/drivers/mempool/mempool.c:115 (discriminator 1)

4ff2baa0: 0x00000000 0x00000000 0x00000000 0x00000040 0x4ff3aaa8 0x4ff2cdfc 0x4ff41000 0x4ff1b000
4ff2bac0: 0x00000000 0x4ff40fde 0x000005d6 0x4800cea0 0x4ff2cd70 0x4ff40fb8 0x4ff40fb8 0x48135b4c
--- 0x4800cea0: transport_drv_sta_tx at C:/Users/Simone.Bernabe/workspace/ZCam/managed_components/espressif__esp_hosted/host/drivers/transport/transport_drv.c:219
--- 0x48135b4c: low_level_output at C:/Users/Simone.Bernabe/esp/v5.4.1/esp-idf/components/esp_netif/lwip/netif/wlanif.c:92

4ff2bae0: 0x0000fc13 0x4ff40fb8 0x4ff3aa90 0x48110082 0x00000000 0x000005b2 0x000005b4 0x000005b4
--- 0x48110082: ip4_output_if_opt_src at C:/Users/Simone.Bernabe/esp/v5.4.1/esp-idf/components/lwip/lwip/src/core/ipv4/ip4.c:1092

4ff2bb00: 0x00000006 0x4ff41014 0x00000006 0x00000001 0x4ff2cdfc 0x4ff3aaa8 0x4ff41000 0x4ff3aa3c
4ff2bb20: 0x0000fc13 0x00000002 0x4ff3aa90 0x481101ec 0x00000000 0x4ff3aaa8 0x4ff41000 0x4ff3aa3c
--- 0x481101ec: ip4_output_if at C:/Users/Simone.Bernabe/esp/v5.4.1/esp-idf/components/lwip/lwip/src/core/ipv4/ip4.c:870

4ff2bb40: 0x0000fc13 0x00000002 0x4ff3aa90 0x4810a86e 0x4ff2df8c 0x05a0ac
--- 0x4810a86e: tcp_output at C:/Users/Simone.Bernabe/esp/v5.4.1/esp-idf/components/lwip/lwip/src/core/tcp_out.c:1365

44 0x00000000 0x02040101
4ff2bb60: 0x0000ffff 0x4ff3ab84 0x4ff1b000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000003
4ff2bb80: 0x0000ff7c 0x00000001 0x00000000 0x00010000 0x0000ffff 0x4ff3ab84 0x00000000 0x4811ecd6
--- 0x4811ecd6: lwip_netconn_do_writemore at C:/Users/Simone.Bernabe/esp/v5.4.1/esp-idf/components/lwip/lwip/src/api/api_msg.c:1833

4ff2bba0: 0x4ff33b40 0x4ff1b000 0x00000000 0x00000005 0x00000004 0x00000002 0x00000001 0x00000003
4ff2bbc0: 0x4ff33b40 0x4ff1b000 0xffffffff 0x48102052 0x00000000 0x00000000 0x00000000 0x4ff33b40
--- 0x48102052: tcpip_thread_handle_msg at C:/Users/Simone.Bernabe/esp/v5.4.1/esp-idf/components/lwip/lwip/src/api/tcpip.c:208
---  (inlined by) tcpip_thread at C:/Users/Simone.Bernabe/esp/v5.4.1/esp-idf/components/lwip/lwip/src/api/tcpip.c:148

4ff2bbe0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff2bc00: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff2bc20: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000000 0x4ff2ae60 0x4ff2ae60 0x00000000
4ff2bc40: 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff2bc60: 0x00000160 0x4ff2b930 0x00011941 0x4ff185a8 0x4ff185a8 0x4ff2bc64 0x4ff185a0 0x00000007
4ff2bc80: 0x4ff2ad28 0x4ff2ad28 0x4ff2bc64 0x00000000 0x00000012 0x4ff2ae60 0x00546974 0x00000000
4ff2bca0: 0x00000000 0x00000000 0x00000000 0x4ff2bc50 0x0000000e 0x00000000 0x00000012 0x00000000
4ff2bcc0: 0x00000000 0x00000000 0x000409f4 0x00000000 0x4ff1b5dc 0x4ff1b644 0x4ff1b6ac 0x00000000
4ff2bce0: 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x48005fd4 0x00000000 0x00000000
--- 0x48005fd4: esp_cleanup_r at C:/Users/Simone.Bernabe/esp/v5.4.1/esp-idf/components/newlib/newlib_init.c:43

4ff2bd00: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff2bd20: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff2bd40: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff2bd60: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff2bd80: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
```

### Steps to reproduce the behavior

1. Setup P4EVK with suggested settings
2. Setup wifi as soft AP with DHCP 
3. Setup a webserver with streaming images (1280x720, downsampled to YUV420)

### Project release version

1.4.1

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 11

### Shell

ZSH

SohKamYung-Espressif, 2025-04-22:
Hi, @simoberny . Could you share the console log output from the P4 from startup?

simoberny, 2025-04-22:
> Hi, [@simoberny](https://github.com/simoberny) . Could you share the console log output from the P4 from startup?

```bash
ESP-ROM:esp32p4-eco2-20240710
Build:Jul 10 2024
rst:0x1 (POWERON),boot:0x30f (SPI_FAST_FLASH_BOOT)
SPI mode:DIO, clock div:1
load:0x4ff33ce0,len:0x177c
load:0x4ff2abd0,len:0xf20
load:0x4ff2cbd0,len:0x33fc
entry 0x4ff2abda
I (25) boot: ESP-IDF v5.4.1 2nd stage bootloader
I (26) boot: compile time Apr 15 2025 16:02:03
I (26) boot: Multicore bootloader
I (27) boot: chip revision: v1.0
I (29) boot: efuse block revision: v0.1
I (32) qio_mode: Enabling default flash chip QIO
I (37) boot.esp32p4: SPI Speed      : 80MHz
I (40) boot.esp32p4: SPI Mode       : QIO
I (44) boot.esp32p4: SPI Flash Size : 16MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (61) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (68) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (74) boot:  2 factory          factory app      00 00 00010000 00600000
I (81) boot:  3 storage          Unknown data     01 82 00610000 00100000
I (87) boot:  4 coco_det         Unknown data     01 82 00710000 00400000
I (95) boot: End of partition table
I (97) esp_image: segment 0: paddr=00010020 vaddr=48160020 size=410b4h (266420) map
I (147) esp_image: segment 1: paddr=000510dc vaddr=30100000 size=0002ch (    44) load
I (148) esp_image: segment 2: paddr=00051110 vaddr=3010002c size=0003ch (    60) load
I (152) esp_image: segment 3: paddr=00051154 vaddr=4ff00000 size=0eec4h ( 61124) load
I (170) esp_image: segment 4: paddr=00060020 vaddr=48000020 size=1541a8h (1393064) map
I (384) esp_image: segment 5: paddr=001b41d0 vaddr=4ff0eec4 size=05ef4h ( 24308) load
I (391) esp_image: segment 6: paddr=001ba0cc vaddr=4ff14e00 size=0342ch ( 13356) load
I (399) boot: Loaded app from partition at offset 0x10000
I (399) boot: Disabling RNG early entropy source...
I (410) hex_psram: vendor id    : 0x0d (AP)
I (410) hex_psram: Latency      : 0x01 (Fixed)
I (410) hex_psram: DriveStr.    : 0x00 (25 Ohm)
I (411) hex_psram: dev id       : 0x03 (generation 4)
I (416) hex_psram: density      : 0x07 (256 Mbit)
I (420) hex_psram: good-die     : 0x06 (Pass)
I (424) hex_psram: SRF          : 0x02 (Slow Refresh)
I (429) hex_psram: BurstType    : 0x00 ( Wrap)
I (433) hex_psram: BurstLen     : 0x03 (2048 Byte)
I (438) hex_psram: BitMode      : 0x01 (X16 Mode)
I (442) hex_psram: Readlatency  : 0x04 (14 cycles@Fixed)
I (447) hex_psram: DriveStrength: 0x00 (1/1)
I (451) MSPI DQS: tuning success, best phase id is 2
I (628) MSPI DQS: tuning success, best delayline id is 11
I esp_psram: Found 32MB PSRAM device
I esp_psram: Speed: 200MHz
I (644) mmu_psram: .rodata xip on psram
I (712) mmu_psram: .text xip on psram
I (713) hex_psram: psram CS IO is dedicated
I (713) cpu_start: Multicore app
I (1160) esp_psram: SPI SRAM memory test OK
I (1169) cpu_start: Pro cpu start user code
I (1170) cpu_start: cpu freq: 360000000 Hz
I (1170) app_init: Application information:
I (1170) app_init: Project name:     ZCam
I (1174) app_init: App version:      0.2.0
I (1177) app_init: Compile time:     Apr 22 2025 10:49:55
I (1183) app_init: ELF file SHA256:  250dab2e5...
I (1187) app_init: ESP-IDF:          v5.4.1
I (1191) efuse_init: Min chip rev:     v0.1
I (1195) efuse_init: Max chip rev:     v1.99
I (1199) efuse_init: Chip rev:         v1.0
[... 74 lines cut ...]
I (2713) transport: Base transport is set-up

I (2717) transport: Slave chip Id[12]
I (2721) hci_stub_drv: Host BT Support: Disabled
I (2725) H_SDIO_DRV: Received INIT event
I (3734) wifi: wifi_init_softap finished. SSID:zcam-p4 password:p5zcam-p4 channel:10
SPIFFS: Total: 956561, Used: 273841
I (3755) main_task: Returned from app_main()
I (3755) webserver: Starting web server on port: '80'
I (3756) webserver: Starting stream server on port: '81'
I (3758) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1
I (3770) sc2336: Detected Camera sensor PID=0xcb3a
I (3919) app_video: version: 0.9.0
I (3919) app_video: driver:  MIPI-CSI
I (3919) app_video: card:    MIPI-CSI
I (3919) app_video: bus:     esp32p4:MIPI-CSI
I (3922) app_video: width=1280 height=720
I (3935) app_camera_pipeline: new elements[0]:0x48c49680, internal:1
I (3935) app_camera_pipeline: new pipeline 0x485412d4, elem_num:1
I (3938) Camera: Start camera stream task
I (3941) app_video: Video Stream Start
I (3946) FbsLoader: The storage free size is 65536 KB
I (3950) FbsLoader: The partition size is 4096 KB
W (3954) FbsLoader: There is only one model in the flatbuffers, ignore the input model name!
```

SohKamYung-Espressif, 2025-04-23:
> I (1236) esp_psram: Adding pool of 30720K of PSRAM memory to heap allocator

From your log, I see you have enabled PSRAM. That should provide more memory for holding images.

You can get some info on your heap memory usage. See the [Heap Memory Allocation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32p4/api-reference/system/mem_alloc.html) documentation for APIs to display the amount of memory being allocated.

ESP-Hosted itself uses some DMA-Capable heap memory for buffer queues to transfer data to and from the ESP32-C6 co-processor. You can adjust the size of the queues via `idf.py menuconfig` -> `Component config` -> `ESP-Hosted config` -> `Hosted SDIO Configuration` -> `Host SDIO Tx/Rx queue size`.

simoberny, 2025-04-23:
> > I (1236) esp_psram: Adding pool of 30720K of PSRAM memory to heap allocator
> 
> From your log, I see you have enabled PSRAM. That should provide more memory for holding images.
> 
> You can get some info on your heap memory usage. See the [Heap Memory Allocation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32p4/api-reference/system/mem_alloc.html) documentation for APIs to display the amount of memory being allocated.
> 
> ESP-Hosted itself uses some DMA-Capable heap memory for buffer queues to transfer data to and from the ESP32-C6 co-processor. You can adjust the size of the queues via `idf.py menuconfig` -> `Component config` -> `ESP-Hosted config` -> `Hosted SDIO Configuration` -> `Host SDIO Tx/Rx queue size`.
> 
> Reducing the size of the queues may reduce your Wi-Fi throughput.

Yeah I'm keeping the RAM usage under control, but it is not a problem related to P4 side. I have plenty of it available. 
Also because the problem show up only when I link to the wifi AP and navigate to the image streaming page. 

I changed the suggested settings above (from 20 to 16) and it seems to work flawlessy now. 
Thanks for the suggestions

simoberny, 2025-04-23:
The heap memory error still happending sporadically (~10 minutes interval with stream open on a webpage)
Now the assertion error is: sdio_push_data_to_queue sdio_drv.c:701

SohKamYung-Espressif, 2025-04-23:
Can you send your `sdkconfig` so we can take a look at your settings?

Meanwhile it might be useful to see how much memory is available before the assert happens. You can do this by using [Heap Memory Debugging calls](https://docs.espressif.com/projects/esp-idf/en/latest/esp32p4/api-reference/system/heap_debug.html).

For example,

```c
#define MEM_DUMP(s) \
    printf("%s free:%lu min-free:%lu dma-free:%u default-free:%u 8bit-free:%u SPIRAM-free:%u\n", s, \
                  esp_get_free_heap_size(), esp_get_minimum_free_heap_size(), \
                  heap_caps_get_largest_free_block(MALLOC_CAP_DMA),\
                  heap_caps_get_largest_free_block(MALLOC_CAP_DEFAULT),\
                  heap_caps_get_largest_free_block(MALLOC_CAP_8BIT),\
                  heap_caps_get_largest_free_block(MALLOC_CAP_SPIRAM));
```

Then do a `MEM_DUMP()` at places in your code, for example, before and after allocating memory for images.

SohKamYung-Espressif, 2025-04-24:
As a test, is it possible to build a complete HTTP response and send with `httpd_resp_send()` instead, and see what happens to memory?

You mentioned that your code works on an ESP32-S3. As a check, you could monitor memory usage while sending data with `httpd_resp_send_chunk()` and see the behaviour.

simoberny, 2025-04-24:
Ok I tested it using httpd_resp_send() by creating a dedicated /image endpoint.
Observation: 
- DMA free memory starts at around 47104 bytes
- First request on /image 
     - DMA free memory drop to 28672 bytes. Initial buffers and queue by esp-hosted? 
- Subsequent requests every ~5 seconds:
    - DMA seems to set around 22528, with some fluctuations
- Increased request rate (~2 per second):
    - The DMA drop down for some time and then get back to the previous value. Until eventually getting to **Not enough heap memory**

SohKamYung-Espressif, 2025-04-24:
> For reference, I'm sending buffer of around 100KB max.

This is the max size of each HTTP chunk?

What is the size of the image?

simoberny, 2025-04-24:
> > For reference, I'm sending buffer of around 100KB max.
> 
> This is the max size of each HTTP chunk?
> 
> What is the size of the image?

I'm sending the image as a single chunk, keeping the socket open during the transfer of the frames. But before every frame I have to send the preamble. 
The typical size of each JPEG image is around 100 KB.
However, based on my recent observations, the issue doesn’t seem to be related to httpd_resp_send_chunk() specifically — it also occurs when using httpd_resp_send() directly.

To give a bit more context I put here some of the code. 
I'm reusing code originally taken from an old esp-who example for the stream endpoint. 

### httpd_resp_send
```c++
    esp_err_t image_handler(httpd_req_t *req)
    {
        Config &cfg = Config::getInstance();
        dl::image::jpeg_img_t jpeg_img;

        if (hw_encode_jpeg(cfg.img_, jpeg_img, JPEG_DOWN_SAMPLING_YUV420, 90) != ESP_OK)
        {
            return ESP_FAIL;
        } 

        httpd_resp_set_type(req, "image/jpeg");
        httpd_resp_send(req, (const char *)jpeg_img.data, jpeg_img.data_size);

        heap_caps_free(jpeg_img.data);
        jpeg_img.data = nullptr;

        return ESP_OK;
    }
```

### httpd_resp_send_chunk
```c++

    esp_err_t stream_handler(httpd_req_t *req)
    {
        Config &cfg = Config::getInstance();

        esp_err_t res = ESP_FAIL;
        dl::image::jpeg_img_t jpeg_img;
        char part_buf[128];

        // Set content type to mixed
        ESP_ERROR_CHECK(httpd_resp_set_type(req, _STREAM_CONTENT_TYPE));

        httpd_resp_set_hdr(req, "Access-Control-Allow-Origin", "*");
        httpd_resp_set_hdr(req, "X-Framerate", "5");

        long long int last_send = 0;

        while (true)
        {
            struct timespec ts = {0};
            clock_gettime(CLOCK_MONOTONIC, &ts);

            if (cfg.img_counter_ != last_send)
            {
                MEM_DUMP("Stream loop");
                
                if (hw_encode_jpeg(cfg.img_, jpeg_img, JPEG_DOWN_SAMPLING_YUV420, 80) != ESP_OK)
                {
                    ESP_LOGE(TAG, "Failed to encode JPEG image");
                    continue;
                }

                res = httpd_resp_send_chunk(req, _STREAM_BOUNDARY, strlen(_STREAM_BOUNDARY));
                if (res == ESP_OK)
                {
                    size_t hlen = snprintf(part_buf, sizeof(part_buf), _STREAM_PART, jpeg_img.data_size, ts.tv_sec, ts.tv_nsec);
                    res = httpd_resp_send_chunk(req, (const char *) part_buf, hlen);
                }

                if (res == ESP_OK)
                {
                    ESP_LOGI(TAG, "Sending %d bytes", jpeg_img.data_size);
                    res = httpd_resp_send_chunk(req, (const char *)jpeg_img.data, jpeg_img.data_size);
                }

                heap_caps_free(jpeg_img.data);
                jpeg_img.data = nullptr;

                if (res != ESP_OK)
                {
                    ESP_LOGE(TAG, "Break stream handler");
                    break;
                }

                last_send = cfg.img_counter_;
            }
        }

        return res;
}
```

SohKamYung-Espressif, 2025-04-24:
An observation: for `httpd_resp_send_chunk()`, you are suppose to send `httpd_resp_send_chunk(req, NULL, 0)` at the end of the transfer as noted in the documentation https://docs.espressif.com/projects/esp-idf/en/v5.4.1/esp32p4/api-reference/protocols/esp_http_server.html#_CPPv421httpd_resp_send_chunkP11httpd_req_tPKc7ssize_t :
 
> Note
>
>    This API is supposed to be called only from the context of a URI handler where httpd_req_t* request pointer is valid.
>
>    When you are finished sending all your chunks, you must call this function with buf_len as 0.
>
>    Once this API is called, all request headers are purged, so request headers need be copied into separate buffers if they are required later.

I looked at your `sdkconfig` and I saw some PSRAM configuration changes:

`CONFIG_SPIRAM_USE_CAPS_ALLOC=y` means memory in PSRAM is allocated using ` heap_caps_malloc(..., MALLOC_CAP_SPIRAM)`.

If you are using `malloc()` instead of `heap_caps_malloc()`, you can change that setting to `CONFIG_SPIRAM_USE_MALLOC=y` instead.

I am currently using the ESP-IDF `examples/protocols/http_server/file_serving` to try to test http chunk transfer.

simoberny, 2025-04-24:
@SohKamYung-Espressif 
1. I remember reading this, but in this case it is a stream so there is no potential end to the chunks. And when I stop the page is closed, the socket is closed as well. So `httpd_resp_send_chunk(req, NULL, 0)` will return error in send. I had already tried it. In fact, this call was also omitted in the original example, likely for the same reason.
2. I'm using CONFIG_SPIRAM_USE_CAPS_ALLOC=y for performance reasons. With this setting enabled, I observed a 50% faster inference on yolov11 network (700ms vs 1100ms).

SohKamYung-Espressif, 2025-04-25:
> I remember reading this, but in this case it is a stream so there is no potential end to the chunks

Understood.

> I'm using CONFIG_SPIRAM_USE_CAPS_ALLOC=y for performance reason

If you use `CONFIG_SPIRAM_USE_MALLOC=y` do you encounter the same DMA heap memory issues?

SohKamYung-Espressif, 2025-04-28:
According to https://docs.espressif.com/projects/esp-idf/en/latest/esp32p4/api-reference/kconfig-reference.html#config-spiram-use 

> - Make RAM allocatable using heap_caps_malloc(..., MALLOC_CAP_SPIRAM) (CONFIG_SPIRAM_USE_CAPS_ALLOC)
> - Make RAM allocatable using malloc() as well (CONFIG_SPIRAM_USE_MALLOC)

If you use `CONFIG_SPIRAM_USE_CAPS_ALLOC` then you have to use `heap_caps_malloc(..., MALLOC_CAP_SPIRAM)` to allocate memory from SPIRAM. Standard `malloc()` will allocate from internal memory.

If you use `CONFIG_SPIRAM_USE_MALLOC`, then `malloc()` may allocate from SPIRAM (there are a few other SPIRAM config options to decide whether memory is allocated from internal memory or SPIRAM).

> May I ask if this is expected behavior using CONFIG_SPIRAM_USE_CAPS_ALLOC? Because it makes the difference in terms of performance with neural network inference.

From https://github.com/espressif/esp-hosted/issues/597#issuecomment-2827420835
> I'm using CONFIG_SPIRAM_USE_CAPS_ALLOC=y for performance reasons. With this setting enabled, I observed a 50% faster inference on yolov11 network (700ms vs 1100ms).

Is your code using `heap_caps_malloc()` or only standard `malloc()`? If only `malloc()`, then memory is only allocated from internal memory, which is faster than SPIRAM. But internal memory is more limited in size than SPIRAM, causing the DMA heap issues.

You can also submit the "performance with neural network inference" question to the ESP-IDF github issue https://github.com/espressif/esp-idf/issues , which may be able to help you optimize your memory usage to get the best performance while minimizing heap issues.

simoberny, 2025-04-29:
> then memory is only allocated from internal memory, which is faster than SPIRAM

It is plausible that the difference in performance depends on this. I'm actually not using either one of malloc/heap_caps_malloc.
So everything depends on the implementations inside esp-dl. By a quick look it seems that they are using `heap_caps_malloc(..., MALLOC_CAP_DEFAULT)` almost everywhere, so I suppose the same as doing malloc, priotizing internal memory. 

I'll move the request to the esp-dl issue, to investigate more on the use of the heap.
Thank you very much to help me investigate and found the root cause of what initially I tought might be a problem with the heap management by esp-hosted. 

Best regards

SohKamYung-Espressif, 2025-04-29:
You are welcome.

Can you mark this issue as closed? You can, of course, reopen this or file a new issue if you discover another issue with esp-hosted.

johnboiles, 2025-05-14:
I'm getting the exact same `esp_dma_capable_malloc(181): Not enough heap memory` error when doing streaming downloads. I've tried the suggestions of decreasing `Host SDIO Tx/Rx queue size` and setting `CONFIG_SPIRAM_USE_MALLOC`. Like @simoberny mentioned, it's not always consistent.

In case it's a clue I've gotten a log like `H_SDIO_DRV: task still writing Rx data to queue!` and sometimes I get a `Stack protection fault`. Absolutely possible I'm doing something wrong in my code but also like @simoberny, the same code works very reliably on ESP32-S3.

Should I open a new issue for this, or continue the discussion here?

johnboiles, 2025-05-14:
`esp_hosted` is allocating these massive ~30KB buffers in `sdio_rx_get_buffer` when in streaming mode and tanks the DMA RAM very quickly:

```
sdio_rx_get_buffer free:21616616 min-free:21604792 lfb-dma:49152 lfb-def:20971520 lfb-8bit:20971520
I (50345) H_SDIO_DRV: buf 1 size: 4608
D (50439) Sol: Downloaded 1.5% (2.9 KB/s, 4096 bytes)
D (50454) Sol: Wrote 1.5% (2.8 KB/s)
D (50454) Sol: Progress callback: 4, 0%
D (50455) Sol: Dispatching Content Sync UI update for state 4
I (50455) H_SDIO_DRV: sdio_rx_get_buffer 16896
I (50455) MEM_TAG: mempool_alloc 1600
sdio_rx_get_buffer free:21614720 min-free:21604288 lfb-dma:45056 lfb-def:20971520 lfb-8bit:20971520
I (50463) MEM_TAG: mempool_alloc 1600
I (50472) H_SDIO_DRV: buf 1 size: 16896
D (50481) Sol: SHOW_WATERMARK(D:UI0) (min remaining stack): 4504 ./src/Misc/Dispatch.cpp:157
I (50487) MEM_TAG: mempool_alloc 1600
I (50489) MEM_TAG: mempool_alloc 1600
I (50664) H_SDIO_DRV: sdio_rx_get_buffer 15360
I (50665) MEM_TAG: mempool_alloc 1600
sdio_rx_get_buffer free:21585228 min-free:21568556 lfb-dma:31744 lfb-def:20971520 lfb-8bit:20971520
I (50667) MEM_TAG: mempool_alloc 1600
I (50670) H_SDIO_DRV: buf 0 size: 15360
D (50679) Sol: SHOW_HEAP() INTERNAL:94k/424k SPIRAM:21520k/28435k DMA:56k/386k 8BIT:21616k/28860k ./src/main.cpp:202
I (50681) MEM_TAG: mempool_alloc 1600
I (50682) MEM_TAG: mempool_alloc 1600
D (50687) Sol: SHOW_WATERMARK(Polling) (min remaining stack): 2460 ./src/main.cpp:203
I (50864) H_SDIO_DRV: sdio_rx_get_buffer 28672
sdio_rx_get_buffer free:21618592 min-free:21568556 lfb-dma:21504 lfb-def:20971520 lfb-8bit:20971520
E (50866) dma_utils: esp_dma_capable_malloc(181): Not enough heap memory
```

Could it be that `esp_hosted` doesn't have any RX throttling to rate limit incoming bytes in [sdio_read_task](https://github.com/espressif/esp-hosted-mcu/blob/main/host/drivers/transport/sdio/sdio_drv.c#L752) and will call [sdio_rx_get_buffer](https://github.com/espressif/esp-hosted-mcu/blob/main/host/drivers/transport/sdio/sdio_drv.c#L673), until it runs out of memory and crashes?

johnboiles, 2025-05-14:
Disabling streaming mode and using `ESP_HOSTED_SDIO_OPTIMIZATION_RX_NONE` instead, It works for a but but then I get this error in the logs:
```
E (47817) H_SDIO_DRV: sdio_get_len_from_slave: Len from slave[2428] exceeds max [1536]
```

SohKamYung-Espressif, 2025-05-15:
Hi, @johnboiles 

To disable SDIO_STREAMING, you have to update the configuration for the C6 co-processor (slave) also. Both the Host and co-processor must enable or disable SDIO STREAMING at the same time.

In the `slave` directory, run `idf.py menuconfig` and go to `Example Configuration` ---> `SDIO configuration` ---> toogle off `Enable SDIO Streaming Mode`.

Now build the code and flash the C6 co-processor.

johnboiles, 2025-05-15:
That's good to know, but I'd rather make streaming work if possible!

SohKamYung-Espressif, 2025-05-15:
FYI, here is the ESP documentation on SDIO streaming https://docs.espressif.com/projects/esp-idf/en/latest/esp32c6/api-reference/peripherals/sdio_slave.html#sending-fifo : 

> Stream Mode: When a buffer is loaded into the driver, the buffer length is included into the packet length requested by host in the incoming communications. This is irrespective of whether previous packets have been sent or not. In other words, the length of the newly loaded buffer is included into the length of the packet requested by the host, even if there are previously unsent packets. This enables the host to receive data from several buffers in a single transfer.

The C6 co-processor sets the maximum transfer size in streaming mode. You can reduce the maximum size of the single transfer in streaming mode by reducing the value of `SDIO_SLAVE_QUEUE_SIZE` in `slave/main/sdio_slave_api.c`.

If your application does not require a SDIO high throughput, you may want to consider using SDIO packet mode transfer to minimize buffer sizes on the host.

johnboiles, 2025-05-15:
Very helpful thank you.

To make sure I understand: it's not surprising that the SDIO buffers could get huge (60k of buffer) if the host MCU is unable to keep up with reading them. Is that correct? I'm doing BLE central, peripheral, writing to a SD card, and driving a LCD at the same time as the download so it's definitely possible the P4 is behind in reading the queues.

I should be able to test that in my minimal repro case by calling `esp_http_client_read' more slowly right?

SohKamYung-Espressif, 2025-05-15:
> To make sure I understand: it's not surprising that the SDIO buffers could get huge (60k of buffer) if the host MCU is unable to keep up with reading them. Is that correct?

Yes, that is correct. The SDIO slave keeps accumulating data until the host transfers all the data in one read (in streaming mode).

> I should be able to test that in my minimal repro case by calling `esp_http_client_read' more slowly right?

Yes, I think that is one way to test it.

mantriyogesh, 2025-05-15:
> The SDIO slave keeps accumulating data until the host transfers all the data in one read

We would also need to see how can we cap on the max size in streaming mode.

simoberny, 2025-05-16:
> In case it's a clue I've gotten a log like `H_SDIO_DRV: task still writing Rx data to queue!` and sometimes I get a `Stack protection fault`. Absolutely possible I'm doing something wrong in my code [...]

I confirm that I also got `H_SDIO_DRV: task still writing Rx data to queue!` and `Stack protection fault` occasionally. 

> Disabling streaming mode and using ESP_HOSTED_SDIO_OPTIMIZATION_RX_NONE instead

First time seeing this flag. I set both on C6 and P4, and it seems stable right now. In my use case, doesn’t seem to be any drawback to not using that optimization. 

> Should I open a new issue for this, or continue the discussion here?

@SohKamYung-Espressif Should I re-open the issue? At least until we figure out if there is a way to limit the buffers getting too huge in streaming mode. Thanks

johnboiles, 2025-05-16:
I've found I can make streaming much mode reliable by setting `CONFIG_LWIP_TCP_WND_DEFAULT` back to the default of 5760. Like @simoberny, I had it set to `65534` as a leftover wifi optimization for the ESP32-S3. I still see my available DMA RAM go down in the course of a download but not as quickly or unevenly. I'm unsure if this improves the issue simply by making downloads slower.

Attached is a minimal repro case that does some high priority no-op work. With `CONFIG_LWIP_TCP_WND_DEFAULT=65534`, I can consistently-ish repro the problem when near my AP (to get good wifi throughput). Sometimes I get SDIO errors after a larger-than-usual buffer:

```
I (38509) H_SDIO_DRV: sdio_rx_get_buffer 25812
download_task free:33847120 min-free:33797844 dma-free:196608 default-free:33030144 8bit-free:33030144 SPIRAM-free:33030144
download_task free:33847100 min-free:33797844 dma-free:196608 default-free:33030144 8bit-free:33030144 SPIRAM-free:33030144
download_task free:33847080 min-free:33797844 dma-free:196608 default-free:33030144 8bit-free:33030144 SPIRAM-free:33030144
download_task free:33847060 min-free:33797844 dma-free:196608 default-free:33030144 8bit-free:33030144 SPIRAM-free:33030144
download_task free:33864832 min-free:33797844 dma-free:196608 default-free:33030144 8bit-free:33030144 SPIRAM-free:33030144
download_task free:33864812 min-free:33797844 dma-free:196608 default-free:33030144 8bit-free:33030144 SPIRAM-free:33030144
download_task free:33864788 min-free:33797844 dma-free:196608 default-free:33030144 8bit-free:33030144 SPIRAM-free:33030144
download_task free:33864768 min-free:33797844 dma-free:196608 default-free:33030144 8bit-free:33030144 SPIRAM-free:33030144
E (39039) sdmmc_io: sdmmc_io_rw_extended: sdmmc_send_cmd returned 0xffffffff
E (39039) H_SDIO_DRV: sdio_read_task: Failed to read data - -1 25812 25812
E (39039) sdmmc_req: handle_idle_state_events unhandled: 00000000 00000002
E (39049) sdmmc_req: handle_idle_state_events unhandled: 00000000 00000002
E (39049) sdmmc_req: handle_idle_state_events unhandled: 00000000 00000002
E (39059) sdmmc_req: handle_idle_state_events unhandled: 00000008 00000002
I (39069) H_SDIO_DRV: sdio_rx_get_buffer 1434
```

And sometimes after that it requests absolutely massive buffers and crashes

```
download_task free:32663632 min-free:32631024 dma-free:237568 default-free:31981568 8bit-free:31981568 SPIRAM-free:31981568
I (27182) H_SDIO_DRV: sdio_rx_get_buffer 1434
I (27186) H_SDIO_DRV: sdio_rx_get_buffer 27246
download_task free:32679928 min-free:32631024 dma-free:237568 default-free:31981568 8bit-free:31981568 SPIRAM-free:31981568
download_task free:32679908 min-free:32631024 dma-free:237568 default-free:31981568 8bit-free:31981568 SPIRAM-free:31981568
download_task free:32679888 min-free:32631024 dma-free:237568 default-free:31981568 8bit-free:31981568 SPIRAM-free:31981568
download_task free:32679868 min-free:32631024 dma-free:237568 default-free:31981568 8bit-free:31981568 SPIRAM-free:31981568
E (27552) sdmmc_io: sdmmc_io_rw_extended: sdmmc_send_cmd returned 0xffffffff
E (27553) H_SDIO_DRV: sdio_read_task: Failed to read data - -1 27246 27246
E (27555) sdmmc_req: handle_idle_state_events unhandled: 00000000 00000002
E (27561) sdmmc_req: handle_idle_state_events unhandled: 00000000 00000002
E (27568) sdmmc_req: handle_idle_state_events unhandled: 00000008 00000000
I (27575) H_SDIO_DRV: sdio_rx_get_buffer 824560
E (27579) dma_utils: esp_dma_capable_malloc(181): Not enough heap memory
```

I suggest re-opening this issue. To me it seems like attempting to allocate more RAM than exists on the system is probably incorrect behavior and should be prevented.

[esp_hosted_crash_repro_v1.zip](https://github.com/user-attachments/files/20259681/esp_hosted_crash_repro_v1.zip)

mantriyogesh, 2025-05-18:
@SohKamYung-Espressif PTAL

johnboiles, 2025-05-21:
What if when `sdio_rx_get_buffer` can't malloc a buffer, and there are still other buffers being processed, it just waits a bit and then tries again? If there are other buffers being processed we know that we will soon get some RAM freed up. It should probably still crash if there are no buffers being processed and it can't malloc a buffer.

For my use case, I was able to improve my reliability by decreasing my L2 cache from 256KB to 128KB (I had `CONFIG_CACHE_L2_CACHE_256KB=y` set). I still see my free DMA ram go from 172KB down to as low as 89KB but it's able to catch back up before it crashes.

mantriyogesh, 2025-05-22:
Hello @johnboiles ,

Sorry to bother in middle of discussion.

We are busy this week for release related activities & please expect delays in resolution to this issue.
Also, May I request you to open new issue (new things tend to open up one after the other as dependency) and cross-reference new issue here if deemed fit. Plus this way issue ownership then would be clear.
 
@simoberny ,

We already had some discussion to address this issue to limit max heap to be used (as part of design instead of limiting through capping, as feasible)

For now, I suggest disabling the streaming mode both sides, by the time this is not fixed.

SohKamYung-Espressif, 2025-05-27:
Hello @johnboiles 

I ran your test code for several minutes and did not see any memory issues. I also tried `CONFIG_LWIP_TCP_WND_DEFAULT=65534` and it is running fine.

In https://github.com/espressif/esp-hosted/issues/597#issuecomment-2887789545 I see this:

```
E (39039) sdmmc_io: sdmmc_io_rw_extended: sdmmc_send_cmd returned 0xffffffff
E (39039) H_SDIO_DRV: sdio_read_task: Failed to read data - -1 25812 25812
```

```
E (27552) sdmmc_io: sdmmc_io_rw_extended: sdmmc_send_cmd returned 0xffffffff
E (27553) H_SDIO_DRV: sdio_read_task: Failed to read data - -1 27246 27246
```

These are SDIO transfer errors, which should not happen.

Which P4 + C6 board are you using? Is it the Espressif [ESP32-P4-Function-EV-Board](https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32p4/esp32-p4-function-ev-board/user_guide.html)? This is what I am using and I do not see those SDIO error messages.

SohKamYung-Espressif, 2025-05-27:
Hello @simoberny . Regarding https://github.com/espressif/esp-hosted/issues/597#issuecomment-2886692803

> I confirm that I also got `H_SDIO_DRV: task still writing Rx data to queue`!

The SDIO driver implements a simple double-buffering scheme to improve performance. There are two threads running: one thread (Thread 1) to transfer received SDIO data into a Rx buffer (done via HW DMA), and another thread (Thread 2) to break up the already received stream data in another Rx buffer into packets for processing by the upper (networking) layer. They switch Rx buffers when completed.

If the system is too busy and Thread 2 gets delayed, the `task still writing Rx data to queue` is displayed because Thread 1 cannot switch buffers as the other task hasn't finished processing its buffer.

Can you check your system task priorities? The default task priority for ESP-Hosted is `23` and it should be the highest in the system.

> In my use case, doesn’t seem to be any drawback to not using streaming mode optimization.

If throughput is less of a concern, then not using streaming mode is an option as it requires fewer system resources.

johnboiles, 2025-05-28:
@SohKamYung-Espressif that looks like a great patch!

simoberny, 2025-06-10:
> [@simoberny](https://github.com/simoberny) [@johnboiles](https://github.com/johnboiles) Checking on the status of this issue.
> 
> Does the patch in [#597 (comment)](https://github.com/espressif/esp-hosted/issues/597#issuecomment-2915841823) help stabilize memory usage on the P4? Or are there still remaining problems?

Sorry, I've been on vacation for two weeks. I just tested the patch and changed the buffer size.
For my use cases, it seems to get stable with queue size of 5. No issue found after some hours of testing. 
I keep streaming mode enabled to do some more test. 
In production I will eventually switch to normal mode to avoid any issue though. 

Thank you

mantriyogesh, 2025-06-12:
Can you please cross-check with latest registry component in main/idf_component.yml if it works fine for you? If so we can close the issue.

You can also use "*" here : https://github.com/espressif/esp-idf/blob/98cd765953dfe0e7bb1c5df8367e1b54bd966cce/examples/wifi/iperf/main/idf_component.yml#L15 so that, it would show,

```
  espressif/esp_hosted:
    version: "*"
    rules:
      - if: "target in [esp32p4, esp32h2]"
 ```

and then 

```
$ rm -rf managed_components/ dependencies.lock sdkconfig build/; idf.py --preview set-target <target_host>
# idf.py -p <host_port> build flash monitor
```

<target_host> is say `esp32p4`

simoberny, 2025-06-12:
@mantriyogesh  I confirm it's working with the latest version of the component. 
The only issue is that the build system could not find `esp_hosted_wifi_config.h`. I had to copy the file from freertos include folder to the api include folder. I don't know why actually

mantriyogesh, 2025-06-12:
yes, this is known issue, at [#58](https://github.com/espressif/esp-hosted-mcu/issues/58)

Let me fix this.

SohKamYung-Espressif, 2025-06-13:
> I'm already including esp_hosted.h (that include esp_hosted_api.h)

I see. Is there a reason you are including `esp_hosted.h`?

SohKamYung-Espressif, 2025-06-13:
No problem.

We're working on a fix for this. In general, you shouldn't have to include any esp-hosted headers if you don't need to call any esp_hosted apis directly.

mantriyogesh, 2025-06-13:
Fixed in 
[image]