espressif/esp-hosted #598: Combining S3+C5, or possibly P4+C5

savenlid, 2025-04-28:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

ESP32-S3 + ESP32-C5, a more complete descripton for this user case

### Use cases

We started with the S3 but today we need 5Ghz wifi ( market requierments )
The quickest solution is to hook up the C5 and let it do the wifi ( disable S3 radio )
Could you make some mentioning of this specific combo considering its 2 pcs esp chips involved.
We need the USB of the S3 and the 5Ghz wifi of the C4
Another option would be P4+C5
Or another competitor chip for example STM32H743 + C5
But the low cost of the esp keep all of us trying to use the esp32 chips to solve everything.

### Alternatives

P4+C5
STM32+C5

SohKamYung-Espressif, 2025-04-29:
Hi, @savenlid ,

Please take a look at ESP-Hosted-MCU https://github.com/espressif/esp-hosted-mcu/ . This is a version of ESP-Hosted that should cater to your requirements.

For the S3 + C5 combination, see https://github.com/espressif/esp-hosted-mcu/blob/main/docs/troubleshooting.md#1-esp-host-to-evaluate-already-has-native-wi-fi , which contains instructions on how to disable the Native Wi-Fi on the S3 so it will use ESP-Hosted to communicate with the C5 co-processor.

For P4 as a host, you can check https://github.com/espressif/esp-hosted-mcu/blob/main/docs/esp32_p4_function_ev_board.md , which contains instructions for the P4 + C6 combination, but can be changed to the C5.

A question: which transport do you plan to use? SPI or SDIO?

savenlid, 2025-04-29:
Thanks for helping me.

I started reading the links you gave.

Regarding SDO versus SPI  i am agnostic.
But when i tried the NG case with a C5 and RPI5, only SPI was supported.
So I assumed only SPI is an option with the C5.

savenlid, 2025-04-29:
Hi

Is this expected ?
I guess I follow what is written on web-page ?

[image]

SohKamYung-Espressif, 2025-04-30:
> Is this expected ?
> I guess I follow what is written on web-page ?

Yes, please use the GPIOs listed on the web page.

The menuconfig values are for a C5 test board and should be updated.

Thank you for bringing it to my attention.

mantriyogesh, 2025-04-30:
> But when i tried the NG case with a C5 and RPI5, only SPI was supported

This is correct. ESP32-C5 **eco1** can work with SPI.
From ESP32-C5 **eco2**, you can expect SDIO, maybe May end. Once IDF supports C5 SDIO, we can easily support on ESP-Hosted.

savenlid, 2025-04-30:
Can I assume this to be the same port ?

[image]

I want to use port spi-3 for my DM9051 eth-mac, currently I have that one on spi-2, need to move to spi3.
I hope that will work.

The slave/co-processor dont drive the clock does it ?
Did not find this in menuconfig.

[image]

SohKamYung-Espressif, 2025-04-30:
> Can I assume this to be the same port ?

Yes. The default config is to use SPI2. No need to change "SPI controller to use".

> The slave/co-processor dont drive the clock does it ?

No. The SPI clock is driven by the SPI master.

"Set the SPI clock frequency" refers to setting the SPI clock frequency on the SPI Master.

Make sure that both the SPI master and slave are using the same SPI Mode (default is Mode 3).

savenlid, 2025-04-30:
After flashing the C5 as a co-processor it keeps repeating the last lines of my text below.
"mem_dump free:108996 min-free:108920 lfb-dma:90112 lfb-def:90112 lfb-8bit:90112"
Is it ok ?
FYI: Nothing is connected to any pins yet, maybe its upset about that ?
Or maybe its normal status info about the memory.

spressif\frameworks\esp-idf-master-3\tools\idf.py'"...
--- Warning: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM5 instead...
--- esp-idf-monitor 1.6.2 on \\.\COM5 115200
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ESP-ROM:esp32c5-eco1-20240726
Build:Jul 26 2024
rst:0x1 (POWERON),boot:0x18 (SPI_FAST_FLASH_BOOT)
SPI mode:DIO, clock div:1
load:0x40855ab0,len:0x1708
load:0x4084c7a0,len:0xd90
--- 0x4084c7a0: esp_bootloader_get_description at C:/Espressif/frameworks/esp-idf-master-3/components/esp_bootloader_format/esp_bootloader_desc.c:39 (discriminator 1)

load:0x4084e9a0,len:0x3680
--- 0x4084e9a0: esp_flash_encryption_enabled at C:/Espressif/frameworks/esp-idf-master-3/components/bootloader_support/src/flash_encrypt.c:86

entry 0x4084c7aa
--- 0x4084c7aa: call_start_cpu0 at C:/Espressif/frameworks/esp-idf-master-3/components/bootloader/subproject/main/bootloader_start.c:25

I (23) boot: ESP-IDF v5.5-dev-698-g030c9957db 2nd stage bootloader
I (23) boot: compile time Apr 30 2025 12:44:00
I (24) boot: chip revision: v0.1
I (25) boot: efuse block revision: v0.1
I (28) boot.esp32c5: SPI Speed      : 80MHz
I (32) boot.esp32c5: SPI Mode       : DIO
I (36) boot.esp32c5: SPI Flash Size : 4MB
I (40) boot: Enabling RNG early entropy source...
I (44) boot: Partition Table:
I (47) boot: ## Label            Usage          Type ST Offset   Length
I (53) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (60) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (66) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  3 ota_0            OTA app          00 10 00010000 00180000
I (79) boot:  4 ota_1            OTA app          00 11 00190000 00180000
I (86) boot: End of partition table
I (89) esp_image: segment 0: paddr=00010020 vaddr=420c0020 size=2ad0ch (175372) map
I (127) esp_image: segment 1: paddr=0003ad34 vaddr=40800000 size=052e4h ( 21220) load
I (131) esp_image: segment 2: paddr=00040020 vaddr=42000020 size=ba104h (762116) map
I (260) esp_image: segment 3: paddr=000fa12c vaddr=408052e4 size=13870h ( 79984) load
I (276) esp_image: segment 4: paddr=0010d9a4 vaddr=40818b80 size=03f30h ( 16176) load
I (284) boot: Loaded app from partition at offset 0x10000
I (284) boot: Disabling RNG early entropy source...
I (294) cpu_start: Unicore app
W (302) clk: esp_perip_clk_init() has not been implemented yet
I (308) cpu_start: Pro cpu start user code
I (308) cpu_start: cpu freq: 240000000 Hz
I (308) app_init: Application information:
I (308) app_init: Project name:     network_adapter
I (313) app_init: App version:      v5.5-dev-698-g030c9957db
I (318) app_init: Compile time:     Apr 30 2025 12:43:05
I (323) app_init: ELF file SHA256:  56d72526d...
I (327) app_init: ESP-IDF:          v5.5-dev-698-g030c9957db
I (333) efuse_init: Min chip rev:     v0.0
I (337) efuse_init: Max chip rev:     v0.99
I (341) efuse_init: Chip rev:         v0.1
I (344) heap_init: Initializing. RAM available for dynamic allocation:
I (351) heap_init: At 40822D00 len 00039CA0 (231 KiB): RAM
I (356) heap_init: At 4085C9A0 len 00002B60 (10 KiB): RAM
I (361) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM
I (367) spi_flash: detected chip: generic
I (370) spi_flash: flash io: dio
W (373) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (385) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (392) sleep_gpio: Enable automatic switching of GPIO sleep configuration
[... 15 lines cut ...]
I (472) h_bt: - BT/BLE (extended)
I (475) fg_mcu_slave: extended capabilities: 0x0
I (484) h_bt: ESP Bluetooth MAC addr: 60:55:f9:fb:cf:96
I (485) BLE_INIT: Using main XTAL as clock source
I (491) BLE_INIT: ble controller commit:[4f138b0]
W (493) BLE_INIT: BLE modem sleep is enabled
I (497) BLE_INIT: Bluetooth MAC: 60:55:f9:fb:cf:96
I (502) phy_init: phy_version 102,3b01d78a,Nov 27 2024,17:24:11
I (1405) phy: libbtbb version: 351e560, Nov 27 2024, 17:24:22
I (1406) SPI_DRIVER: Using SPI interface
I (1406) gpio: GPIO[3]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1411) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1420) SPI_DRIVER: SPI Ctrl:1 mode: 3, Freq:ConfigAtHost
GPIOs: MOSI: 7, MISO: 2, CS: 10, CLK: 6 HS: 3 DR: 4

I (1430) SPI_DRIVER: Hosted SPI queue size: Tx:20 Rx:20
I (1435) gpio: GPIO[10]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1443) gpio: GPIO[10]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1453) SPI_DRIVER: Slave chip Id[12]
I (1455) fg_mcu_slave: Initial set up done
I (1459) slave_ctrl: event ESPInit
mem_dump free:108996 min-free:108920 lfb-dma:90112 lfb-def:90112 lfb-8bit:90112
mem_dump free:108996 min-free:108920 lfb-dma:90112 lfb-def:90112 lfb-8bit:90112
mem_dump free:108996 min-free:108920 lfb-dma:90112 lfb-def:90112 lfb-8bit:90112
mem_dump free:108996 min-free:108920 lfb-dma:90112 lfb-def:90112 lfb-8bit:90112

SohKamYung-Espressif, 2025-05-02:
> I suspect the esp32-s3 is not an option for esp-hosted

Yes, you are correct. The default host configuration for ESP-Hosted is for ESP32-P4 and ESP32-H2.

Add `esp32s3` as a target for `esp_wifi_remote` and `esp_hosted`.

SohKamYung-Espressif, 2025-05-02:
> After flashing the C5 as a co-processor it keeps repeating the last lines of my text below.
> "mem_dump free:108996 min-free:108920 lfb-dma:90112 lfb-def:90112 lfb-8bit:90112"
> Is it ok ?

Yes, this is fine. It is periodically printed by the slave `app_main.c` https://github.com/espressif/esp-hosted-mcu/blob/e6489fbbfe3b35e837b34e97cc34912c59e210a6/slave/main/app_main.c#L941

mantriyogesh, 2025-05-02:
> https://github.com/espressif/esp-hosted/issues/598#issuecomment-2841723920

The P4 and H2, by default, do not have Wi-Fi with them. that is the reason ESP-Hosted is defaulted for them.

ESP32-S3 by default has Wi-Fi. Following link includes a table and legend below, shows which all chipsets are supported:
https://github.com/espressif/esp-hosted-mcu?tab=readme-ov-file#6-decide-the-communication-bus-in-between-host-and-slave

To evaluate ESP32-S3 as ESP-Hosted Slave, you might need to disable default Wi-Fi using https://github.com/espressif/esp-hosted-mcu/blob/main/docs/troubleshooting.md#1-esp-host-to-evaluate-already-has-native-wi-fi

savenlid, 2025-05-02:
My instinct made me turn the reset as active low.
It did not work, it worked better with active high.
But its not active high ?, its asserted low ?

[image]

This make you become unsecure about the hand-shake/ and data-ready polarities also.
It seems for the S3 as host only SPI2 is avaliable ( not spi3 ) ?

E (1996) transport: Identified slave [esp32c5] != Expected [esp32]
                run 'idf.py menuconfig' at host to reselect the slave?
                Aborting..

c5 is not an option.

[image]

savenlid, 2025-05-03:
I managed to get it working in iper.
Somethings are confusing when reset is described as active high, you get very insecure what else can be strange.
However after manually adding esp32s3 as a host in en yml file I got it working.
I never tested the iperf functionallity tough.
I wanted a better example where I acutally could run trafffic in the s3 via the c5.
I thought i do the example/protocols/http_server simple.
I made sure I had a screaming fresh copy from github of espressif.
git clone --recursive https://github.com/espressif/esp-idf.git
I installed all chips.
I updated it.
everything was fine.
I did add the "modules"
idf.py add-dependency "espressif/esp_wifi_remote"
idf.py add-dependency "espressif/esp_hosted"
I had to manually add esp32c5 as an option in slave yml file to be able to choose it in menuconfig for wifi-remote.
(this always feels very nervous, as I do something forbidden or at least its sure to to fail)

setting the target gives a less than reassuring output:
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: esp_stubs
NOTICE: Skipping optional dependency: esp_stubs
NOTICE: Skipping optional dependency: esp_stubs
NOTICE: Using component placed at /home/pi/esp250305/esp-idf/examples/common_components/protocol_examples_common for dependency "protocol_examples_common", specified in /home/pi/esp250305/esp-idf/examples/protocols/http_server/simple/main/idf_component.yml
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: esp_stubs

However, equal to the iperf example it seems to work ok.
But both examples are unhappy with GPIO4 as IRQ, it seems to have been spoken for by someone else

I (389) os_wrapper_esp: GPIO [17] configuring as Interrupt
I (389) spi: No element in any queue found
I (399) os_wrapper_esp: GPIO [4] configuring as Interrupt
E (399) gpio: gpio_install_isr_service(526): GPIO isr service already installed

The ISR I can suspect to be only 1 single ISR for all GPIO, so maybe its what it says.
A bit unclear.

Success
[image]

now a question 2 you gentlemens.

My plan is to convert out current system S3 to a S3+C5 (for the sake of 5Ghz wifi capability)
I also have ETH (DM9051) works today.
RNDIS over USB (works in your examples )
I2C-Master ( works today)

Will I succeed or am I a dead man walking ?

Will I get the DM9051 in on the SPI3 over the gpio-mux, or will it be possible to run both the C5 and DM9051 on SPI2-port using 2 chip-selects ?

Thank you.

SohKamYung-Espressif, 2025-05-05:
> Somethings are confusing when reset is described as active high, you get very insecure what else can be strange.

You are referring to this ESP-Hosted SPI setting in `menuconfig`: `Reset GPIO Config (RESET: Active High)`

You can check `Symbol info` ('?' key) for that setting which explains what it means:

```
Help:

"If Active High, High->Low->High will trigger reset (Low will trigger reset)                         
If Active Low, Low->High->Low will trigger reset (High will trigger reset)"
```

> I had to manually add esp32c5 as an option in slave yml file to be able to choose it in menuconfig for wifi-remote.

I checked wifi-remote and you are right. esp32c5 is not currently a slave option (probably a temporary removal).

As a temporary measure, you can use an earlier version of wifi-remote that has the C5 as an option, like v0.8.5 https://components.espressif.com/components/espressif/esp_wifi_remote/versions/0.8.5?language=en

You can use it by running:

```
idf.py add-dependency "espressif/esp_wifi_remote^0.8.5" 
```

> NOTICE: Skipping optional dependency: espressif/esp_hosted

Just a check: you are seeing this while using the S3 as the ESP-Hosted host? If yes, please check that you have disabled wifi related settings as mentioned in the link https://github.com/espressif/esp-hosted-mcu/blob/main/docs/troubleshooting.md#1-esp-host-to-evaluate-already-has-native-wi-fi 

Can you provide the console output from your ESP-Hosted master (S3?) and slave (C5?) from boot-up onwards?

savenlid, 2025-05-05:
FYI #1: as I wrote, its working, but with failure to register an GPIO4 as IRQ.

Yes I did disable all wifi in the host (S3 ).
Once I forgot and got linking error.

FYI #2:  you don't find this text confusing ?
"**If Active High, High**->Low->High will trigger reset (**Low will trigger reset**)                         
If Active Low, Low->High->Low will trigger reset (High will trigger reset)"

Question #1
Why do the ISR fail ?
Complete boot below as requested:

pi@RPI16:~/esp250305/esp-idf/examples/protocols/http_server/simple $ idf.py moni
tor
Executing action: monitor
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32-S3
Running idf_monitor in directory /home/pi/esp250305/esp-idf/examples/protocols/h                                                                                                             ttp_server/simple
Executing "/home/pi/.espressif/python_env/idf5.5_py3.11_env/bin/python /home/pi/                                                                                                             esp250305/esp-idf/tools/idf_monitor.py -p /dev/ttyUSB0 -b 115200 --toolchain-pre                                                                                                             fix xtensa-esp32s3-elf- --target esp32s3 --revision 0 /home/pi/esp250305/esp-idf                                                                                                             /examples/protocols/http_server/simple/build/simple.elf /home/pi/esp250305/esp-i                                                                                                             df/examples/protocols/http_server/simple/build/bootloader/bootloader.elf -m '/ho                                                                                                             me/pi/.espressif/python_env/idf5.5_py3.11_env/bin/python' '/home/pi/esp250305/es                                                                                                             p-idf/tools/idf.py'"...
--- esp-idf-monitor 1.6.2 on /dev/ttyUSB0 115200
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x15d8
load:0x403c8700,len:0xd24
--- 0x403c8700: _stext at ??:?

load:0x403cb700,len:0x2f84
entry 0x403c8924
--- 0x403c8924: call_start_cpu0 at /home/pi/esp250305/esp-idf/components/bootloa                                                                                                             der/subproject/main/bootloader_start.c:25

I (24) boot: ESP-IDF v5.5-dev-3372-g38628f98b9-dirty 2nd stage bootloader
I (25) boot: compile time May  4 2025 02:09:55
I (25) boot: Multicore bootloader
I (27) boot: chip revision: v0.1
I (30) boot: efuse block revision: v1.2
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (37) boot.esp32s3: SPI Mode       : DIO
I (41) boot.esp32s3: SPI Flash Size : 2MB
I (45) boot: Enabling RNG early entropy source...
I (49) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (71) boot:  2 factory          factory app      00 00 00010000 00100000
I (78) boot: End of partition table
I (81) esp_image: segment 0: paddr=00010020 vaddr=3c060020 size=260a4h (155812)                                                                                                              map
I (116) esp_image: segment 1: paddr=000360cc vaddr=3fc95900 size=02d40h ( 11584)                                                                                                              load
I (119) esp_image: segment 2: paddr=00038e14 vaddr=40374000 size=07204h ( 29188)                                                                                                              load
I (126) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=57af4h (359156)                                                                                                              map
I (191) esp_image: segment 4: paddr=00097b1c vaddr=4037b204 size=0a690h ( 42640)                                                                                                              load
I (200) esp_image: segment 5: paddr=000a21b4 vaddr=600fe000 size=0001ch (    28)                                                                                                              load
I (207) boot: Loaded app from partition at offset 0x10000
I (208) boot: Disabling RNG early entropy source...
I (218) cpu_start: Multicore app
I (227) cpu_start: Pro cpu start user code
I (227) cpu_start: cpu freq: 160000000 Hz
I (227) app_init: Application information:
I (227) app_init: Project name:     simple
I (231) app_init: App version:      v5.5-dev-3372-g38628f98b9-dirty
I (237) app_init: Compile time:     May  4 2025 02:24:08
I (242) app_init: ELF file SHA256:  dbcf0aca0...
I (246) app_init: ESP-IDF:          v5.5-dev-3372-g38628f98b9-dirty
I (252) efuse_init: Min chip rev:     v0.0
I (256) efuse_init: Max chip rev:     v0.99
[... 50 lines cut ...]
I (2019) spi: Received INIT event
I (2889) example_connect: Connecting to NSP5G...
I (2929) H_API: esp_wifi_remote_connect
I (2949) example_connect: Waiting for IP(s)
I (12759) esp_wifi_remote: esp_wifi_internal_reg_rxcb: sta: 0x4202385c
[... cut ...]
I (17049) example: Starting server on port: '80'
I (17059) example: Registering URI handlers

Question #2: 
My plan is to convert our current system S3 -> S3+C5 (for the sake of 5Ghz wifi capability)
I also have ETH (DM9051) works today.
RNDIS over USB (works in your examples )
I2C-Master ( works today)

Will I succeed or am I a dead man walking ?

Will I get the DM9051 in on the SPI3 over the gpio-mux, or will it be possible to run both the C5 and DM9051 on SPI2-port using 2 chip-selects ?

Thank you

savenlid, 2025-05-07:
> > But when i tried the NG case with a C5 and RPI5, only SPI was supported
> 
> This is correct. ESP32-C5 **eco1** can work with SPI. From ESP32-C5 **eco2**, you can expect SDIO, maybe May end. Once IDF supports C5 SDIO, we can easily support on ESP-Hosted.

If C5 eco 2 works with SDIO, how ill the connections be made between S3 and C5 ?
I need to plan my HW ahead of time.

mantriyogesh, 2025-05-07:
Hello @savenlid ,

We have C5+P4 SDIO board, whose schematic we can share to you.
However, I would suggest you to wait for some more time, as I do not want you to trouble reworking if something changes in hardware.

Around week or two wait should be better.
If impatient, just mail me, I would copy you ${\color{yellow} \text{existing}}$ schematic.

savenlid, 2025-05-07:
@mantriyogesh 

thanks,

I wait a couple of weeks, we wont produce until we can get wroom modules with eco2

mantriyogesh, 2025-05-07:
sure thanks.

savenlid, 2025-05-07:
@mantriyogesh 

Hi again,

I was successfully using the S5 as wifi co-processor in the examples/protocols/http_server/simple.

But when I wanted to add my DM9051 spi ethernet only SPI 0,1,2 is available, not #3.
Do you know why ?
Is not spi0 the the flash it boots from, spi1 PSRAM ?
AFAIK i used SPI2 to connect the C5.
If connecting C5 over SDIO ( in the future ) will SPI2 be free again ?

[image]

savenlid, 2025-05-07:
this is confusing for me

I see some HSPI/FSPI definitions

#define FSPI_HOST   SPI2_HOST
#define HSPI_HOST   SPI3_HOST

but in esp-hosted you can only select both or none ?

[image]

mantriyogesh, 2025-05-07:
Important thing despite to know, even if SPI slave states say 60MHz or so in slave mode, We could get it reliable till 40MHz, depending upon the chipset. Obviously higher frequencies (>20MHz) would only make any meaning only if you have PCB. Please refrain from using higher frequencies using jumper cables for SPI, as it would just not work due to signal issues.

If you free up some peripheral, it would be usable . for example, you use sdio for hosted. still everything else (apart from that peripheral and GPIOs) from chip is usable (depends how you handle it) for something else.

savenlid, 2025-05-07:
thanks,

I also would have preferred 0,1,2,3 throughout all docs.

But for the core question, it seems the wifi-hosted is taking port 2 ?
And I only have port 0,1,2  to choose from for my spi-dm9051.
So where did port 3 go ?
I don't know what to do.
I assumed 0 = FLASH, PSRAM =1
Remains 2 and 3
Did the wifi hosted (HSPI/FSPI) take 2 or 3 ?

SohKamYung-Espressif, 2025-05-08:
Hi, @savenlid ,

Regarding https://github.com/espressif/esp-hosted/issues/598#issuecomment-2858008728

That Menuconfig for SPI is controlled by https://github.com/espressif/esp-idf/blob/master/examples/common_components/protocol_examples_common/Kconfig.projbuild#L253C1-L259C91 : 

```
        if EXAMPLE_USE_SPI_ETHERNET
            config EXAMPLE_ETH_SPI_HOST
                int "SPI Host Number"
                range 0 2
                default 1
                help
                    Set the SPI host used to communicate with the SPI Ethernet Controller.
```

Taking a look at https://github.com/espressif/esp-idf/blob/master/components/hal/include/hal/spi_types.h#L18C1-L28C21 : 

```c
/**
 * @brief General purpose SPI Host Controller ID.
 */
typedef enum {
    SPI1_HOST = 0,  ///< SPI1
    SPI2_HOST = 1,  ///< SPI2
#if SOC_SPI_PERIPH_NUM > 2
    SPI3_HOST = 2,  ///< SPI3
#endif
    SPI_HOST_MAX,   ///< invalid host value
} spi_host_device_t;
```

To use SPI3, select `2` as `SPI Host Number`.

ESP-Hosted is using `SPI2` which maps to the value of `1`.

savenlid, 2025-05-09:
I try to modify some examples to get hosted-wifi and spi-ethernet simultaneously.

The example project example/wifi/iperf  already have both components (esp_wifi_remote/esp_hosted) already added.
When I add hosted/wifi-remote to some other projects I often get below problem.

**idf.py add-dependency "espressif/esp_wifi_remote"
idf.py add-dependency "espressif/esp_hosted"
idf.py update-dependencies**

Any hints to me why this happens ?

only beginning  tail if link is  shown below.

**multiple definition of `wifi_init'**

Another question, why do some examples have a parallel components directory separated from the normal components as described in the docs.
I refer to some examples having
**example_eth_init**(&s_eth_handles, &s_eth_port_cnt);  //  number of available ports is put into your eth_port_cnt here
But in documentation this is non-existing, I get compile error for not existing.
Doc says:
https://components.espressif.com/components/espressif/ethernet_init
**ethernet_init_all**(&eth_handles, &eth_port_cnt);

Why this parallel component-dir with example_xxxx ?

In this doc:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/network/esp_eth.html
A 3rd method for a single eth interface init seems to exist.
esp_eth_config_t config = ETH_DEFAULT_CONFIG(mac, phy); // apply default driver configuration
esp_eth_handle_t eth_handle = NULL; // after the driver is installed, we will get the handle of the driver
esp_eth_driver_install(&config, &eth_handle); // install driver

The above is a bit confusing to me, is there a legacy reason ?

pi@RPI16:~/esps3host/esp-idf/examples/ethernet/basic $ idf.py build
Executing action: all (aliases: build)
Running ninja in directory /home/pi/esps3host/esp-idf/examples/ethernet/basic/build
Executing "ninja all"...
[1/1] cd /home/pi/esps3host/esp-idf/examples/ethernet/basic/build/bootloader/esp-idf/esptool_...loader 0x0 /home/pi/esps3host/esp-idf/examples/ethernet/basic/build/bootloader/bootloader.bi
Bootloader binary size 0x52e0 bytes. 0x2d20 bytes (35%) free.
[4/6] Linking CXX executable ethernet_basic.elf
FAILED: ethernet_basic.elf

.........
.........

x.a  -u __cxx_fatal_exception  -u uart_vfs_include_dev_init  -u usb_serial_jtag_vfs_include_dev_init  -u usb_serial_jtag_connection_monitor_include  -u esp_vfs_include_console_register  -u vfs_include_syscalls_impl  -u esp_vfs_include_nullfs_register  -u include_esp_phy_override  -lphy  -lbtbb  esp-idf/esp_phy/libesp_phy.a  -lphy  -lbtbb  esp-idf/esp_phy/libesp_phy.a  -lphy  -lbtbb && :
/home/pi/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/espressif__esp_hosted/libespressif__esp_hosted.a(rpc_slave_if.c.obj): 

in function `wifi_init':/home/pi/esps3host/esp-idf/examples/ethernet/basic/managed_components/espressif__esp_hosted/host/drivers/rpc/slaveif/rpc_slave_if.c:170: 

multiple definition of `wifi_init'; esp-idf/espressif__console_cmd_wifi/libespressif__console_cmd_wifi.a(console_wifi.c.obj):/home/pi/esps3host/esp-idf/examples/ethernet/basic/managed_components/espressif__console_cmd_wifi/console_wifi.c:95: first defined here
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/pi/esps3host/esp-idf/examples/ethernet/basic/build/log/idf_py_stderr_output_161916 and /home/pi/esps3host/esp-idf/examples/ethernet/basic/build/log/idf_py_stdout_output_161916

SohKamYung-Espressif, 2025-05-09:
Hi, @savenlid https://github.com/espressif/esp-hosted/issues/598#issuecomment-2865415117 is rather large to digest.

Regarding

>multiple definition of `wifi_init';

Could you provide the example project (includes `sdkconfig`, source code, top level CMake, `idf_component.yml` etc.) as a zip file that we can extract and run (execute `idf.py set-target esp32s3; idf.py menuconfig; idf.py build`) to see the issue?

1. which ESP-IDF example are you using?
2. what commands did you run, or files edited, to added wifi-remote and esp-hosted?
3. what other changes from made to the example?

savenlid, 2025-05-09:
Sorry

FYI, I have both spi-eth-dm905 and spi-hosted-c5 working simultaneously now.
By modifiyng:
examples/wifi/getting_started/station/
Thanks for helping me achieve this.

To the problem #1.
Its examples/network/sta2eth  but similar problem happened at others also.
I did not include /build  or /managed_componets to keep size down

[sta2eth.zip](https://github.com/user-attachments/files/20120009/sta2eth.zip)

#2, why do some examples bring in their own component-lib ( not compatible with the offical/docs)
IE:
example_eth_init(&s_eth_handles, &s_eth_port_cnt);   // esp-idf/example/xyz/managed_components
Versus
ethernet_init_all(&eth_handles, &eth_port_cnt);    // esp-idf/components

SohKamYung-Espressif, 2025-05-09:
Thanks for the summary.

> Its examples/network/sta2eth but similar problem happened at others also

I will check the zip. We think we know what is happening and will verity the fix with your zip file. Will let you know.

>why do some examples bring in their own component-lib ( not compatible with the offical/docs)
>IE:
>example_eth_init(&s_eth_handles, &s_eth_port_cnt); // esp-idf/example/xyz/managed_components
>Versus
>ethernet_init_all(&eth_handles, &eth_port_cnt); // esp-idf/components

ESP-Hosted doesn't handle ethernet configuration, so best to post this question to ESP-IDF.

I see you have already posted one question on eth SPI https://github.com/espressif/esp-idf/issues/15919 . You can post this question there also.

SohKamYung-Espressif, 2025-05-14:
@savenlid Commit https://github.com/espressif/esp-hosted-mcu/commit/6e118a303afc352011a72355c56850743ea3dbb1 should fix this issue:

> multiple definition of `wifi_init'

mantriyogesh, 2025-05-14:
@savenlid , Closing the issue for now.

Please feel free to re-open if the fix doesn't work.
Please create a new issue if something else is still not as per your expectations.

closed 2025-05-14 by mantriyogesh