espressif/esp-hosted #599: Issues in getting IP address (ip route table) after successfully WIFI STA CONNECT b/n host and esp32s3 in the esphosted fg implementation

praveenm-itas, 2025-04-30:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Issues in getting IP address (ip route table) after successfully WIFI STA CONNECT Between host and esp32s3 in the esphosted fg implementation.

1. Ported ESP32 Host side driver module code ,referred porting guide to Android based MYIR LT527  Allwinner board.

2. .SPI communication works fine , able to get esp32_spi: process_init_event: esp boot-up event processed message
3. using c-support hostcontrol test , using get_ap_scan_list -> getting all the AP list at host
4 . STA_CONNECT is connecting to the pre-configured wifi AP configured with SSID and PWD in ctrl_config.h
log on both esp32 side anf host side its show WIFI AP connected.- wifi:I (390942) slave_ctrl: connected to ap SSID:'ITAS-Air5',
but when ping from Host to the IP address of AP its unreachable, and also not able to ping any external IP address.

### Actual behavior (suspected bug)

when I ping from Host to the IP address of AP its unreachable, and also not able to ping any external IP address. can you please help what i am missing here after connection successful.

Logs attached for the following scenario ESP32 side and host side & c-support terminal log

[esp32_connect.txt](https://github.com/user-attachments/files/19974547/esp32_connect.txt)
[esp32_init.txt](https://github.com/user-attachments/files/19974543/esp32_init.txt)
[esp_app1.txt](https://github.com/user-attachments/files/19974544/esp_app1.txt)
[esp_host_connect.txt](https://github.com/user-attachments/files/19974545/esp_host_connect.txt)
[esp_host_init.txt](https://github.com/user-attachments/files/19974546/esp_host_init.txt)

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

Using c-support hostcontrol test , using get_ap_scan_list -> getting all the AP list at host
STA_CONNECT is connecting to the pre-configured wifi AP configured with SSID and PWD in ctrl_config.h
log on both esp32 side anf host side its show WIFI AP connected.- wifi:I (390942) slave_ctrl: connected to ap SSID:'ITAS-Air5',
> but when ping from Host to the IP address of AP its unreachable, and also not able to ping any external IP address.

### Project release version

ESP-Hosted-FG-release_0.0.5.tgz for rpi4b and ESP-Hosted-FG-release_0.0.5.tgz for esp32 s3 wroom 1

### System architecture

ARM 64-bit (Apple M1/M2, Raspberry Pi 4/5)

### Operating system

Linux

### Operating system version

linux-5.15 

### Shell

ZSH

mantriyogesh, 2025-04-30:
From what you've described, it seems you're connecting an external device to the ESP's SoftAP, and the ESP is also connecting to an external AP as a station. Now, you want to ping from the external device to the external AP. Is that correct?

[image]

If this is the issue, let me clarify, softap and sta interfaces are isolated; packets from SoftAP clients do not get routed automatically like this.
Simplest solution is Wi-Fi-NAT-router example:  https://github.com/espressif/esp-idf/tree/master/examples/wifi/softap_sta

praveenm-itas, 2025-04-30:
Just to describe  what we are doing:
 We are connecting Android based (MYIR LT527 Allwinner board) Host  with ESP32S3  via SPI interface
Now we want to achieve the WIFI connectivity on this Android based Host.
(here the  AP is our office  wifi router)

"From what you've described, it seems you're connecting an external device to the ESP's SoftAP, and the ESP is also connecting to an external AP as a station. Now, you want to ping from the external device to the external AP. Is that correct?"
>> its like the ESP is also connecting to an external AP ( our office  wifi router) as a station,   and trying to ping from the host machine ( which is connected via SPI with ESP32)  ,

mantriyogesh, 2025-04-30:
I understand now.

you are using normal station mode.

After the station connects to the AP, we bring the network interface up, but do not run a DHCP client by default.

This is because the DHCP client is a high-level feature, and the specific client used (e.g., dhclient, udhcpd, etc.) varies across Linux distributions. Our code doesn't include dhcp client code (and sit should also not include in my opinion)

In the ESP-Hosted code:

After CONNECT_AP succeeds, we call up_sta_netdev()
([code link](https://github.com/espressif/esp-hosted/blob/61471c115ba46253f739ac8cb07a632648ba8f87/esp_hosted_fg/host/linux/host_control/c_support/test_utils.c#L272-L324))
This brings up the ethsta0 interface and sets its MAC address, but does not assign an IP.

The Python support code provides a `--set_dhcp` flag
([code link](https://github.com/espressif/esp-hosted/blob/61471c115ba46253f739ac8cb07a632648ba8f87/esp_hosted_fg/host/linux/host_control/python_support/py_parse/process.py#L21-L25))
This tries to run `dhclient`, **assuming** it's available.

After connecting, you can check with `ifconfig` or `ip addr` that the `ethsta0` interface is up, with correct (non zero) MAC address.
You can manually run:
 
```
sudo dhclient -v ethsta0
```
to get IP on `ethsta0` interface.

In case you wish to hook, you can add up system commands 
```
system("sudo dhclient ethsta0 -r"); /* remove dhcp */
system("sudo dhclient ethsta0 -v"); /* perform dhcp client on staeth0 */
```

before this line, https://github.com/espressif/esp-hosted/blob/61471c115ba46253f739ac8cb07a632648ba8f87/esp_hosted_fg/host/linux/host_control/c_support/test_utils.c#L531

Please do not hesitate to comment should you have any confusion.

Although, I understand we could have embedded above system commands (just the same way we did not python), with some config flag for USE_DHCP_CLIENT , by default enabled.

praveenm-itas, 2025-05-05:
Hi @mantriyogesh ,
Below are the details  after connecting from Host to ESp32 via esp32_test utils commands

(1)  # ifconfig -a 

ethsta0   Link encap:Ethernet  HWaddr 24:ec:4a:7e:1b:90
          inet6 addr: fe80::26ec:4aff:fe7e:1b90/64 Scope: Link
          UP BROADCAST RUNNING  MTU:1500  Metric:1
          RX packets:528 errors:0 dropped:1 overruns:0 frame:0 
          TX packets:14 errors:0 dropped:0 overruns:0 carrier:0 
          collisions:0 txqueuelen:1000 
          RX bytes:55271 TX bytes:1368 

ethap0    Link encap:Ethernet  HWaddr 24:ec:4a:7e:1b:91
          inet6 addr: fe80::26ec:4aff:fe7e:1b91/64 Scope: Link
          UP BROADCAST RUNNING  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0 
          TX packets:10 errors:0 dropped:0 overruns:0 carrier:0 
          collisions:0 txqueuelen:1000 
          RX bytes:0 TX bytes:936 

App_main() - control_test() log also shows below messages  for MAC address.
MAC address 24:ec:4a:7e:1b:90 set to ethsta0 interface
ethsta0 interface up

(2) to get IP on ethsta0 interface:
I used dhcp client,  currently I am running from terminal (busybox_arm64) , once this works i can integrate in code.
as below after connect.
# ./busybox_arm64 udhcpc -i ethsta0 -v
udhcpc: started, v1.31.1
udhcpc: executing /usr/share/udhcpc/default.script deconfig
udhcpc: entering listen mode: raw
udhcpc: created raw socket
udhcpc: sending discover
udhcpc: waiting 3 seconds
udhcpc: unrelated/bogus packet, ignoring
udhcpc: waiting 3 seconds
udhcpc: unrelated/bogus packet, ignoring
udhcpc: waiting 3 seconds
udhcpc: unrelated/bogus packet, ignoring
udhcpc: waiting 2 seconds
udhcpc: unrelated/bogus packet, ignoring
udhcpc: waiting 2 seconds
udhcpc: unrelated/bogus packet, ignoring
udhcpc: waiting 2 seconds
udhcpc: unrelated/bogus packet, ignoring
udhcpc: waiting 1 seconds
udhcpc: unrelated/bogus packet, ignoring
udhcpc: waiting 1 seconds
udhcpc: sending discover
udhcpc: waiting 3 seconds
udhcpc: unrelated/bogus packet, ignoring

Not obtaining the IP.

3)  checked log-in router and this device is not listed even though its connected.

But the Esp32 Host side log shows/ lists what are the other laptops/devices connected to that router
03-21 14:41:10.179     0     0 I [    C0] esp32_spi: spi_data_ready_interrupt_handler: esp spi_data_ready_interrupt_handler
03-21 14:41:10.190     0     0 I [    C0] esp32_spi: spi_interrupt_handler: esp spi_interrupt_handler
03-21 14:41:10.198  2690  2690 D spi_rx  : 00000000bd7d5a68: f6 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
03-21 14:41:10.210  2690  2690 D spi_rx  : 00000000230df70a: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
03-21 14:41:10.222  2690  2690 D spi_rx  : 0000000036701b13: 00 00 3c 00 0c 00 94 09 00 00 00 00 24 ec 4a 7e  ..<.........$.J~
03-21 14:41:10.222     0     0 I [    C0] esp32_spi: spi_interrupt_handler: esp spi_interrupt_handler
03-21 14:41:10.234  2690  2690 D spi_rx  : 0000000034f799ca: 1b 90 00 17 61 12 8d 3d 08 06 00 01 08 00 06 04  ....a..=........
03-21 14:41:10.253    84    84 D rx      : 000000005f5de02a: 00 00 68 00 0c 00 27 1d 00 00 00 00 24 ec 4a 7e  ..h...'.....$.J~
03-21 14:41:10.264    84    84 D rx      : 00000000460bb1cc: 1b 90 34 6f 24 eb dd 75 86 dd 60 00 80 fc 00 32  ..4o$..u..`....2
03-21 14:41:10.275    84    84 D rx      : 00000000d552f53f: 11 01 fe 80 00 00 00 00 00 00 81 81 c4 95 53 51  ..............SQ
03-21 14:41:10.286    84    84 D rx      : 000000006b05af9c: 79 db ff 02 00 00 00 00 00 00 00 00 00 00 00 00  y...............
03-21 14:41:10.297    84    84 D rx      : 00000000ba939bae: 00 fb 14 e9 14 e9 00 32 72 4f 00 00 00 00 00 01  .......2rO......
[... 3 lines cut ...]
03-21 14:41:11.167     0     0 I [    C0] esp32_spi: spi_data_ready_interrupt_handler: esp spi_data_ready_interrupt_handler
03-21 14:41:11.177     0     0 I [    C0] esp32_spi: spi_interrupt_handler: esp spi_interrupt_handler
03-21 14:41:11.186  2690  2690 D spi_rx  : 00000000704ac74a: f6 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
03-21 14:41:11.197  2690  2690 D spi_rx  : 00000000a360e4c8: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
03-21 14:41:11.210     0     0 I [    C0] esp32_spi: spi_interrupt_handler: esp spi_interrupt_handler
03-21 14:41:11.210  2690  2690 D spi_rx  : 00000000b92864c7: 02 00 41 00 0c 00 00 17 10 00 00 00 01 08 00 63  ..A............c
03-21 14:41:11.230  2690  2690 D spi_rx  : 00000000761424e0: 74 72 6c 45 76 6e 74 02 33 00 08 03 10 af 02 fa  trlEvnt.3.......
03-21 14:41:11.241    66    66 D rx      : 00000000b92864c7: 02 00 41 00 0c 00 00 17 10 00 00 00 01 08 00 63  ..A............c
03-21 14:41:11.252    66    66 D rx      : 00000000761424e0: 74 72 6c 45 76 6e 74 02 33 00 08 03 10 af 02 fa  trlEvnt.3.......
03-21 14:41:11.263    66    66 D rx      : 000000007cefd6c2: 12 2b 12 09 49 54 41 53 2d 41 69 72 35 22 11 65  .+..ITAS-Air5".e
03-21 14:41:11.274    66    66 D rx      : 00000000bc97215e: 65 3a 32 32 3a 66 34 3a 32 65 3a 35 63 3a 64 61  e:22:f4:2e:5c:da
03-21 14:41:11.285    66    66 D rx      : 000000006e89556a: 28 0e 30 d4 ff ff ff ff ff ff ff ff 01           (.0..........
03-21 14:41:51.940   293   293 W healthd : battery none chg=u
03-21 14:42:51.942   293   293 W healthd : battery none chg=u
03-21 14:43:51.944   293   293 W healthd : battery none chg=u
03-21 14:44:14.517     0     0 I [    C7] esp32_spi: esp_hard_start_xmit: esp_hard_start_xmit called....
[... cut ...]
03-21 14:44:14.551     0     0 I [    C0] esp32_spi: spi_interrupt_handler: esp spi_interrupt_handler
03-21 14:44:14.562  2690  2690 D spi_rx  : 000000000cb1c761: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
03-21 14:44:51.946   293   293 W healthd : battery none chg=u
03-21 14:45:51.947   293   293 W healthd : battery none chg=u
03-21 14:46:51.949   293   293 W healthd : battery none chg=u
03-21 14:47:06.072   192   192 I logd    : logdr: UID=0 GID=0 PID=2723 b tail=0 logMask=80 pid=0 start=0ns deadline=0ns

mantriyogesh, 2025-05-06:
Please let us know once you have update @praveenm-itas.

In my opinion, it should have worked with the above steps. However, I understand it did not work as I expected.
I can check the above logs and come to a conclusion.

Also, similar to host logs, Would it be possible for you to enable slave side SPI level logs?
If so, it would be faster to locate the problem, if the request from host was reached or not.

praveenm-itas, 2025-05-07:
Hi @mantriyogesh ,

Attached below 3 logs
Full ESP32 logs from the beginning.
Complete dmesg log from the host, also from the start.
Commands fired and their output in terminal

127|t527-demo-car:/data/local/tmp #  ifconfig ethsta0 192.168.1.233    
                                                                                                                                    
t527-demo-car:/data/local/tmp #  ip route show
192.168.1.0/24 dev ethsta0 proto kernel scope link src 192.168.1.233 

t527-demo-car:/data/local/tmp # ping 192.168.1.1                                                                                                                                                           
connect: Network is unreachable

127|t527-demo-car:/data/local/tmp # ping -I ethsta0 192.168.1.1
PING 192.168.1.1 (192.168.1.1) from 192.168.1.233 ethsta0: 56(84) bytes of data.
From 192.168.1.233: icmp_seq=1 Destination Host Unreachable
From 192.168.1.233: icmp_seq=2 Destination Host Unreachable
From 192.168.1.233: icmp_seq=3 Destination Host Unreachable
From 192.168.1.233: icmp_seq=4 Destination Host Unreachable
From 192.168.1.233: icmp_seq=5 Destination Host Unreachable
From 192.168.1.233: icmp_seq=6 Destination Host Unreachable
From 192.168.1.233: icmp_seq=7 Destination Host Unreachable
From 192.168.1.233: icmp_seq=8 Destination Host Unreachable
From 192.168.1.233: icmp_seq=9 Destination Host Unreachable
From 192.168.1.233: icmp_seq=10 Destination Host Unreachable

[Host_side_full_log.txt](https://github.com/user-attachments/files/20076868/Host_side_full_log.txt)

[ESP32_side_log.txt](https://github.com/user-attachments/files/20076875/ESP32_side_log.txt)

[commands_terminal_log.txt](https://github.com/user-attachments/files/20076877/commands_terminal_log.txt)

mantriyogesh, 2025-05-07:
What is currently happening, 
```
I (855013) slave_ctrl: Stop Heatbeat
I (855013) slave_ctrl: Stopping HB timer
I (856668) wifi:<ba-add>idx:1 (ifx:0, ee:22:f4:2e:5c:da), tid:7, ssn:2, winSize:64
I (1089504) slave_ctrl: HB timer started for 20 sec

I (1090192) slave_ctrl: Wifi already configured earlier, ignore
I (1090192) slave_ctrl: Request successful
I (1090365) slave_ctrl: Get station mac address
I (1090365) slave_ctrl: mac [24:ec:4a:7e:1b:90] 
I (1091328) slave_ctrl: Get softap mac address
I (1091328) slave_ctrl: mac [24:ec:4a:7e:1b:91] 
I (1091576) slave_ctrl: Stop Heatbeat
I (1091577) slave_ctrl: Stopping HB timer
```

You are using `stress.out`, which is more of testing utility. This is more of to test if any inconsistencies while running the commands back to back. but in the same script it just runs the commands back to back ${\color{yellow} \text{blindly}}$

${\color{yellow} \text{It is just a demonstration}}$. So is the same and simplified in `test.c`, which creates `test.out` output, which could be used in following way described.

Can you just simply try these things:
1. load the kernel module (you would have done it already)
2. cd c_support/
change `ctrl_config.h` with your desired SSID and Password along lines:
https://github.com/espressif/esp-hosted/blob/61471c115ba46253f739ac8cb07a632648ba8f87/esp_hosted_fg/host/linux/host_control/c_support/ctrl_config.h#L84-L85

2. build the test.out with new changes

```
make test
```
3. connect station mode
```
sudo ./test.out sta_connect
```

This would show something like:
```
------ ESP-Hosted FW [FG-0.0.6.0.0] ------
Connect to AP[Yogesh]

Requested operation complete
Sleeping for some time just to showcase heartbeat
ctrl_app_parse_resp:503 Info: Connect band_mode is 0
ethsta0 interface down
MAC address XXXXXXXXX set to ethsta0 interface
ethsta0 interface up
```

4. Once the interface up with MAC address like above, then either invoke dhcp client or static ip

```
sudo dhclient -v ethsta0
ping 8.8.8.8
```

<details>
<summary> logs at my side to refer </summary>

```
ym2@rpi2 ~/h/e/h/l/h/c_support [1]> vim ctrl_config.h
ym2@rpi2 ~/h/e/h/l/h/c_support> make test
gcc -C -Wall --sysroot= -I/home/ym2/h2_old_18jan/esp_hosted_fg/host/linux/host_control/c_support/../../../../common/protobuf-c -I/home/ym2/h2_old_18jan/esp_hosted_fg/host/linux/host_control/c_support/../../../../common/include -I/home/ym2/h2_old_18jan/esp_hosted_fg/host/linux/host_control/c_support/../../../control_lib/include -I/home/ym2/h2_old_18jan/esp_hosted_fg/host/linux/host_control/c_support/../../../control_lib/src/include -I/home/ym2/h2_old_18jan/esp_hosted_fg/host/linux/host_control/c_support/../../../virtual_serial_if/include -I/home/ym2/h2_old_18jan/esp_hosted_fg/host/linux/host_control/c_support/../../../components/include -I/home/ym2/h2_old_18jan/esp_hosted_fg/host/linux/host_control/c_support/../../port/include -I. /home/ym2/h2_old_18jan/esp_hosted_fg/host/linux/host_control/c_support/../../../../common/protobuf-c/protobuf-c/protobuf-c.c /home/ym2/h2_old_18jan/esp_hosted_fg/host/linux/host_control/c_support/../../../../common/esp_hosted_config.pb-c.c /home/ym2/h2_old_18jan/esp_hosted_fg/host/linux/host_control/c_support/../../../control_lib/src/ctrl_core.c /home/ym2/h2_old_18jan/esp_hosted_fg/host/linux/host_control/c_support/../../../control_lib/src/ctrl_api.c /home/ym2/h2_old_18jan/esp_hosted_fg/host/linux/host_control/c_support/../../../virtual_serial_if/src/serial_if.c /home/ym2/h2_old_18jan/esp_hosted_fg/host/linux/host_control/c_support/../../../components/src/esp_queue.c /home/ym2/h2_old_18jan/esp_hosted_fg/host/linux/host_control/c_support/../../port/src/platform_wrapper.c ./test_utils.c -lpthread -lrt test.c -o test.out
ym2@rpi2 ~/h/e/h/l/h/c_support> sudo ./test.out sta_connect
Enable heartbeat with duration 20
Heartbeat operation successful
------ ESP-Hosted FW [FG-0.0.6.0.0] ------
Connect to AP[Yogesh]

[... 46 lines cut ...]
Internet Systems Consortium DHCP Client 4.4.3-P1
Copyright 2004-2022 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Listening on LPF/ethsta0/40:4c:ca:4a:a8:50
Sending on   LPF/ethsta0/40:4c:ca:4a:a8:50
Sending on   Socket/fallback
DHCPDISCOVER on ethsta0 to 255.255.255.255 port 67 interval 8
DHCPOFFER of 172.20.10.5 from 172.20.10.1
DHCPREQUEST for 172.20.10.5 on ethsta0 to 255.255.255.255 port 67
DHCPACK of 172.20.10.5 from 172.20.10.1
bound to 172.20.10.5 -- renewal in 1631 seconds.
ym2@rpi2 ~/h/e/h/l/h/c_support>
ym2@rpi2 ~/h/e/h/l/h/c_support> ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.
64 bytes from 8.8.8.8: icmp_seq=1 ttl=115 time=62.2 ms
64 bytes from 8.8.8.8: icmp_seq=2 ttl=115 time=43.8 ms
64 bytes from 8.8.8.8: icmp_seq=3 ttl=115 time=54.0 ms
^C
--- 8.8.8.8 ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2003ms
rtt min/avg/max/mdev = 43.819/53.325/62.170/7.506 ms
```
</details>

mantriyogesh, 2025-05-07:
<details>
<summary> ESP logs at my side to refer </summary>

```
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0x1e (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875730,len:0x173c
load:0x4086b910,len:0xe9c
--- 0x4086b910: esp_bootloader_get_description at /Users/yogesh/esp-idf/components/esp_bootloader_format/esp_bootloader_desc.c:39 (discriminator 1)

load:0x4086e610,len:0x30ec
--- 0x4086e610: esp_flash_encryption_enabled at /Users/yogesh/esp-idf/components/bootloader_support/src/flash_encrypt.c:89

entry 0x4086b91a
--- 0x4086b91a: call_start_cpu0 at /Users/yogesh/esp-idf/components/bootloader/subproject/main/bootloader_start.c:25

I (23) boot: ESP-IDF v5.5-dev-2303-gc864ee3701-dirty 2nd stage bootloader
I (24) boot: compile time May  7 2025 15:15:23
I (24) boot: chip revision: v0.0
I (26) boot: efuse block revision: v0.1
I (29) boot.esp32c6: SPI Speed      : 80MHz
I (33) boot.esp32c6: SPI Mode       : DIO
I (37) boot.esp32c6: SPI Flash Size : 4MB
I (41) boot: Enabling RNG early entropy source...
I (45) boot: Partition Table:
I (48) boot: ## Label            Usage          Type ST Offset   Length
I (54) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (61) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (67) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (74) boot:  3 ota_0            OTA app          00 10 00010000 00180000
I (80) boot:  4 ota_1            OTA app          00 11 00190000 00180000
I (87) boot: End of partition table
I (90) esp_image: segment 0: paddr=00010020 vaddr=420c0020 size=25bc8h (154568) map
I (126) esp_image: segment 1: paddr=00035bf0 vaddr=40800000 size=0a428h ( 42024) load
I (136) esp_image: segment 2: paddr=00040020 vaddr=42000020 size=ba9b4h (764340) map
I (277) esp_image: segment 3: paddr=000fa9dc vaddr=4080a428 size=158c4h ( 88260) load
I (297) esp_image: segment 4: paddr=001102a8 vaddr=4081fcf0 size=03cc4h ( 15556) load
I (308) boot: Loaded app from partition at offset 0x10000
I (309) boot: Disabling RNG early entropy source...
I (318) cpu_start: Unicore app
I (327) cpu_start: Pro cpu start user code
I (327) cpu_start: cpu freq: 160000000 Hz
I (327) app_init: Application information:
I (327) app_init: Project name:     network_adapter
I (332) app_init: App version:      FG-0.0.6.0.0
I (336) app_init: Compile time:     May  7 2025 15:15:15
I (341) app_init: ELF file SHA256:  3e8fba45d...
I (345) app_init: ESP-IDF:          v5.5-dev-2303-gc864ee3701-dirty
I (351) efuse_init: Min chip rev:     v0.0
I (355) efuse_init: Max chip rev:     v0.99
I (359) efuse_init: Chip rev:         v0.0
I (363) heap_init: Initializing. RAM available for dynamic allocation:
I (369) heap_init: At 408297F0 len 00052E20 (331 KiB): RAM
I (374) heap_init: At 4087C610 len 00002F54 (11 KiB): RAM
I (380) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM
I (386) spi_flash: detected chip: generic
I (389) spi_flash: flash io: dio
W (392) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (404) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (410) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (417) coexist: coex firmware version: e727207
I (435) coexist: coexist rom version 5b8dcfa
I (436) main_task: Started on CPU0
I (436) main_task: Calling app_main()
I (436) NETWORK_ADAPTER: *********************************************************************
I (443) NETWORK_ADAPTER:                 ESP-Hosted Firmware version :: FG-0.0.6.0.0
I (450) NETWORK_ADAPTER:                 Transport used :: SPI only
[... 116 lines cut ...]
I (6049) wifi:connected with Yogesh, aid = 1, channel 6, BW20, bssid = 36:a2:64:36:92:12
I (6055) wifi:security: WPA2-PSK, phy:11ax, rssi:-46, cipher(pairwise:0x3, group:0x3), pmf:0,
I (6064) wifi:pm start, type: 1, twt_start:0

I (6068) wifi:pm start, type:1, aid:0x1, trans-BSSID:36:a2:64:36:92:12, BSSID[5]:0x12, mbssid(max-indicator:0, index:0), he:1
I (6079) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (6087) wifi:set rx beacon pti, rx_bcn_pti: 10, bcn_timeout: 25000, mt_pti: 10, mt_time: 10000
I (6095) wifi:[ADDBA]TX addba request, tid:0, dialogtoken:1, bufsize:32, A-MSDU:0(not supported), policy:1(IMR), ssn:0(0x0)
I (6106) wifi:[ADDBA]TX addba request, tid:7, dialogtoken:2, bufsize:32, A-MSDU:0(not supported), policy:1(IMR), ssn:0(0x20)
I (6117) wifi:[ADDBA]TX addba request, tid:5, dialogtoken:3, bufsize:32, A-MSDU:0(not supported), policy:1(IMR), ssn:0(0x0)
I (6128) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (6134) wifi:[ADDBA]RX addba response, status:0, tid:0/tb:1(0xa1), bufsize:32, batimeout:0, txa_wnd:32
I (6142) wifi:[ADDBA]RX addba response, status:0, tid:7/tb:1(0xa1), bufsize:32, batimeout:0, txa_wnd:32
I (6152) wifi:[ADDBA]RX addba response, status:0, tid:5/tb:1(0xa1), bufsize:16, batimeout:0, txa_wnd:16
I (6161) slave_ctrl: Manual Wi-Fi connected, no event raised
I (6166) slave_ctrl: connected to ap SSID:'Yogesh', password:'PlantMoreTrees123'
I (6173) slave_ctrl: req_connect_ap_handler:655 Set resp to Success
I (13549) slave_ctrl: Stop Heatbeat
I (13550) slave_ctrl: Stopping HB timer
W (35199) wifi:<ba-add>idx:0, ifx:0, tid:6, TAHI:0x1001292, TALO:0x3664a236, (ssn:0, win:64, cur_ssn:0), CONF:0xc0006005
W (39279) wifi:<ba-add>idx:1, ifx:0, tid:0, TAHI:0x1001292, TALO:0x3664a236, (ssn:2, win:64, cur_ssn:2), CONF:0xc0000005

<Note: nothing else ahead in esp side log, after 'connected', which is significant difference in your and mine log >
```
</details>

mantriyogesh, 2025-05-07:
# ESP-Hosted: Documentation & Usage Notes

## 📚 Recommended Documentation

- **Low-Level RPC APIs (Host–Slave Communication)**  
  [Control APIs Documentation](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/common/ctrl_apis.md)

- **Higher-Level User-Space APIs**  
  Easier-to-use APIs can be found in [test.h](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/host/linux/host_control/c_support/test.h) and used directly in your application once the demo is working.

- **Control Path Design**  
  [Control Path Overview](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/common/contrl_path.md)

---

## 🚦 Using `enable_wifi()` and Related Functions

- `enable_wifi()`, `disable_wifi()`, and `get_ap_scan_list()` are **not required** for a basic Wi-Fi connection.
- Wi-Fi and Bluetooth are **enabled by default** at boot.  Some users needed to manually disable, that was the reason 'disable_xxx()' and 'enable_xxx()' for Wi-Fi and Bluetooth was introduced.
- To connect to an AP, use `sta_connect()` directly — it sets up the mode, performs scan, and connects.

### ⚠️ Current Behavior (Bug?)

- Calling `enable_wifi()` **clears previous Wi-Fi state**, even if Wi-Fi is already enabled.
- After calling `enable_wifi()`, you **must call `sta_connect()` again** to reconnect.
- This behavior is expected to improve in future releases (e.g., `enable_wifi()` becoming idempotent).

---

## 🔔 Event Subscription

You can register and handle asynchronous Wi-Fi events such as disconnects.

- Events documentation:  
  [Control Path Events](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/common/ctrl_apis.md#2-control-path-events)

- For example:  
  On **station disconnect**, you can trigger an **asynchronous RPC request** to reconnect.

---

## 🔁 Synchronous vs Asynchronous Requests

- RPC (or Control) Responses and Events are 'Asynchronous' in nature.
- Only RPC Requests could be Synchronous or Asynchronous.
 - Request becomes asynchronous if response callback was provided in the Request.
 - If response callback is not populated, it remains NULL. it would be trated as Synchronous request.
 - Any Request is by default Synchronous.
 - Synchronous Request is called 'synchronous' because it blocks for the response. 
 - Asynchronous Request is called 'asynchronous' because it **does not ** block for the response.

| Source Context      | Send Sync RPC Request | Send Async RPC Request |
|---------------------|------------------|-------------------|
| Synchronous Thread  | ✅ Supported     | ✅ Supported      |
| Asynchronous Thread | ❌ Not Supported | ✅ Supported      |

- Event callbacks are invoked from the **RX thread**, which must remain:
  - **Non-blocking**
  - **Fast**
- Async response callback also are called from **RX thread**, which must remain:
  - **Non-blocking**
  - **Fast**

praveenm-itas, 2025-05-07:
I tried  building with test.c  and  tried as per the steps you mentioned ,
Only triggered connect command  this time ( SSID & pwd in  .h file)

> 1|t527-demo-car:/data/local/tmp # ./esp32_wifi_control_test sta_connect

------ ESP-Hosted FW [FG-0.0.6.0.0] ------
Connect to AP[ITAS-Air5]ctrl_app_send_req -> req.msg_id :107 
4. Protobuf msg size4. Protobuf msg size tx_len = 36 
5. Allocate protobuf msg 
6. Assign response callback 
7. Start timeout for response for async only 
8. Pack in protobuf and send the request 
Serial connection: transport_pserial_send
Serial handle available
 before serial_drv_write 
serial_drv_write, .... just before ko write 
serial_drv_write SUCCESS 
9. Free hook for application
10. Cleanup

Requested operation complete
Sleeping for some time just to showcase heartbeat
ctrl_app_parse_resp:505 Info: Connect band_mode is 0
ethsta0 interface down
MAC address 24:ec:4a:7e:1b:90 set to ethsta0 interface
ethsta0 interface up
^C

After this assigned Static IP 
t527-demo-car:/data/local/tmp # ifconfig ethsta0 192.168.1.233                                                                                                                                             
t527-demo-car:/data/local/tmp # ip route show                                                                                                                                                              
192.168.1.0/24 dev ethsta0 proto kernel scope link src 192.168.1.233 
t527-demo-car:/data/local/tmp # ping 192.168.1.1                                                                                                                                                           
connect: Network is unreachable
2|t527-demo-car:/data/local/tmp # ping -I ethsta0 192.168.1.1
PING 192.168.1.1 (192.168.1.1) from 192.168.1.233 ethsta0: 56(84) bytes of data.
From 192.168.1.233: icmp_seq=12 Destination Host Unreachable
From 192.168.1.233: icmp_seq=13 Destination Host Unreachable
From 192.168.1.233: icmp_seq=14 Destination Host Unreachable
^C
--- 192.168.1.1 ping statistics ---
17 packets transmitted, 0 received, +3 errors, 100% packet loss, time 16420ms
pipe 3
127|t527-demo-car:/data/local/tmp # ping 8.8.8.8                                                                                                                                                           
connect: Network is unreachable
2|t527-demo-car:/data/local/tmp # ping -I ethsta0 8.8.8.8                                                                                                                                                  
PING 8.8.8.8 (8.8.8.8) from 192.168.1.233 ethsta0: 56(84) bytes of data.
From 192.168.1.233: icmp_seq=1 Destination Host Unreachable
From 192.168.1.233: icmp_seq=2 Destination Host Unreachable
From 192.168.1.233: icmp_seq=3 Destination Host Unreachable
^C
--- 8.8.8.8 ping statistics ---
6 packets transmitted, 0 received, +3 errors, 100% packet loss, time 5092ms
pipe 4

praveenm-itas, 2025-05-07:
DHCP client:

1|t527-demo-car:/data/local/tmp # ./busybox_arm64 udhcpc -i ethsta0 -v                                                                                                                                     
udhcpc: started, v1.31.1
udhcpc: executing /usr/share/udhcpc/default.script deconfig
udhcpc: entering listen mode: raw
udhcpc: created raw socket
udhcpc: sending discover
udhcpc: waiting 3 seconds
udhcpc: unrelated/bogus packet, ignoring
udhcpc: waiting 2 seconds
udhcpc: unrelated/bogus packet, ignoring
udhcpc: waiting 1 seconds
udhcpc: sending discover
udhcpc: waiting 3 seconds
udhcpc: received a packet
udhcpc: xid a929a87f (our is 6f1a4884), ignoring packet
udhcpc: waiting 3 seconds
udhcpc: unrelated/bogus packet, ignoring
udhcpc: waiting 1 seconds
udhcpc: unrelated/bogus packet, ignoring
udhcpc: sending discover
udhcpc: waiting 3 seconds
udhcpc: unrelated/bogus packet, ignoring
udhcpc: waiting 3 seconds
udhcpc: unrelated/bogus packet, ignoring
udhcpc: waiting 3 seconds
udhcpc: unrelated/bogus packet, ignoring
udhcpc: waiting 1 seconds
udhcpc: executing /usr/share/udhcpc/default.script leasefail

mantriyogesh, 2025-05-07:
I am not understanding why this should fail !

Let me think and come back on this.

mantriyogesh, 2025-05-08:
# ESP-Hosted SPI Slave Extra Debug Logs

## Rationale

The STA connect request is reaching the slave, but IP assignment is not completing. These debug logs are added to validate if the **DHCP Discover** request (or other packets) from the host are reaching the ESP32-S3 over SPI, and whether the slave is successfully pushing responses back. This will help identify where packet loss or processing issues may occur.

---

## 1. Slave RX Logs

- [ ] **Before RX validation**
  - **File:** `spi_slave_api.c`
  - **Line:** [L584](https://github.com/espressif/esp-hosted/blob/61471c115ba46253f739ac8cb07a632648ba8f87/esp_hosted_fg/esp/esp_driver/network_adapter/main/spi_slave_api.c#L584)
  - **Code to add:**
    ```c
    ESP_LOG_BUFFER_HEXDUMP("raw_rx", rx_buf_handle.payload, 32, ESP_LOG_INFO);
    ```

- [ ] **After RX validation, before processing**
  - **File:** `app_main.c`
  - **Lines:** [L383–384](https://github.com/espressif/esp-hosted/blob/61471c115ba46253f739ac8cb07a632648ba8f87/esp_hosted_fg/esp/esp_driver/network_adapter/main/app_main.c#L383)
  - **Replace existing:**
    ```c
    ESP_LOGD(TAG, "Rx pkt: type:%u\n",buf_handle->if_type);
    ESP_LOG_BUFFER_HEXDUMP(TAG, payload, payload_len, ESP_LOG_DEBUG);
    ```
  - **With:**
    ```c
    ESP_LOGI(TAG, "Rx pkt: type:%u\n", buf_handle->if_type);
    ESP_LOG_BUFFER_HEXDUMP(TAG, payload, payload_len, ESP_LOG_INFO);
    ```

---

## 2. Slave TX Logs

- [ ] **Before pushing TX buffer to SPI**
  - **File:** `spi_slave_api.c`
  - **Line:** [L813](https://github.com/espressif/esp-hosted/blob/61471c115ba46253f739ac8cb07a632648ba8f87/esp_hosted_fg/esp/esp_driver/network_adapter/main/spi_slave_api.c#L813)
  - **Code to add:**
    ```c
    ESP_LOGI(TAG, "Tx pkt: type:%u\n", tx_buf_handle.if_type);
    ESP_LOG_BUFFER_HEXDUMP(TAG, tx_buf_handle.payload, tx_buf_handle.payload_len, ESP_LOG_INFO);
    ```

- [ ] **TX buffer just after SPI transaction**
  - **File:** `spi_slave_api.c`
  - **Line:** [L569](https://github.com/espressif/esp-hosted/blob/61471c115ba46253f739ac8cb07a632648ba8f87/esp_hosted_fg/esp/esp_driver/network_adapter/main/spi_slave_api.c#L569)
  - **Code to add:**
    ```c
    ESP_LOG_BUFFER_HEXDUMP("raw_tx", spi_trans->tx_buffer, 32, ESP_LOG_INFO);
    ```

---

## Flashing Steps (ESP32-S3)

Once the code changes are made:

```bash
# 1. Set up environment (assuming earlier manual flashing is already done and can be re-used as is)
cd esp_hosted_fg/esp/esp_driver/network_adapter

# 2. Build-Flash-Logs of ESP32-S3
idf.py flash monitor
```

Please help to reproduce the scenario with `udhcpc` and get us the text logs at ESP (above monitor from start), host logs (dmesg from system bootup), command-response you fired in terminal

praveenm-itas, 2025-05-08:
Added the  all log messages provided 
1. After build and flash ,  I just run sta_connect 
2. then udhcpc client from host

Attached all three logs
1.  Host side
2. ESP32_side ( messages at the end are during  DHCP client  running from host side)
3. terminal command logs

[Host_side_log.txt](https://github.com/user-attachments/files/20100451/Host_side_log.txt)

[Esp32_side_log.txt](https://github.com/user-attachments/files/20100454/Esp32_side_log.txt)

[test_command_terminal_log.txt](https://github.com/user-attachments/files/20100468/test_command_terminal_log.txt)

kindly let me know any other info needed.

praveenm-itas, 2025-05-08:
Attaching another set of  complete logs  ( in case needed) from second test

1.  run sta_connect
2.  run udhcpc client from host

Attached all three set of logs
Host side
ESP32_side ( messages at the end are during DHCP client running from host side)
terminal command logs

[ESP32_side_log.txt](https://github.com/user-attachments/files/20101873/ESP32_side_log.txt)

[Host_side_log.txt](https://github.com/user-attachments/files/20101875/Host_side_log.txt)

[Terminal_command_log.txt](https://github.com/user-attachments/files/20101878/Terminal_command_log.txt)

mantriyogesh, 2025-05-09:
Hwllo @praveenm-itas ,

I checked you spi level logs at both esp and host.

The issue does not appear to be at the SPI level - the communication between ESP32 and the host is working properly, with packets being transferred in both directions.

From logs
```
1|t527-demo-car:/data/local/tmp # ./busybox_arm64 udhcpc -i ethsta0 -v
udhcpc: started, v1.31.1
udhcpc: executing /usr/share/udhcpc/default.script deconfig
udhcpc: entering listen mode: raw
udhcpc: created raw socket
udhcpc: sending discover
udhcpc: waiting 3 seconds
udhcpc: unrelated/bogus packet, ignoring
udhcpc: waiting 3 seconds
udhcpc: unrelated/bogus packet, ignoring
udhcpc: waiting 2 seconds
udhcpc: received a packet
udhcpc: sending select for 192.168.1.33
udhcpc: waiting 3 seconds
udhcpc: received a packet
udhcpc: xid e0203918 (our is 282a05ac), ignoring packet
udhcpc: waiting 3 seconds
udhcpc: unrelated/bogus packet, ignoring
udhcpc: waiting 3 seconds
udhcpc: unrelated/bogus packet, ignoring
udhcpc: waiting 2 seconds
udhcpc: unrelated/bogus packet, ignoring
udhcpc: waiting 2 seconds
udhcpc: unrelated/bogus packet, ignoring
udhcpc: waiting 2 seconds
udhcpc: unrelated/bogus packet, ignoring
udhcpc: waiting 1 seconds
udhcpc: received a packet
udhcpc: lease of 192.168.1.33 obtained, lease time 86400
udhcpc: executing /usr/share/udhcpc/default.script bound
udhcpc: entering listen mode: none
t527-demo-car:/data/local/tmp # 
```

```
udhcpc: xid e0203918 (our is 282a05ac), ignoring packet
```
What this means the transaction id with which dhcp offer was sent, did not match to one originated by dhcp client. (may be dhcp server not configured correctly or this is dhcp offer for another client running)

Summary till now is,
- All the messages are being received at udhcpc as well. but I suspect some 
The problem is occurring at the network application layer, specifically with the DHCP protocol.
- The fact that DHCP eventually succeeds after multiple retries suggests that the issue is intermittent rather than a complete failure, which is typical of application-layer protocol implementation issues rather than hardware or low-level communication problems.

**However, If you still have doubt on SPI level or ESP-Hosted**, we can have debug session call.
If call happens, I would need some things before the hand prepared, so as we would use the call time for crucial activities.

1. Send a photo of your esp and host connections
2. Please gather the hardware connections you have used in tabular form
3. wireshark sniffer
- In case you do not have dedicated sniffer with you, use mac laptop as sniffer
- In case you do not have mac laptop, please configure linux laptop as sniffer
- test your sniffer works
4. host terminal access
- I would need to type commands remotely over your screen - anydesk software in such debugs is useful)
5. esp log display real time 

Some issues I want to highlight,
1. There are many ARPs sent from slave to router, with no responses
2. DHCP client doesn't recognise the responses received from wifi router

Line of treatment:
1. You can give a try to different AP and use (unused) static ip first.
- localise if ping are processing without involvement of udhcpc
- Use less crowded AP, for lessor confusion 
2. If (1) doesn't work, we can prepare for debug call.

Please suggest your inputs on this. Any inputs are welcome.

praveenm-itas, 2025-05-09:
Hi @mantriyogesh ,
It is good to know that SPI and  esp - host communication works properly,
let me check with  options/try you mentioned and dhcp client &  diff AP and come back with their result and  further details.

praveenm-itas, 2025-05-13:
Hi @mantriyogesh ,

1) Now With static IP  its able to ping the router 192.168.1.1
t527-demo-car:/data/local/tmp # ifconfig ethsta0 192.168.1.35 
t527-demo-car:/data/local/tmp #  ip route show
192.168.1.0/24 dev ethsta0 proto kernel scope link src 192.168.1.35 

2|t527-demo-car:/data/local/tmp # ping -I ethsta0 192.168.1.1                                                                                                                                              
PING 192.168.1.1 (192.168.1.1) from 192.168.1.35 ethsta0: 56(84) bytes of data.
From 192.168.1.35: icmp_seq=37 Destination Host Unreachable
From 192.168.1.35: icmp_seq=38 Destination Host Unreachable
From 192.168.1.35: icmp_seq=39 Destination Host Unreachable
64 bytes from 192.168.1.1: icmp_seq=1 ttl=64 time=51130 ms
64 bytes from 192.168.1.1: icmp_seq=2 ttl=64 time=50823 ms
64 bytes from 192.168.1.1: icmp_seq=3 ttl=64 time=49948 ms
64 bytes from 192.168.1.1: icmp_seq=4 ttl=64 time=50388 ms
64 bytes from 192.168.1.1: icmp_seq=5 ttl=64 time=50360 ms
64 bytes from 192.168.1.1: icmp_seq=6 ttl=64 time=50005 ms
64 bytes from 192.168.1.1: icmp_seq=7 ttl=64 time=49265 ms
64 bytes from 192.168.1.1: icmp_seq=8 ttl=64 time=48673 ms
64 bytes from 192.168.1.1: icmp_seq=9 ttl=64 time=48540 ms
64 bytes from 192.168.1.1: icmp_seq=10 ttl=64 time=50082 ms
64 bytes from 192.168.1.1: icmp_seq=11 ttl=64 time=53701 ms
64 bytes from 192.168.1.1: icmp_seq=12 ttl=64 time=54091 ms
64 bytes from 192.168.1.1: icmp_seq=13 ttl=64 time=53543 ms
64 bytes from 192.168.1.1: icmp_seq=14 ttl=64 time=53111 ms
64 bytes from 192.168.1.1: icmp_seq=15 ttl=64 time=52201 ms
64 bytes from 192.168.1.1: icmp_seq=16 ttl=64 time=52421 ms
64 bytes from 192.168.1.1: icmp_seq=17 ttl=64 time=51622 ms
64 bytes from 192.168.1.1: icmp_seq=18 ttl=64 time=51140 ms
64 bytes from 192.168.1.1: icmp_seq=19 ttl=64 time=50275 ms
64 bytes from 192.168.1.1: icmp_seq=20 ttl=64 time=49422 ms
64 bytes from 192.168.1.1: icmp_seq=21 ttl=64 time=49633 ms
64 bytes from 192.168.1.1: icmp_seq=22 ttl=64 time=49631 ms
64 bytes from 192.168.1.1: icmp_seq=23 ttl=64 time=49393 ms
64 bytes from 192.168.1.1: icmp_seq=24 ttl=64 time=49705 ms
64 bytes from 192.168.1.1: icmp_seq=25 ttl=64 time=49880 ms
64 bytes from 192.168.1.1: icmp_seq=26 ttl=64 time=49859 ms
64 bytes from 192.168.1.1: icmp_seq=27 ttl=64 time=49011 ms
64 bytes from 192.168.1.1: icmp_seq=28 ttl=64 time=48215 ms
64 bytes from 192.168.1.1: icmp_seq=29 ttl=64 time=47286 ms
64 bytes from 192.168.1.1: icmp_seq=30 ttl=64 time=47118 ms
64 bytes from 192.168.1.1: icmp_seq=31 ttl=64 time=46201 ms
64 bytes from 192.168.1.1: icmp_seq=32 ttl=64 time=45293 ms
64 bytes from 192.168.1.1: icmp_seq=33 ttl=64 time=45616 ms
64 bytes from 192.168.1.1: icmp_seq=34 ttl=64 time=51168 ms

> From 192.168.1.35: icmp_seq=37 Destination Host Unreachableable    
couldn't initially reach the AP.

> 64 bytes from 192.168.1.1: icmp_seq=1 ttl=64 time=51130 ms
Later success, but with high latency  time=51130 ms

2) Check timestamp difference between request & reply.
# ping -I ethsta0 192.168.1.1 
 
# tcpdump -i ethsta0 icmp 
15:13:45.229022 IP 192.168.1.35 > 192.168.1.1: ICMP echo request, id 5, seq 291, length 64
15:13:46.253252 IP 192.168.1.35 > 192.168.1.1: ICMP echo request, id 5, seq 292, length 64
15:13:47.699941 IP 192.168.1.1 > 192.168.1.35: ICMP echo reply, id 5, seq 1, length 64
15:13:48.345294 IP 192.168.1.1 > 192.168.1.35: ICMP echo reply, id 5, seq 2, length 64
15:13:48.633544 IP 192.168.1.35 > 192.168.1.1: ICMP echo request, id 5, seq 293, length 64
15:13:48.642194 IP 192.168.1.35 > 192.168.1.1: ICMP echo request, id 5, seq 294, length 64
15:13:49.280775 IP 192.168.1.35 > 192.168.1.1: ICMP echo request, id 5, seq 295, length 64
15:13:50.065299 IP 192.168.1.1 > 192.168.1.35: ICMP echo reply, id 5, seq 3, length 64
15:13:50.281040 IP 192.168.1.35 > 192.168.1.1: ICMP echo request, id 5, seq 296, length 64
15:13:50.332004 IP 192.168.1.1 > 192.168.1.35: ICMP echo reply, id 5, seq 4, length 64
15:13:51.282652 IP 192.168.1.35 > 192.168.1.1: ICMP echo request, id 5, seq 297, length 64
15:13:52.301619 IP 192.168.1.35 > 192.168.1.1: ICMP echo request, id 5, seq 298, length 64
15:13:53.325035 IP 192.168.1.35 > 192.168.1.1: ICMP echo request, id 5, seq 299, length 64
[... 10 lines cut ...]
15:14:04.621446 IP 192.168.1.35 > 192.168.1.1: ICMP echo request, id 5, seq 310, length 64
15:14:05.650826 IP 192.168.1.35 > 192.168.1.1: ICMP echo request, id 5, seq 311, length 64
15:14:06.669171 IP 192.168.1.35 > 192.168.1.1: ICMP echo request, id 5, seq 312, length 64
15:14:07.725236 IP 192.168.1.35 > 192.168.1.1: ICMP echo request, id 5, seq 313, length 64
15:14:08.749476 IP 192.168.1.35 > 192.168.1.1: ICMP echo request, id 5, seq 314, length 64
15:14:09.100199 IP 192.168.1.1 > 192.168.1.35: ICMP echo reply, id 5, seq 7, length 64
15:14:09.723127 IP 192.168.1.1 > 192.168.1.35: ICMP echo reply, id 5, seq 8, length 64
15:14:09.748960 IP 192.168.1.35 > 192.168.1.1: ICMP echo request, id 5, seq 315, length 64
15:14:10.765325 IP 192.168.1.35 > 192.168.1.1: ICMP echo request, id 5, seq 316, length 64

What we can observe is ::

ICMP Echo Requests from 192.168.1.35 are being sent consistently 
ICMP Echo Replies from 192.168.1.1 are delayed

15:13:50.065299 IP 192.168.1.1 > 192.168.1.35: ICMP echo reply, id 5, seq 3, length 64
ICMP replies for old requests (reply to seq=3 comes after requests up to seq=295).

As we observed in earlier log , DHCP Client also getting IP lease intermittently, observed xid mismatch most of the time,  this also could be due to  slow reply.

Logs when Ping is going , in case needed.

[ESP32_side_logs.txt](https://github.com/user-attachments/files/20185246/ESP32_side_logs.txt)

[host_side_log_13_05_2025.txt](https://github.com/user-attachments/files/20185279/host_side_log_13_05_2025.txt)

mantriyogesh, 2025-05-13:
> ICMP Echo Replies from 192.168.1.1 are delayed

I'm seeing delayed ICMP Echo Replies from 192.168.1.1 and trying to figure out if the router is the issue.

The ESP’s receiver should be fast enough, even with logging enabled. The logging might add up to (hypothetically, at the max 100ms of jitter), but not delays of several seconds — that's suspicious.

From the logs, the SPI link seems to be working instantly, so it's likely not the cause of the delay (this long).

Have you tried using a different AP or hotspot?

Keep in mind, your laptop might be on 5GHz while the ESP uses 2.4GHz. Testing with a different standard AP or hotspot can help narrow down the issue.

praveenm-itas, 2025-05-15:
Hi @mantriyogesh ,  Managed to get another router/AP to test this.  Below is the details, compare to previous one this PING reply and ICMP reply time is better, 

------ ESP-Hosted FW [FG-0.0.6.0.0] ------
Connect to AP[Itas_tata2G]ctrl_app_send_req -> req.msg_id :107 
4. Protobuf msg size4. Protobuf msg size tx_len = 43 
5. Allocate protobuf msg 
6. Assign response callback 
7. Start timeout for response for async only 
8. Pack in protobuf and send the request 
Serial connection: transport_pserial_send
Serial handle available
 before serial_drv_write 
serial_drv_write, .... just before ko write 
serial_drv_write SUCCESS 
9. Free hook for application
10. Cleanup

Requested operation complete
Sleeping for some time just to showcase heartbeat
ctrl_app_parse_resp:505 Info: Connect band_mode is 0
ethsta0 interface down
MAC address 24:ec:4a:7e:1b:90 set to ethsta0 interface
ethsta0 interface up
2025-03-21 14:22:39 >  App EVENT: Heartbeat event [0]
2025-03-21 14:22:59 >  App EVENT: Heartbeat event [1]
^C

t527-demo-car:/data/local/tmp # ip route show
192.168.1.0/24 dev ethsta0 proto kernel scope link src 192.168.1.35 

t527-demo-car:/data/local/tmp # ping -I ethsta0 192.168.1.254
PING 192.168.1.254 (192.168.1.254) from 192.168.1.35 ethsta0: 56(84) bytes of data.
64 bytes from 192.168.1.254: icmp_seq=1 ttl=64 time=3726 ms
64 bytes from 192.168.1.254: icmp_seq=2 ttl=64 time=4155 ms
64 bytes from 192.168.1.254: icmp_seq=3 ttl=64 time=3836 ms
64 bytes from 192.168.1.254: icmp_seq=4 ttl=64 time=3449 ms
64 bytes from 192.168.1.254: icmp_seq=5 ttl=64 time=2592 ms
64 bytes from 192.168.1.254: icmp_seq=6 ttl=64 time=3235 ms
64 bytes from 192.168.1.254: icmp_seq=7 ttl=64 time=3382 ms
64 bytes from 192.168.1.254: icmp_seq=8 ttl=64 time=3201 ms
64 bytes from 192.168.1.254: icmp_seq=9 ttl=64 time=4791 ms
64 bytes from 192.168.1.254: icmp_seq=10 ttl=64 time=4375 ms
64 bytes from 192.168.1.254: icmp_seq=11 ttl=64 time=4662 ms
64 bytes from 192.168.1.254: icmp_seq=12 ttl=64 time=4346 ms
64 bytes from 192.168.1.254: icmp_seq=13 ttl=64 time=3476 ms
64 bytes from 192.168.1.254: icmp_seq=14 ttl=64 time=4753 ms
64 bytes from 192.168.1.254: icmp_seq=15 ttl=64 time=4255 ms
64 bytes from 192.168.1.254: icmp_seq=16 ttl=64 time=4655 ms
64 bytes from 192.168.1.254: icmp_seq=17 ttl=64 time=4459 ms
64 bytes from 192.168.1.254: icmp_seq=18 ttl=64 time=3493 ms
64 bytes from 192.168.1.254: icmp_seq=19 ttl=64 time=4048 ms
64 bytes from 192.168.1.254: icmp_seq=20 ttl=64 time=3295 ms
64 bytes from 192.168.1.254: icmp_seq=21 ttl=64 time=2771 ms
64 bytes from 192.168.1.254: icmp_seq=22 ttl=64 time=4556 ms
64 bytes from 192.168.1.254: icmp_seq=23 ttl=64 time=4403 ms
64 bytes from 192.168.1.254: icmp_seq=24 ttl=64 time=4334 ms
64 bytes from 192.168.1.254: icmp_seq=25 ttl=64 time=3643 ms
64 bytes from 192.168.1.254: icmp_seq=26 ttl=64 time=3362 ms
64 bytes from 192.168.1.254: icmp_seq=27 ttl=64 time=3545 ms
64 bytes from 192.168.1.254: icmp_seq=28 ttl=64 time=3472 ms
64 bytes from 192.168.1.254: icmp_seq=29 ttl=64 time=4387 ms
64 bytes from 192.168.1.254: icmp_seq=30 ttl=64 time=4695 ms
64 bytes from 192.168.1.254: icmp_seq=31 ttl=64 time=4707 ms
64 bytes from 192.168.1.254: icmp_seq=32 ttl=64 time=4433 ms
64 bytes from 192.168.1.254: icmp_seq=33 ttl=64 time=3620 ms
64 bytes from 192.168.1.254: icmp_seq=34 ttl=64 time=5047 ms
64 bytes from 192.168.1.254: icmp_seq=35 ttl=64 time=4394 ms
64 bytes from 192.168.1.254: icmp_seq=36 ttl=64 time=4420 ms
64 bytes from 192.168.1.254: icmp_seq=37 ttl=64 time=4013 ms
64 bytes from 192.168.1.254: icmp_seq=38 ttl=64 time=4480 ms
[... 21 lines cut ...]
14:25:15.305995 IP 192.168.1.35 > 192.168.1.254: ICMP echo request, id 1, seq 59, length 64
14:25:15.474539 IP 192.168.1.254 > 192.168.1.35: ICMP echo reply, id 1, seq 54, length 64
14:25:16.094097 IP 192.168.1.254 > 192.168.1.35: ICMP echo reply, id 1, seq 55, length 64
14:25:16.306207 IP 192.168.1.35 > 192.168.1.254: ICMP echo request, id 1, seq 60, length 64
14:25:16.653836 IP 192.168.1.254 > 192.168.1.35: ICMP echo reply, id 1, seq 56, length 64
14:25:17.117487 IP 192.168.1.254 > 192.168.1.35: ICMP echo reply, id 1, seq 57, length 64
14:25:17.306164 IP 192.168.1.35 > 192.168.1.254: ICMP echo request, id 1, seq 61, length 64
14:25:17.668123 IP 192.168.1.254 > 192.168.1.35: ICMP echo reply, id 1, seq 58, length 64
14:25:18.307831 IP 192.168.1.35 > 192.168.1.254: ICMP echo request, id 1, seq 62, length 64
14:25:19.341216 IP 192.168.1.35 > 192.168.1.254: ICMP echo request, id 1, seq 63, length 64
14:25:19.689479 IP 192.168.1.254 > 192.168.1.35: ICMP echo reply, id 1, seq 59, length 64
14:25:20.343523 IP 192.168.1.35 > 192.168.1.254: ICMP echo request, id 1, seq 64, length 64
14:25:20.955372 IP 192.168.1.254 > 192.168.1.35: ICMP echo reply, id 1, seq 60, length 64
14:25:21.344903 IP 192.168.1.35 > 192.168.1.254: ICMP echo request, id 1, seq 65, length 64
14:25:21.997874 IP 192.168.1.254 > 192.168.1.35: ICMP echo reply, id 1, seq 61, length 64
14:25:22.346312 IP 192.168.1.35 > 192.168.1.254: ICMP echo request, id 1, seq 66, length 64
14:25:22.844112 IP 192.168.1.254 > 192.168.1.35: ICMP echo reply, id 1, seq 62, length 64
14:25:23.347598 IP 192.168.1.35 > 192.168.1.254: ICMP echo request, id 1, seq 67, length 64
14:25:23.702911 IP 192.168.1.254 > 192.168.1.35: ICMP echo reply, id 1, seq 63, length 64
14:25:24.349628 IP 192.168.1.35 > 192.168.1.254: ICMP echo request, id 1, seq 68, length 64
14:25:25.389114 IP 192.168.1.35 > 192.168.1.254: ICMP echo request, id 1, seq 69, length 64
14:25:25.737383 IP 192.168.1.254 > 192.168.1.35: ICMP echo reply, id 1, seq 64, length 64
14:25:26.395157 IP 192.168.1.35 > 192.168.1.254: ICMP echo request, id 1, seq 70, length 64
14:25:26.759818 IP 192.168.1.254 > 192.168.1.35: ICMP echo reply, id 1, seq 65, length 64
14:25:27.397325 IP 192.168.1.35 > 192.168.1.254: ICMP echo request, id 1, seq 71, length 64

praveenm-itas, 2025-05-15:
2)   Next case  for Dynamic IP

------ ESP-Hosted FW [FG-0.0.6.0.0] ------
Connect to AP[Itas_tata2G]ctrl_app_send_req -> req.msg_id :107 
4. Protobuf msg size4. Protobuf msg size tx_len = 43 
5. Allocate protobuf msg 
6. Assign response callback 
7. Start timeout for response for async only 
8. Pack in protobuf and send the request 
Serial connection: transport_pserial_send
Serial handle available
 before serial_drv_write 
serial_drv_write, .... just before ko write 
serial_drv_write SUCCESS 
9. Free hook for application
10. Cleanup

Requested operation complete
Sleeping for some time just to showcase heartbeat
ctrl_app_parse_resp:505 Info: Connect band_mode is 0
ethsta0 interface down
MAC address 24:ec:4a:7e:1b:90 set to ethsta0 interface
ethsta0 interface up
2025-03-21 14:15:18 >  App EVENT: Heartbeat event [0]
2025-03-21 14:15:38 >  App EVENT: Heartbeat event [1]

1|t527-demo-car:/data/local/tmp # ifconfig -a
ethsta0   Link encap:Ethernet  HWaddr 24:ec:4a:7e:1b:90
          inet6 addr: 2402:e280:218e:3f1:26ec:4aff:fe7e:1b90/64 Scope: Global
          inet6 addr: fe80::26ec:4aff:fe7e:1b90/64 Scope: Link
          UP BROADCAST RUNNING  MTU:1500  Metric:1
          RX packets:68 errors:0 dropped:3 overruns:0 frame:0 
          TX packets:53 errors:0 dropped:0 overruns:0 carrier:0 
          collisions:0 txqueuelen:1000 
          RX bytes:7385 TX bytes:5108

t527-demo-car:/data/local/tmp # ip route show                                                                                                                                                              
t527-demo-car:/data/local/tmp # /vendor/bin//busybox udhcpc -i ethsta0 -v
Adapter index 19
MAC 24:ec:4a:7e:1b:90
udhcpc (v1.23.0.git) started
Executing /usr/share/udhcpc/default.script deconfig
Entering listen mode: raw
Opening raw socket on ifindex 19
Got raw socket fd
Created raw socket
Adapter index 19
MAC 24:ec:4a:7e:1b:90
Sending discover...
Waiting on select 3 seconds
Received a packet
xid 2253bf0 (our is 6d752862), ignoring packet
Waiting on select 2 seconds
Adapter index 19
MAC 24:ec:4a:7e:1b:90
Sending discover...
Waiting on select 3 seconds
Received a packet
Adapter index 19
MAC 24:ec:4a:7e:1b:90
Sending select for 192.168.1.39...
Waiting on select 3 seconds
Received a packet
xid 2253bf0 (our is 6d752862), ignoring packet
Waiting on select 3 seconds
Received a packet
Waiting on select 2 seconds
Received a packet
Lease of 192.168.1.39 obtained, lease time 86400
Executing /usr/share/udhcpc/default.script bound
Entering listen mode: none
t527-demo-car:/data/local/tmp # ip route show                                                                                                                                                              
t527-demo-car:/data/local/tmp # ifconfig -a         
ethsta0   Link encap:Ethernet  HWaddr 24:ec:4a:7e:1b:90
          inet6 addr: 2402:e280:218e:3f1:26ec:4aff:fe7e:1b90/64 Scope: Global
          inet6 addr: fe80::26ec:4aff:fe7e:1b90/64 Scope: Link
          UP BROADCAST RUNNING  MTU:1500  Metric:1
          RX packets:127 errors:0 dropped:3 overruns:0 frame:0 
          TX packets:88 errors:0 dropped:0 overruns:0 carrier:0 
          collisions:0 txqueuelen:1000 
          RX bytes:14881 TX bytes:9200 
               
t527-demo-car:/data/local/tmp # ip route show                                                                                                                                                              
t527-demo-car:/data/local/tmp # ping -I ethsta0 192.168.1.254
ping: Warning: source address might be selected on device other than ethsta0.
PING 192.168.1.254 (192.168.1.254) from 0.0.0.0 ethsta0: 56(84) bytes of data.

>> Here  DHCP  leased IP also, but IP route show is not showing  its 0.0.0.0

your input will help me to check further.

mantriyogesh, 2025-05-16:
> https://github.com/espressif/esp-hosted/issues/599#issuecomment-2883425492

If the AP is pingable, the 8.8.8.8 would be pingable if your network setting allow so.

Check the default route and add to AP ip and try again ping.
/etc/resolv.conf also can be updated with 8.8.8.8 once pingable.

Please cross-check if there is some issue on your Linux wrt networking.

The Per packet logs at host and slave should be strictly commented. they would slow down the packets excessively.

in normal operation no logs should be printed either side including not not limited to interrupts and per packet logs.

praveenm-itas, 2025-06-04:
Hi @mantriyogesh ,

I have been trying this to include the above  with Android  stack (to trigger from wifi settings) to seamlessly scan and connect  using the Abstraction layer which internally invokes esp32 apis

One issue i am observing is:

The MAC address is assigned by  ESP32 Host controller APIs  only after  CTRL_RESP_CONNECT_AP  ( i.e Connect i established)
it is assigning esp32 Mac addr to  then network Interface type STA(ethsta0) and setup the interface_up.

But In android, the flow is bit different where after loading driver and Vendor_HAL  library ,  during the initialisation of HAL it is checking for interface type  ethsat0 in this case , but the MAC addr is 0:0:0:0:0   during this time. and flow is failing in this case.

So any alternate way to resolve this , can we assign MAC before that ?,  I tried querying mac addr using get_mac_addr API and its returning Mac addr  without CONNECT.  

Thanks,
Praveen

praveenm-itas, 2025-07-22:
Hi @mantriyogesh ,

I have completed the basic Android part integration , and able to connect using Settings->Wifi

Able to get dhcp lease & ping  external IP 8.8.8.8.8  google.com  also, but speed looks very slow

t527-demo-car:/ $  ip addr show ethsta0
18: ethsta0: <BROADCAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UNKNOWN group default qlen 1000
    link/ether 24:ec:4a:7e:1b:90 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.39/24 brd 192.168.1.255 scope global ethsta0
       valid_lft forever preferred_lft forever
    inet6 2402:e280:218e:3f1:a10a:b8fd:8b72:32c0/64 scope global temporary dynamic 
       valid_lft 172799sec preferred_lft 85807sec
    inet6 2402:e280:218e:3f1:4191:c064:a538:2a97/64 scope global dynamic mngtmpaddr stable-privacy 
       valid_lft 172799sec preferred_lft 172799sec
    inet6 fe80::26ec:4aff:fe7e:1b90/64 scope link 
       valid_lft forever preferred_lft forever

t527-demo-car:/ $ ping -I ethsta0 8.8.8.8                                                                                                                                                                  
PING 8.8.8.8 (8.8.8.8) from 192.168.1.39 ethsta0: 56(84) bytes of data.
64 bytes from 8.8.8.8: icmp_seq=1 ttl=119 time=10947 ms
64 bytes from 8.8.8.8: icmp_seq=2 ttl=119 time=10302 ms
64 bytes from 8.8.8.8: icmp_seq=3 ttl=119 time=10279 ms
64 bytes from 8.8.8.8: icmp_seq=4 ttl=119 time=9993 ms
64 bytes from 8.8.8.8: icmp_seq=5 ttl=119 time=9608 ms
64 bytes from 8.8.8.8: icmp_seq=6 ttl=119 time=10717 ms
64 bytes from 8.8.8.8: icmp_seq=7 ttl=119 time=11214 ms
64 bytes from 8.8.8.8: icmp_seq=8 ttl=119 time=11295 ms
64 bytes from 8.8.8.8: icmp_seq=9 ttl=119 time=11014 ms
64 bytes from 8.8.8.8: icmp_seq=10 ttl=119 time=10440 ms
64 bytes from 8.8.8.8: icmp_seq=11 ttl=119 time=10531 ms
64 bytes from 8.8.8.8: icmp_seq=12 ttl=119 time=10186 ms
64 bytes from 8.8.8.8: icmp_seq=13 ttl=119 time=10756 ms
64 bytes from 8.8.8.8: icmp_seq=14 ttl=119 time=12395 ms
64 bytes from 8.8.8.8: icmp_seq=15 ttl=119 time=12821 ms
64 bytes from 8.8.8.8: icmp_seq=16 ttl=119 time=11988 ms
64 bytes from 8.8.8.8: icmp_seq=17 ttl=119 time=11808 ms
64 bytes from 8.8.8.8: icmp_seq=18 ttl=119 time=13364 ms
64 bytes from 8.8.8.8: icmp_seq=19 ttl=119 time=12899 ms

1) Now As I see there is  FG 1.0 released   - tag/release fg-v1.0.0.0.0, , Shall I take this complete release, or you suggest Master branch tip?.

2)  Currently its in default Bandwidth ,  in case need to change  to 20 MHz or 40 MHz bandwidth, where all changes need to incorporated.

mantriyogesh, 2025-07-23:
There is something wrong. I would like to know this better if some debug session is feasible from your side to provide isolated control of Linux and ESP.

But before that, I suggest to run the [raw transport throughput](https://gitlab.espressif.cn:6688/app-frameworks/esp_hosted/-/blob/master/esp_hosted_fg/docs/Linux_based_host/Raw_TP_Testing.md?ref_type=heads) in both directions one-by-one, to first rule out any issues at the transport level.

Also, Are there any additional logs that you enabled at host or slave side?

mantriyogesh, 2025-09-09:
This looks serious. Can you please send us the full dmesg log and slave side logs?

How easy is this to reproduce btw, @praveenm-itas ?

mantriyogesh, 2025-09-09:
Can you also please state the git commits you use at slave and host? 
It is FG, if i am not wrong.