espressif/esp-hosted #601: Change control interface SDIO to UART

alexsvn, 2025-05-16:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

Change control interface SDIO to UART
Hosted Wi-Fi control is based on SDIO. Can it be changed to UART channel for control? What changes are needed and what should be noted

### Use cases

Na

mantriyogesh, 2025-05-16:
Hello @alexsvn ,

Are you using Linux or MCU?

- MCU
 - For MCU we had added uart support, but that makes it everything over uart (control, data) in commit 
https://github.com/espressif/esp-hosted-mcu/commit/84550663bedd4a1a4af003fa8204b618554f3ff9
 - see line https://github.com/espressif/esp-hosted-mcu/commit/84550663bedd4a1a4af003fa8204b618554f3ff9#diff-57f3aa59b9682eb1decaa8f9494716b0e4dd35cec07b3c94d6231049747bf7ebR89-R95
- Linux Not supported yet

Can you please state your requirement little clear? Any intentions of such separation if we can understand, to proceed if this has to be incorporated or possibly leave open for later implementation.

alexsvn, 2025-05-16:
Dear @mantriyogesh,
Linux platform
Resources are limited and control areas are restricted;
Is there a UART based control path that does not run wireless data; Or how to remove wireless data processing from the current MCU hosted control path, give me some suggestions
BR,
Alex

mantriyogesh, 2025-05-16:
I can suggest something on this on Monday earliest, with some details.

mantriyogesh, 2025-05-21:
> Or how to remove wireless data processing from the current MCU hosted control path

# Tx Path

## Option 1: Comment calls for esp_hosted_tx() & drop for non ESP_PRIV_IF and ESP_SERIAL_IF

Actually, this should be simple in MCU. If you wish to only run control path and avoid any other 'traffic type' to be allowed to use the bus, you can just have a look at 

```
esp_hosted_tx()
```

This function is called from all the traffic when actual tx is need to be done.

If you wish to disallow station traffic to flow, just comment 

https://github.com/espressif/esp-hosted-mcu/blob/478d946a5db8ca8f66053eda90d27966a9fee3a3/host/drivers/transport/transport_drv.c#L242

and drop packet (or route through other transport maybe, as per your requirement)

For softap traffic, similar:
https://github.com/espressif/esp-hosted-mcu/blob/478d946a5db8ca8f66053eda90d27966a9fee3a3/host/drivers/transport/transport_drv.c#L259

## Option 2) Edit esp_hosted_tx() & drop for non ESP_PRIV_IF and ESP_SERIAL_IF

https://github.com/espressif/esp-hosted-mcu/blob/478d946a5db8ca8f66053eda90d27966a9fee3a3/host/drivers/transport/uart/uart_drv.c#L511-L554

change to :

```diff
int esp_hosted_tx(uint8_t iface_type, uint8_t iface_num,
		uint8_t * wbuffer, uint16_t wlen, uint8_t buff_zcopy,
		void (*free_wbuf_fun)(void* ptr))
{
	interface_buffer_handle_t buf_handle = {0};
	void (*free_func)(void* ptr) = NULL;
	uint8_t pkt_prio = PRIO_Q_OTHERS;
	uint8_t transport_up = is_transport_tx_ready();

	if (free_wbuf_fun)
		free_func = free_wbuf_fun;

	if (!wbuffer || !wlen ||
		(wlen > MAX_PAYLOAD_SIZE) ||
		!transport_up) {
		ESP_LOGE(TAG, "tx fail: NULL buff, invalid len (%u) or len > max len (%u), transport_up(%u))",
				wlen, MAX_PAYLOAD_SIZE, transport_up);
		H_FREE_PTR_WITH_FUNC(free_func, wbuffer);
		return ESP_FAIL;
	}

++ /* code start */
++    if ( (iface_type != ESP_SERIAL_IF) &&
++          (iface_type != ESP_PRIV_IF)) {
++          H_FREE_PTR_WITH_FUNC(free_func, wbuffer);
++          return ESP_OK;
++    }
++ /* code end */

	buf_handle.payload_zcopy = buff_zcopy;
	buf_handle.if_type = iface_type;
	buf_handle.if_num = iface_num;
	buf_handle.payload_len = wlen;
	buf_handle.payload = wbuffer;
	buf_handle.priv_buffer_handle = wbuffer;
	buf_handle.free_buf_handle = free_func;

	if (buf_handle.if_type == ESP_SERIAL_IF)
		pkt_prio = PRIO_Q_SERIAL;
	else if (buf_handle.if_type == ESP_HCI_IF)
		pkt_prio = PRIO_Q_BT;

	g_h.funcs->_h_queue_item(to_slave_queue[pkt_prio], &buf_handle, HOSTED_BLOCK_MAX);
	g_h.funcs->_h_post_semaphore(sem_to_slave_queue);

#if ESP_PKT_STATS
	if (buf_handle.if_type == ESP_STA_IF)
		pkt_stats.sta_tx_in_pass++;
#endif

	return ESP_OK;
}
```

# Rx path

Do not handle Rx for traffic type other than serial and priv types:

Change 
https://github.com/espressif/esp-hosted-mcu/blob/478d946a5db8ca8f66053eda90d27966a9fee3a3/host/drivers/transport/uart/uart_drv.c#L328-L377

to 

```diff
static int is_valid_uart_rx_packet(uint8_t *rxbuff_a, uint16_t *len_a, uint16_t *offset_a)
{
    struct esp_payload_header * h = (struct esp_payload_header *)rxbuff_a;
    uint16_t len = 0, offset = 0;
#if H_UART_CHECKSUM
    uint16_t rx_checksum = 0, checksum = 0;
#endif

    if (!h || !len_a || !offset_a)
        return 0;

    /* Fetch length and offset from payload header */
    len = le16toh(h->len);
    offset = le16toh(h->offset);

    if ((!len) ||
        (len > MAX_PAYLOAD_SIZE) ||
        (offset != sizeof(struct esp_payload_header))) {

        /* Free up buffer, as one of following -
         * 1. no payload to process
         * 2. input packet size > driver capacity
         * 3. payload header size mismatch,
         * wrong header/bit packing?
         * */
        return 0;
    }

#if H_UART_CHECKSUM
    rx_checksum = le16toh(h->checksum);
    h->checksum = 0;
    checksum = compute_checksum((uint8_t*)h, len + offset);

    if (checksum != rx_checksum) {
        ESP_LOGE(TAG, "UART RX rx_chksum[%u] != checksum[%u]. Drop.",
                checksum, rx_checksum);
        return 0;
    }
#endif

++    if ((h->if_type != ESP_SERIAL_IF) &&  (h->if_type != ESP_PRIV_IF)) {
++        return 0;
++    }

#if ESP_PKT_STATS
    if (h->if_type == ESP_STA_IF)
        pkt_stats.sta_rx_in++;
#endif

    *len_a = len;
    *offset_a = offset;

    return 1;
}
```

mantriyogesh, 2025-05-21:
# Linux - only control path on uart

##### Background of 'uart' used in hosted linux

Although we support UART in Linux, for Bluetooth, the UART driver is directly claimed by `hci_uart` kernel module, **independent** of espressif esp-hosted kernel module.

##### Ensure you are in control of uart bus

What I mean, the uart is not in control of hosted, when it is used directly by hci_uart.

Ensure the uart bus is not claimed by anyone, that you intend to use.
You can enable the bus in device tree referring to your SoC document (if not by default enabled)

If no uart is expected to be used for **bluetooth**, Possibly kernel should be rebuilt without `CONFIG_BT_HCIUART`.

For PoC purpose, just do not load `hci_uart` manually or as dependency. 

In any case, IMO, the first one claiming the uart bus would go through. in your software, you would fail fatally, if uart bus is already claimed (as you need to claim)

- Option 1) keeping design intact - Kernel module way
Add up your uart driver similar to one `spi` as in https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/host/linux/host_driver/esp32/spi/esp_spi.c
\
which would define in your uart driver, alternatives for
\
https://github.com/espressif/esp-hosted/blob/92335c3758ebaa76a72371688eea3d62125eee91/esp_hosted_fg/host/linux/host_driver/esp32/spi/esp_spi.c#L77-L80

- Option 2) Handle uart directly in user space in your application itself.

----
Similar to MCU changes stated in earlier comment, Linux [Rx path ](https://github.com/espressif/esp-hosted/blob/92335c3758ebaa76a72371688eea3d62125eee91/esp_hosted_fg/host/linux/host_driver/esp32/main.c#L506) and [Tx path](https://github.com/espressif/esp-hosted/blob/92335c3758ebaa76a72371688eea3d62125eee91/esp_hosted_fg/host/linux/host_driver/esp32/main.c#L642) in main.c of kernel module would need to be changed, disallow non ESP_SERIAL_IF and ESP_PRIV_IF.

mantriyogesh, 2025-05-21:
Summary
---------

MCU: If you can use MCU as host, then solution is almost there just just small tweaks.
Linux: This requires some development efforts, with reference to existing bus implementation

alexsvn, 2025-07-08:
Dear @mantriyogesh ,
Thank you very much for your reply. As I am busy with other matters recently, I have not verified this yet.

Your reply is mainly based on the interception processing done by the control end driver. Can the control (using uart "hci has been removed") and data (keeping sdio) be separated on the esp32 side? The host side only needs to change the esps0 controlled by host_control to the host uart (control instructions use the control path). Is this processing feasible? Please give me some suggestions

thank you!

mantriyogesh, 2025-07-08:
Yes, in fact, it is easily feasible using existing bus, SPI. (I know your question is for UART, addressed in later part of comment)

The incoming data from the host to the slave hits the function, https://github.com/espressif/esp-hosted/blob/5f5daf930fc20f48787d62d8b3de6b5a9d81942f/esp_hosted_fg/esp/esp_driver/network_adapter/main/esp_hosted_coprocessor.c#L422
which segregates the traffic using the `if_type`, https://github.com/espressif/esp-hosted/blob/5f5daf930fc20f48787d62d8b3de6b5a9d81942f/esp_hosted_fg/common/include/adapter.h#L58-L66

Process Rx packet is done such that,
- Functions start
```
static void process_rx_pkt(interface_buffer_handle_t *buf_handle)
{

	struct esp_payload_header *header = NULL;

...
	header = (struct esp_payload_header *) buf_handle->payload;
...
```

- Wifi Station mode host->slave packets processing (Optional in your case)
``` 
	if (buf_handle->if_type == ESP_STA_IF && station_connected) {
...
			ret = esp_wifi_internal_tx(ESP_IF_WIFI_STA, payload, payload_len);
...
```

- Wi-Fi softap mode host->slave packets processing  (Optional in your case)
```
	} else if (buf_handle->if_type == ESP_AP_IF && softap_started) {
...
		esp_wifi_internal_tx(ESP_IF_WIFI_AP, payload, payload_len);
```

- **Serial packets  (Mandatory in your case)**
```
	} else if (buf_handle->if_type == ESP_SERIAL_IF) {
...
		process_serial_rx_pkt(buf_handle->payload);
```

- **Priv packets  (Mandatory - Needed for internal processing)**
```
	} else if (buf_handle->if_type == ESP_PRIV_IF) {
		process_priv_pkt(payload, payload_len);
	}
```
- Bluetooth packets (VHCI) -  (Optional in your case)
```
#if defined(CONFIG_BT_ENABLED) && BLUETOOTH_HCI
	else if (buf_handle->if_type == ESP_HCI_IF) {
		process_hci_rx_pkt(payload, payload_len);
	}
#endif
```

- Raw throughput (physical link testing, not used in production) -  (Optional in your case)
```
#if TEST_RAW_TP
	else if (buf_handle->if_type == ESP_TEST_IF) {
		debug_update_raw_tp_rx_count(payload_len);
	}
#endif
```
```
...
...
} // Function ends
```

As you need UART as transport, the ${\color{orange} \text {UART driver needs to be amended to the ESP kernel module driver}}$, in parallel or similar to SPI, at [esp_hosted_fg/host/linux/host_driver/esp32/spi](https://github.com/espressif/esp-hosted/tree/master/esp_hosted_fg/host/linux/host_driver/esp32/spi). 

> [!Note]
> I do not want to confuse you here, but still I had to mention `Bluetooth over UART bypass mode`. It is important to distinguish this from the UART over Bluetooth mode, which is documented [here](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/UART_setup.md). Please be aware that this is a distinct and specialized mode that exclusively handles Bluetooth packets.
> 
> The default Bluetooth over UART configuration is transparent to ESP-Hosted. 
> ${\color{red} \text {This mode is not suited for your case}}$
> **What does this mean?**
> The HCI over UART driver is directly managed by the base Linux driver, meaning that the packets do not interact with the ESP-Hosted kernel driver at all.
> 
> **Why do we mention this? **
> We recognize that users may not be familiar with this mode, which sometimes what they wish to do.
>
> Is it possible to modify this to work with normal packets instead of HCI packets?  
> It is uncertain; you may need to review your SoC's HCI over UART driver. However, in my opinion, this approach is not advisable. The previously discussed implementation is cleaner and more maintainable, utilizing normal data/control frames sent via `/dev/esps0` file operations, which operate over the serial interface (SERIAL_IF) with the ESP-Hosted header.