espressif/esp-hosted #605: Station dump

miglio, 2025-05-30:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

Hello,

I tried esp_hosted_ng but I notice that when working as a AP with a client connected, the command:

`iw dev wlan0 station dump`

provides no output.
Do you think this feature could be added to the firmware? Is something that we could do on our own, or it requires access to binary components of the firmware?

### Use cases

This is very useful to list clients connected to the AP

mantriyogesh, 2025-05-30:
@kapilkedawat PTAL

kapilkedawat, 2025-05-30:
@miglio For SoftAP, we actually don’t have much configuration or firmware code — almost all the handling is done by hostapd, so it's better to use hostapd_cli to get more information.

closed 2025-06-24 by miglio

kapilkedawat, 2025-06-24:
@miglio somehow I missed the last comment, are you still facing RSSI issue?

miglio, 2025-06-24:
@kapilkedawat yes, the issue is stille there (but, let me say, is not a blocking one).