espressif/esp-hosted #607: Does not build for Linux 6.15.1

libschmudie-so, 2025-06-07:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

esp-hosted builds successfully

### Actual behavior (suspected bug)

esp-hosted does not build successfully

### Error logs or terminal output

```shell
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- -C /tmp/build-moekC8/linux-6.15.1 M=/tmp/build-moekC8/esphosted/esp_hosted_ng/host modules
make[1]: Entering directory '/tmp/build-moekC8/linux-6.15.1'
make[2]: Entering directory '/tmp/build-moekC8/esphosted/esp_hosted_ng/host'
  CC [M]  sdio/esp_sdio.o
sdio/esp_sdio.c:8:10: fatal error: utils.h: No such file or directory
    8 | #include "utils.h"
      |          ^~~~~~~~~
compilation terminated.
make[4]: *** [/tmp/build-moekC8/linux-6.15.1/scripts/Makefile.build:203: sdio/esp_sdio.o] Error 1
make[3]: *** [/tmp/build-moekC8/linux-6.15.1/Makefile:2003: .] Error 2
make[2]: *** [/tmp/build-moekC8/linux-6.15.1/Makefile:248: __sub-make] Error 2
make[2]: Leaving directory '/tmp/build-moekC8/esphosted/esp_hosted_ng/host'
```

### Steps to reproduce the behavior

Build Linux 6.15.1 for arm
Try to build esp-hosted-ng for it

### Project release version

master

### System architecture

ARM 32-bit (Raspberry Pi 32-bit)

### Operating system

Linux

### Operating system version

Debian 12

### Shell

ZSH

mantriyogesh, 2025-06-08:
Thanks for reporting @libschmudie-so , we were unaware as we did not get build issue.

We are working on it. Let me get back with the fix.

mantriyogesh, 2025-06-08:
[586.patch](https://github.com/user-attachments/files/20642792/586.patch)

The change is in review, anyway, by the time you can apply this patch to get ahead

libschmudie-so, 2025-06-08:
Unfortunately even with this patch the issue still persists.

```
HEAD is now at 0699edf Merge branch 'docs/c5_changes' into 'master'
patching file esp_hosted_ng/host/Makefile
patching file esp_hosted_ng/host/esp_debugfs.c
patching file esp_hosted_ng/host/sdio/esp_sdio.c
make: Entering directory '/tmp/build-BbkEBV/esphosted/esp_hosted_ng/host'
rm -rf *.o sdio/*.o spi/*.o *.ko
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- -C /tmp/build-BbkEBV/linux-6.15.1 M=/tmp/build-BbkEBV/esphosted/esp_hosted_ng/host clean
make[1]: Entering directory '/tmp/build-BbkEBV/linux-6.15.1'
make[2]: Entering directory '/tmp/build-BbkEBV/esphosted/esp_hosted_ng/host'
make[2]: Leaving directory '/tmp/build-BbkEBV/esphosted/esp_hosted_ng/host'
make[1]: Leaving directory '/tmp/build-BbkEBV/linux-6.15.1'
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- -C /tmp/build-BbkEBV/linux-6.15.1 M=/tmp/build-BbkEBV/esphosted/esp_hosted_ng/host modules
make[1]: Entering directory '/tmp/build-BbkEBV/linux-6.15.1'
make[2]: Entering directory '/tmp/build-BbkEBV/esphosted/esp_hosted_ng/host'
  CC [M]  sdio/esp_sdio.o
sdio/esp_sdio.c:8:10: fatal error: utils.h: No such file or directory
    8 | #include "utils.h"
      |          ^~~~~~~~~
compilation terminated.
make[4]: *** [/tmp/build-BbkEBV/linux-6.15.1/scripts/Makefile.build:203: sdio/esp_sdio.o] Error 1
make[3]: *** [/tmp/build-BbkEBV/linux-6.15.1/Makefile:2003: .] Error 2
make[2]: *** [/tmp/build-BbkEBV/linux-6.15.1/Makefile:248: __sub-make] Error 2
make[2]: Leaving directory '/tmp/build-BbkEBV/esphosted/esp_hosted_ng/host'
make[1]: *** [Makefile:248: __sub-make] Error 2
make[1]: Leaving directory '/tmp/build-BbkEBV/linux-6.15.1'
make: *** [Makefile:60: all] Error 2
make: Leaving directory '/tmp/build-BbkEBV/esphosted/esp_hosted_ng/host'
```

mantriyogesh, 2025-06-08:
okay. this patch built for me on older kernels.

Let me check how can I build explicitly on your kernel (6.15.1).

I used the within system kernel compiler (non-cross, native gcc).

Possibly workaround is to append 'include/' at https://github.com/espressif/esp-hosted/blob/0699edf7e4f48028f6d4bd6032418ef7c7395c11/esp_hosted_ng/host/sdio/esp_sdio.c#L8 to change to 
```
#include "include/utils.h"
```
similar to https://github.com/espressif/esp-hosted/blob/0699edf7e4f48028f6d4bd6032418ef7c7395c11/esp_hosted_ng/host/sdio/esp_sdio.c#L23

But let me checkout the 6.15.1 kernel linux and build on vm.

mantriyogesh, 2025-06-09:
After a lengthy kernel build of version 6.15.1, I modified the Makefile to ensure it works, as indicated in the attached patch.

## Build Log with 6.15.1
```
[I] u64@ubuntuarm:/home/u64/tmp/esp_hosted_ng/host{} $ make -C /home/u64/tmp/linux_6.15.x/linux-6.15.1 M=/home/u64/tmp/esp_hosted_ng/host ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- modules

make: Entering directory '/home/u64/tmp/linux_6.15.x/linux-6.15.1'
make[1]: Entering directory '/home/u64/tmp/esp_hosted_ng/host'
  CC [M]  sdio/esp_sdio.o
  CC [M]  sdio/esp_sdio_api.o
  CC [M]  esp_bt.o
  CC [M]  main.o
  CC [M]  esp_cmd.o
  CC [M]  esp_utils.o
  CC [M]  esp_cfg80211.o
  CC [M]  esp_stats.o
  CC [M]  esp_debugfs.o
  CC [M]  esp_log.o
  LD [M]  esp32_sdio.o
  MODPOST Module.symvers
  CC [M]  esp32_sdio.mod.o
  CC [M]  .module-common.o
  LD [M]  esp32_sdio.ko
  BTF [M] esp32_sdio.ko
make[1]: Leaving directory '/home/u64/tmp/esp_hosted_ng/host'
make: Leaving directory '/home/u64/tmp/linux_6.15.x/linux-6.15.1'
[I] u64@ubuntuarm:/home/u64/tmp/esp_hosted_ng/host{} $
```

[586.patch](https://github.com/user-attachments/files/20649394/586.patch)

Please assess the attached patch and let me know your thoughts

mantriyogesh, 2025-06-09:
base commit for patch: 98d9b1565db41b45d89339cd0dc4062690b324ab

mantriyogesh, 2025-06-09:
Please don't hesitate to re-open the issue if you continue to experience problems with the build.
According to our testing, the issue should now be resolved.