espressif/esp-hosted #612: Porting esp-hosted NG SDIO to OpenWRT 24 cannot perform Wi-Fi scan

zhengruiw02, 2025-06-20:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I've studied esp-hosted NG for a time. I can successfully port host to Raspberry Pi 4 with SDIO/SPI works with ESP32-C6 -DevkitC1 V1.2.

When I tried to port host to MT7688 with OpenWRT latest version, I can see wlan0 there with correct MAC address of my ESP32-C6 board. But then I want to use `iw` command to scan Wi-Fi. I expected I can scan some AP and list them.

### Actual behavior (suspected bug)

Use `iw` command to scan Wi-Fi and it failed. Seems after init esp32_sdio module, some SDIO command from host send out is not responding from ESP.

### Error logs or terminal output

```shell
Log from host like below.

root@OpenWrt:~# ls -l
-rwxr-xr-x    1 root     root       2439652 May 19 23:03 esp32_sdio.ko
-rwxr-xr-x    1 root     root               101 May 19 23:03 mtk_sdio_init.sh
root@OpenWrt:~# cat mtk_sdio_init.sh 
modprobe bluetooth;
modprobe cfg80211;
insmod ./esp32_sdio.ko resetpin=6;
sleep 4;
ifconfig wlan0 up
root@OpenWrt:~# ./mtk_sdio_init.sh 
[  109.322815] esp32_sdio: esp_probe: ESP network device detected
[  109.334891] esp32_sdio: get_firmware_data: Rx Pre ====== 0
[  109.345841] esp32_sdio: get_firmware_data: Rx Pos ======  0
[  109.357607] esp32_sdio: get_firmware_data: Tx Pre ======  0
[  109.368740] esp32_sdio: get_firmware_data: Tx Pos ======  10
[  109.393067] esp32_sdio: process_esp_bootup_event: Received ESP bootup event
[  109.406973] esp32_sdio: process_event_esp_bootup: Bootup Event tag: 3
[  109.419767] esp32_sdio: esp_validate_chipset: Chipset=ESP32-C6 ID=0d detected over SDIO
[  109.435658] esp32_sdio: process_event_esp_bootup: Bootup Event tag: 0
[  109.448437] esp32_sdio: process_event_esp_bootup: Bootup Event tag: 1
[  109.461211] esp32_sdio: process_fw_data: ESP chipset's last reset cause:
[  109.474503] esp32_sdio: print_reset_reason: POWERON_RESET
[  109.485241] esp32_sdio: check_esp_version: ESP-Hosted Version: NG-1.0.4.0.2
[  109.504733] ------------[ cut here ]------------
[  109.513989] WARNING: CPU: 0 PID: 6 at backports-6.12.6/net/wireless/util.c:233 0x82c07028 [cfg80211@ec7b2600+0x56000]
[  109.535095] Modules linked in: esp32_sdio(O) pppoe ppp_async nft_fib_inet nf_flow_table_inet rtl8xxxu(O) pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mmc_spi mac80211(O) cfg80211(O) ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda slhc rfcomm of_mmc_spi nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c hidp crc7 crc_itu_t crc_ccitt compat(O) btsdio bnep bluetooth ntfs3 hid evdev input_core nls_utf8 nls_iso8859_1 nls_cp437 rfkill ecdh_generic ecc sha512_generic seqiv sha3_generic jitterentropy_rng drbg kpp hmac geniv rng ecb cmac uas mtk_sd mmc_block cqhci usb_storage sdhci_pltfm sdhci mmc_core leds_gpio ohci_platform ohci_hcd fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod scsi_common
[  109.535965]  ehci_hcd gpio_button_hotplug(O) vfat fat exfat usbcore nls_base usb_common crc16 crc32c_generic
[  109.733479] CPU: 0 PID: 6 Comm: kworker/0:0H Tainted: G           O       6.6.90 #0
[  109.748660] Workqueue: ESP_EVENTS_WORKQUEUE 0x843a211c [esp32_sdio@51e44361+0xa000]
[  109.763850] Stack : 8189fc5c 00000000 841e0000 81838a00 00000000 00000000 00000000 00000000
[  109.780449]         00000000 00000000 00000000 00000000 00000000 00000001 8189fc20 1a618f01
[  109.797042]         8189fcb8 00000000 00000000 8189fb40 00000038 804c23a4 00000000 0000000b
[  109.813638]         64735f32 25692ae8 4b77a428 35406f69 804de66c 804de66c 00000009 00000000
[  109.830235]         00000000 000000e9 841e0000 841d83f0 00000000 802ae7ec 00000000 80790000
[  109.846832]         ...
[  109.851674] Call Trace:
[  109.851693] [<804c23a4>] 0x804c23a4
[  109.863429] [<802ae7ec>] 0x802ae7ec
[  109.870339] [<80006928>] 0x80006928
[  109.877248] [<80006930>] 0x80006930
[  109.884157] [<804a26fc>] 0x804a26fc
[  109.891067] [<82c07028>] 0x82c07028 [cfg80211@ec7b2600+0x56000]
[  109.902804] [<82c07028>] 0x82c07028 [cfg80211@ec7b2600+0x56000]
[  109.914541] [<80020c0c>] 0x80020c0c
[  109.921452] [<82c07028>] 0x82c07028 [cfg80211@ec7b2600+0x56000]
[... 274 lines cut ...]
8. Flash ESP32-C6 with this firmware.

### Project release version

v1.0.4.0.2

### System architecture

other (details in Additional context)

### Operating system

Linux

### Operating system version

OpenWRT 24

### Shell

Bash

### Additional context

Host board ARCH is MIPS

mantriyogesh, 2025-06-25:
@kapilkedawat @Shreyas0-7 PTAL

zhengruiw02, 2025-06-26:
Hi @kapilkedawat , I followed porting guide.

My MediaTek host development board have SDIO interface with DAT0/DAT1/DAT2/DAT3/CLK 10k ohm pull-up resistors. And it comes with Micro-SD slot. I tested it can mount my SD card.
We use solder wiring and try to make all connection shorten and same length.
I also use external power source to supply on 3.3V/5V both my MediaTek host development board and ESP32-C6-DevkitC1.

[image]

I will try to run raw_throughput to see result.

zhengruiw02, 2025-06-26:
Hi @kapilkedawat , I've ran throughput test, and found as below"
1. From ESP to host, no data rate.
2. From host to ESP, have little in beginning then also no date rate later.

Log as below.

ESP to Host log.
```
root@OpenWrt:~# modprobe bluetooth;
root@OpenWrt:~# modprobe cfg80211
root@OpenWrt:~# insmod ./esp32_sdio.ko resetpin=2 raw_tp_mode=1
root@OpenWrt:~# rmmod esp32_sdio.ko 
root@OpenWrt:~# insmod ./esp32_sdio.ko resetpin=2 raw_tp_mode=2
root@OpenWrt:~# [  274.061012] mmc0: queuing unknown CIS tuple 0x01 [d9 01 ff] (3 bytes)
[  274.080900] mmc0: queuing unknown CIS tuple 0x1a [01 01 00 02 07] (5 bytes)
[  274.098775] mmc0: queuing unknown CIS tuple 0x1b [c1 41 30 30 ff ff ff ff] (8 bytes)
[  274.117572] mmc0: new high speed SDIO card at address 0001
[  274.133042] esp32_sdio: esp_probe: ESP network device detected
[  274.145171] esp32_sdio: get_firmware_data: Rx Pre ====== 0
[  274.156126] esp32_sdio: get_firmware_data: Rx Pos ======  0
[  274.167890] esp32_sdio: get_firmware_data: Tx Pre ======  0
[  274.179019] esp32_sdio: get_firmware_data: Tx Pos ======  10
[  274.202858] esp32_sdio: process_esp_bootup_event: Received ESP bootup event
[  274.216756] esp32_sdio: process_event_esp_bootup: Bootup Event tag: 3
[  274.229550] esp32_sdio: esp_validate_chipset: Chipset=ESP32-C6 ID=0d detected over SDIO
[  274.245445] esp32_sdio: process_event_esp_bootup: Bootup Event tag: 0
[  274.258236] esp32_sdio: process_event_esp_bootup: Bootup Event tag: 1
[  274.271028] esp32_sdio: process_fw_data: ESP chipset's last reset cause:
[  274.284323] esp32_sdio: print_reset_reason: POWERON_RESET
[  274.295043] esp32_sdio: check_esp_version: ESP-Hosted Version: NG-1.0.4.0.2
[  274.312406] ------------[ cut here ]------------
[  274.321657] WARNING: CPU: 0 PID: 6 at backports-6.12.6/net/wireless/util.c:233 0x82c07028 [cfg80211@305dc425+0x56000]
[  274.342760] Modules linked in: esp32_sdio(O) pppoe ppp_async nft_fib_inet nf_flow_table_inet rtl8xxxu(O) pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mmc_spi mac80211(O) cfg80211(O) ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda slhc rfcomm of_mmc_spi nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c hidp crc7 crc_itu_t crc_ccitt compat(O) btsdio bnep bluetooth ntfs3 hid evdev input_core nls_utf8 nls_iso8859_1 nls_cp437 rfkill ecdh_generic ecc sha512_generic seqiv sha3_generic jitterentropy_rng drbg kpp hmac geniv rng ecb cmac uas mtk_sd mmc_block cqhci usb_storage sdhci_pltfm sdhci mmc_core leds_gpio ohci_platform ohci_hcd fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod scsi_common
[  274.343630]  ehci_hcd gpio_button_hotplug(O) vfat fat exfat usbcore nls_base usb_common crc16 crc32c_generic [last unloaded: esp32_sdio(O)]
[  274.546490] CPU: 0 PID: 6 Comm: kworker/0:0H Tainted: G           O       6.6.90 #0
[  274.561669] Workqueue: ESP_EVENTS_WORKQUEUE 0x843a211c [esp32_sdio@b0281bf0+0xa000]
[  274.576860] Stack : 8189fc5c 00000000 84390000 81838100 00000000 00000000 00000000 00000000
[  274.593457]         00000000 00000000 00000000 00000000 00000000 00000001 8189fc20 b262660d
[  274.610052]         8189fcb8 00000000 00000000 8189fb40 00000038 804c23a4 00000000 00000008
[  274.626648]         64735f32 36b8122e 54302d19 62406f69 804de66c 804de66c 00000009 00000000
[  274.643245]         00000000 000000e9 84390000 843883f0 00000000 802ae7ec 00000000 80790000
[  274.659841]         ...
[  274.664684] Call Trace:
[  274.664703] [<804c23a4>] 0x804c23a4
[  274.676440] [<802ae7ec>] 0x802ae7ec
[  274.683350] [<80006928>] 0x80006928
[  274.690259] [<80006930>] 0x80006930
[  274.697168] [<804a26fc>] 0x804a26fc
[  274.704077] [<82c07028>] 0x82c07028 [cfg80211@305dc425+0x56000]
[  274.715816] [<82c07028>] 0x82c07028 [cfg80211@305dc425+0x56000]
[  274.727552] [<80020c0c>] 0x80020c0c
[  274.734463] [<82c07028>] 0x82c07028 [cfg80211@305dc425+0x56000]
[  274.746198] [<804a6494>] 0x804a6494
[  274.753108] [<82c07028>] 0x82c07028 [cfg80211@305dc425+0x56000]
[  274.764845] [<80020cec>] 0x80020cec
[  274.771761] [<82c07028>] 0x82c07028 [cfg80211@305dc425+0x56000]
[  274.783503] [<8298e7a4>] 0x8298e7a4 [rfkill@bcd29053+0x2000]
[  274.794728] [<82c02568>] 0x82c02568 [cfg80211@305dc425+0x56000]
[  274.806466] [<82c00da0>] 0x82c00da0 [cfg80211@305dc425+0x56000]
[  274.818212] [<843a8c64>] 0x843a8c64 [esp32_sdio@b0281bf0+0xa000]
[  274.830124] [<843a2074>] 0x843a2074 [esp32_sdio@b0281bf0+0xa000]
[  274.842040] [<843a22a8>] 0x843a22a8 [esp32_sdio@b0281bf0+0xa000]
[  274.853967] [<80037944>] 0x80037944
[  274.860876] [<8003c748>] 0x8003c748
[  274.867787] [<8003c748>] 0x8003c748
[  274.874700] [<80037fac>] 0x80037fac
[  274.881609] [<804c48ac>] 0x804c48ac
[  274.888517] [<80037cd8>] 0x80037cd8
[  274.895427] [<8003df00>] 0x8003df00
[  274.902336] [<8003de24>] 0x8003de24
[... 162 lines cut ...]
[ 3866.392455] esp32_sdio: log_raw_tp_stats_timer_cb: 10-11 sec       0 kbits/sec
[ 3866.392455] 
[ 3867.432456] esp32_sdio: log_raw_tp_stats_timer_cb: 11-12 sec       0 kbits/sec
[ 3867.432456] 
[ 3868.472455] esp32_sdio: log_raw_tp_stats_timer_cb: 12-13 sec       0 kbits/sec
[ 3868.472455] 
[... cut ...]
[ 3872.632455] 
[ 3873.672455] esp32_sdio: log_raw_tp_stats_timer_cb: 17-18 sec       0 kbits/sec
[ 3873.672455] 
[ 3874.712455] esp32_sdio: log_raw_tp_stats_timer_cb: 18-19 sec       0 kbits/sec
[ 3874.712455] 
[ 3875.752455] esp32_sdio: log_raw_tp_stats_timer_cb: 19-20 sec       0 kbits/sec
[ 3875.752455] 
[ 3876.792456] esp32_sdio: log_raw_tp_stats_timer_cb: 20-21 sec       0 kbits/sec
[ 3876.792456] 
[ 3877.832455] esp32_sdio: log_raw_tp_stats_timer_cb: 21-22 sec       0 kbits/sec
[ 3877.832455] 
```

mantriyogesh, 2025-06-26:
The Event WQ is crashing, which prevents the state transitions from enabling the SDIO lines, rendering them unusable. Based on the experiment, I see two potential issues:

1. The crashing of the Event WQ is blocking the state transitions necessary for activating the SDIO lines. We need to determine why the event queue fails specifically on your CPU/MPU.

2. The mysterious -145 error. It's unclear where or how this error arises. For standard functions like sdio_memcpy_toio or sdio_io_rw_extended, -145 is not a common error I recognize.

Before diving into extensive debugging, I recommend checking a few hardware aspects:
- Ensure the input voltage is sufficient for the Linux MPU and that the MCU is not under heavy load.
- Verify that the SDIO lines are receiving adequate voltage (3.3V).
- Confirm that you are applying 5V to C6 through the correct input pin. Alternatively, you might consider powering the C6 manually using the standard USB UART port.
- Test C6 independently with a simple WiFi station example, such as the one found at https://github.com/espressif/esp-idf/tree/release/v5.5/examples/wifi/getting_started/station, to see if the ESP32-C6 functions correctly in isolation.

I am uncertain about the origin of the -145 error, as it does not seem to occur in standard SDIO operations.

mantriyogesh, 2025-07-07:
---
@zhengruiw02 ,

> tx_process: Failed to send data: -84 512 512

The -84 and -110 errors are quite common when using jumper wires in your setup. These errors typically indicate that the SDIO driver is either not getting properly probed or is experiencing instability due to hardware-related issues.

Because of this, the communication may work intermittently — sometimes succeeding, sometimes failing — due to errors like [-84](https://elixir.bootlin.com/u-boot/v2025.01/source/include/linux/errno.h#L92) or [-110](https://elixir.bootlin.com/u-boot/v2025.01/source/include/linux/errno.h#L132).

While I'm fairly confident this is the root cause, it would still be helpful to check the slave logs for additional clues at the same time.

Have you had a chance to verify the points mentioned in [this comment](https://github.com/espressif/esp-hosted/issues/612#issuecomment-3007117709)?
It's essential to double-check the hardware recommendations there — skipping them might lead to hard-to-debug failures later on.

---
@kapilkedawat (Espressif)
> root@OpenWrt:~# [   59.524868] esp32_sdio: wait_and_decode_cmd_resp: Command[0x1] timed out
> [   59.531583] esp32_sdio: cmd_init_interface: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22
> [   59.540910] esp32_sdio: esp_add_card: esp_add_network_ifaces(adapter) failure, ret: -1
> [   59.548860] Unable to handle kernel read from unreadable memory at virtual address 0000000000000000
> [   59.557915] Mem abort info:

As a consequence of the above issue, we’re encountering a decoding of NULL, likely due to the instability it causes. Adding a few safeguards or sanity checks would help improve robustness, perhaps including logic to gracefully unload the driver if it fails to initialize correctly.

mantriyogesh, 2025-07-07:
@zhengruiw02, If you cannot (at all) omit jumper wires and not move to PCB, we strongly suggest testing the **SDIO 1-bit mode**, instead of the 4-bit mode, as an evaluation alternative.

Please let us know if you wish to see the SDIO schematic we use as PCB between the Raspberry Pi and ESP. I can share with you my mail ID, if you wish to cross-check.

mantriyogesh, 2025-07-25:
@kapilkedawat PTAL

Shreyas0-7, 2025-07-28:
Hey @zhengruiw02 
can you please check if your host source code has these values defined correctly
static struct ieee80211_rate esp_rates[] = {
	{.bitrate = 10, .hw_value = WIFI_PHY_RATE_1M_L, },
	{.bitrate = 20, .hw_value = WIFI_PHY_RATE_2M_L, },
	{.bitrate = 55, .hw_value = WIFI_PHY_RATE_5M_L, .hw_value_short = WIFI_PHY_RATE_5M_S},
	{.bitrate = 110, .hw_value = WIFI_PHY_RATE_11M_L, .hw_value_short = WIFI_PHY_RATE_11M_S},
	{.bitrate = 60, .hw_value = WIFI_PHY_RATE_6M, },
	{.bitrate = 90, .hw_value = WIFI_PHY_RATE_9M, },
	{.bitrate = 120, .hw_value = WIFI_PHY_RATE_12M, },
	{.bitrate = 180, .hw_value = WIFI_PHY_RATE_18M, },
	{.bitrate = 240, .hw_value = WIFI_PHY_RATE_24M, },
	{.bitrate = 360, .hw_value = WIFI_PHY_RATE_36M, },
	{.bitrate = 480, .hw_value = WIFI_PHY_RATE_48M, },
	{.bitrate = 540, .hw_value = WIFI_PHY_RATE_54M, },
};

zhengruiw02, 2025-07-29:
> Hey [@zhengruiw02](https://github.com/zhengruiw02) can you please check if your host source code has these values defined correctly static struct ieee80211_rate esp_rates[] = { {.bitrate = 10, [...]

Hi Shreyas, thanks for your reply.
I've checked those code in "esp_hosted_ng/host/esp_cfg80211.c"  are untouched as you mentioned. This also strange thing. I'm going to debug from mac80211 of OpenWRT package to see what happen there. ( Source is http://mirror2.openwrt.org/sources/backports-6.12.6.tar.xz).

Compare to MT7688 host, on RPi4 with OpenWRT same mac80211 package does not have this warning.

Shreyas0-7, 2025-07-30:
Hey @zhengruiw02 
Have you made any changes on platform/kernel can you please check if there is any updated package available from vendor?

zhengruiw02, 2025-07-31:
Hi @Shreyas0-7 
I also trying on this to update package and build again.

BTW I have more investigation on it, and located issue comes in `esp_add_wiphy` function.
I alloc new memory and memcpy `esp_rate` structure to this new memory, then pass it to `esp_wifi_bands_2ghz.bitrates` and it works. Then I revert code, and adding more debug prints, found that root cause is after `wiphy_register(wiphy)` called. Seems data is corrupted after it called. If did memcpy to new memory can help on this.

I'm compairing to other Wi-Fi drivers on cfg80211 part, for init/create. Found boardcom driver would like to use `kmemdup` for cfg80211_ops and also for `wiphy->bands` struct.  Refer to [this](https://github.com/torvalds/linux/blob/master/drivers/net/wireless/broadcom/brcm80211/brcmfmac/cfg80211.c#L7711)

After I use memcpy to workaround as above, still have other issue when driver trying to connecting to SSID.  When calling `esp_cfg80211_scan` have NULL pointer on `*request->wdev`. Not sure related to memory corrupted or not. I want to try to use `kmemdup` for the driver init first.

zhengruiw02, 2025-08-11:
Hi all,

After some tries, I made ESP-hosted host code as a patch and apply it to OpenWRT package of mac80211-backports, also modified Makefile to acitvate the build of ESP SDIO driver, the insmod assert messange gone, and I can connect to AP with ESP32-C6 on my host platform now. Thanks for your helping.

Close this ticket since solved.

closed 2025-08-11 by zhengruiw02

mantriyogesh, 2025-08-11:
Hi @zhengruiw02,

Would you mind sharing the changes you made to adapt this for your host environment, along with the relevant host details? This would be very helpful for other team members who might encounter similar situations.

We'd also appreciate the opportunity to review these changes to determine if any updates should be incorporated into our main codebase.

Thank you for your time and efforts!

zhengruiw02, 2025-08-12:
OK, let me simplify it as below steps.

Background: OpenWRT is using [Linux backports project](https://backports.docs.kernel.org/) on wireless drivers, which means not using mainstream kernel mac80211 library.

1. Verify your [OpenWRT](https://github.com/openwrt/openwrt) build system. Make sure you can build and run firmware without ESP-hosted.
2. In your OpenWRT build system, find out which backport source version is using, by checking at [`package/kernel/mac80211/Makefile`](https://github.com/openwrt/openwrt/tree/main/package/kernel/mac80211/Makefile). Then download it for further test and make diff use.
For example below, you can download your source from [this](http://mirror2.openwrt.org/sources/backports-6.14.11.tar.xz)
```
PKG_NAME:=mac80211

PKG_VERSION:=6.14.11
PKG_RELEASE:=1
PKG_LICENSE:=GPL-2.0-only
PKG_LICENSE_FILES:=COPYING

PKG_SOURCE_URL:=http://mirror2.openwrt.org/sources/
PKG_HASH:=1f6620485f0b33046e87aa2c6ac04ae529e9b5e83e318652d89939b192c6e348

PKG_SOURCE:=backports-$(PKG_VERSION).tar.xz
```
4. Move ESP-hosted-NG code in, according to other wireless drivers. Then you make diff patch from your changes. I will attach a diff file for example.
5. Move this patch file to OpenWRT project `package/kernel/mac80211/patches/build`, and rename like `999-mt7688-esp32.patch` to make sure it will be patch at last.
6. Modifiy `package/kernel/mac80211/Makefile` to make sure build of esp-hosted driver. From my patch file, I defined `WLAN_VENDOR_ESP` and `ESP_HOSTED_NG` to build the module, so add those two defines in `config-y` variable.
7. Test the integration by `make package/kernel/mac80211/{prepare,compile}  -j${JOBS} V=99 ` command with verbose, which will apply patches and compile the mac80211 backport with drivers.
8. You should find `esp32_sdio.ko` from `build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/mac80211-regular/backports-6.14.11/drivers/net/wireless/esp/host/esp32_sdio.ko` some path like this.

[999-mt7688-esp32.patch](https://github.com/user-attachments/files/21725276/999-mt7688-esp32.patch)