espressif/esp-hosted #620: Can connect to iPhone hotspot only if option “Optimize compatibility” is enabled

giuliobenetti, 2025-07-01:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

It should connect without enabling that option possibly.

### Actual behavior (suspected bug)

Need to enable Optimize option

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

Setup an iPhone as hotspot and don’t enable Optimize compatibility option

### Project release version

Latest

### System architecture

ARM 64-bit (Apple M1/M2, Raspberry Pi 4/5)

### Operating system

Linux

### Operating system version

Buildroot BSP

### Shell

sh

mantriyogesh, 2025-07-01:
Yes, if you're using an iPhone hotspot, please ensure "Maximize Compatibility" is enabled in your iPhone’s hotspot settings. This setting forces the hotspot to:
- Use 2.4 GHz only (ESP32 is typically limited to 2.4 GHz)
- Disable Wi-Fi 6 (802.11ax)
- Use 802.11 b/g/n modes that are compatible with embedded Wi-Fi devices

Without this option, the iPhone may switch to 5 GHz or Wi-Fi 6, which some ESP slave cannot connect to, if limited by their Wi-Fi capabilities.

May we know the slave ESP chip used?

giuliobenetti, 2025-07-02:
@Shreyas0-7 can you please test the throughput at 5Ghz and 2.4Ghz? Thanks in advance

giuliobenetti, 2025-07-03:
> Hey [@giuliobenetti](https://github.com/giuliobenetti) we are doing some internal testing right now, will update throughput soon

Thanks a lot. I’m going to setup an embedded Linux board(RK3562) with have with C6 using C5 with very short lines to keep a decent Signal Integrity and test as well.
Then let you know the results too.

giuliobenetti, 2025-10-06:
I finally found time to put my hands back on this. Unfortunately when I try to flash with this command:
`
esptool.py --chip esp32c5 --force --port /dev/ttyS2 --baud 460800 write_flash \
    0x0 /lib/firmware/bootloader.bin \
    0x8000 /lib/firmware/partition-table.bin \
    0x10000 /lib/firmware/network_adapter.bin
`

I get this result:
`esptool.py v4.10.0
Serial port /dev/ttyS2
Connecting...
Failed to get PID of a device on /dev/ttyS2, using standard reset sequence.
.
Chip is ESP32-C5 (revision v1.0)
Features: WiFi 6, BT 5, IEEE802.15.4
Crystal is 48MHz
MAC: 30:ed:a0:ff:fe:e4:4f:c4
BASE MAC: 30:ed:a0:e4:4f:c4
MAC_EXT: ff:fe
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Unexpected chip id in image. Expected 23 but value was 0. Is this image for a different chip model?

A fatal error occurred: /lib/firmware/bootloader.bin is not an ESP32-C5 image. Use --force to flash anyway.`

Is there a fix or a workaround? I'm using esptool 4.10 and --force is not an option even if pointed in log.

mantriyogesh, 2025-10-07:
The command generated after idf.py build for me, for esp32c5 after following

```
rm -rf managed_components/ dependencies.lock sdkconfig build/ sdkconfig.old; idf.py set-target esp32c5
idf.py build
```

is

```
python -m esptool --chip esp32c5 -b 2000000 \
  --before default_reset --after hard_reset \
  write_flash --flash_mode dio --flash_size 4MB --flash_freq 80m \
  0x2000  build/bootloader/bootloader.bin \
  0x8000  build/partition_table/partition-table.bin \
  0xd000  build/ota_data_initial.bin \
  0x10000 build/esp_hosted_mcu_wifi.bin

```

Never use force.

Some options in this command even if not mentioned, would be auto populated by idf.py.

Your effective command after changes could be:
```
esptool.py --chip esp32c5 --port /dev/ttyS2 -b 460800 \
  write_flash 0x2000 /lib/firmware/bootloader.bin \
               0x8000 /lib/firmware/partition-table.bin \
               0x10000 /lib/firmware/network_adapter.bin
```

There is major diff that you are pushing the bootloader on 0x0. 
Please ignore partitions that you do not have and change the partition binary names as per your application name.

giuliobenetti, 2025-10-07:
@mantriyogesh thank you. I've programmed it successfully after applying this patch pending:
https://github.com/espressif/esp-hosted/pull/669

Unfortunately I have SI issues because of wires too long and crossed. So I'm going to setup an A20 board with 2.54 pins and use Jumper Wires. Then I come back to you.

giuliobenetti, 2025-10-15:
Nothing to do neither with wires less than 5cm and far from each other. All 47K pull-ups are there, 22R series for CLK.
It begins to work only at 5Mhz with 1-wire but when it tries with wpa_supplicant to connect to either 2.4 or 5Ghz network it fails with error 2.

I proceed remastering to be sure to have correct SIignal Integrity.
If you @mantriyogesh have any other ideas to give a fast try please let me know.
To do this try I had to setup an Olinuxino A20 micro using SDC1 port. Unfortunately all pins are grouped together and no GND pin is close. The best would have been to have GND-CLK-GND-D0-GND-D1-... and so on, but nothing to do.

At this point I trust your ESP32C5-MINI-1 in place of C6-MINI-1 module and remaster carefully.

giuliobenetti, 2025-10-16:
@mantriyogesh While taking picture you see below I've realized than GND return path and 3V3 rail are still too long.
I'm going to shorten 3 wires:
- 3V3
- GND
- 3V3 for pull-ups
the more I can.

In case yes I can program an esp32c5 pogram test. Which FW do you suggest me to build, flash and try?

Thanks a lot for suppot

[image]
[image]
[image]

mantriyogesh, 2025-10-17:
> They are, CMD, DAT0-1-2-3 all have 47K. For C6 and other models 51K was pointed while for C5 10K seems to be
the right value. Would it make sense to change from 47K to 10K? At 5Mhz I don't see the point that much, but let me
know if it makes sense for you and also I'm curious for the reason of 10K vs 51K.

10K is also sufficient. In our all schematics, every Hosted based PCB, the hardware team has used 51K Ohm.
I am not sure from hardware perspective, but all i got to know is it should not much matter.

If you feel, i should get to bottom of (from my experience, as stated, shouldn't matter much), please let me know. i will check with hardware team.

Sorry, I did not understand the intention of `mmc1` node device tree config.

From your log, 
```
[   26.249364] esp32_sdio mmc0:0001:2: probe with driver esp32_sdio failed with error -22
```
it looked like you are using mmc0 for hosted sdio. please let me know if not the case?

mantriyogesh, 2025-10-22:
@kapilkedawat Please take a look.

@giuliobenetti , Can you please provide the firmware through https://wetransfer.com/ or something?
Any definitive steps to easily reproduce the issue?

We would try with Raspberry Pi 4 or 5 as Linux our side..

Shreyas0-7, 2025-10-24:
Hey @giuliobenetti 
I went through logs provided by you it seems like we are able to get till association but AP is not sending M1 eapol message. We might need wireshark capture to debug this furthermore.
Also I tried ng with my own iPhone and I was able to connect with `Maximise Compatibility` on as well off.
This is conf and configuration which I used
- command: `wpa_supplicant -c supp.conf -i wlan0 -dd`
- configuration
```
ctrl_interface=/var/run/wpa_supplicant
network={
	ssid="iPhone" //YOUR SSID
		psk="passphrase" //YOUR PASSWORD
		key_mgmt=WPA-PSK
		ieee80211w=1
}
```