espressif/esp-hosted #622: The bluetoothctl tool of Bluez is unable to execute commands such as scanning and connection

FrankWT, 2025-07-04:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

The Bluetooth function can be tested using the "bluetoothctl" tool

### Actual behavior (suspected bug)

After loading the esp32_sdio.ko driver and executing bluetooth, if the bluetoothctl tool is executed next, it will get stuck and unable to execute subsequent commands. However, when using the hcitool lescan and btmgmt tools for testing, the device's Bluetooth can scan and broadcast.

Executing "bluetoothctl -v" is normal. The version of bluetoothctl is 5.54.

### Error logs or terminal output

```shell
Zeratul login: 
Zeratul login: 
Zeratul login: root
Mar  1 12:15:20 login[437]: root login on 'console'
Hello Zeratul!
[root@Zeratul:~]# insmod_mmc 
[root@Zeratul:~]# [  922.968292] mmc1: error ocr is 0x20ffff00.

[root@Zeratul:~]# insmod /system/lib/modules/esp32_sdio.ko 
[root@Zeratul:~]# 
[root@Zeratul:~]# 
[root@Zeratul:~]# hciconfig -a
hci0:	Type: Primary  Bus: SDIO
	BD Address: A0:85:E3:B0:85:1A  ACL MTU: 517:10  SCO MTU: 0:0
	DOWN 
	RX bytes:209 acl:0 sco:0 events:30 errors:0
	TX bytes:61 acl:0 sco:0 commands:0 errors:0
	Features: 0x00 0x00 0x00 0x00 0x60 0x00 0x00 0x00
	Packet type: DM1 DH1 HV1 
	Link policy: 
	Link mode: SLAVE ACCEPT 

[root@Zeratul:~]# 
[root@Zeratul:~]# 
[root@Zeratul:~]# dbus-daemon --system --fork
dbus-daemon[496]: Failed to start message bus: Failed to bind socket "/var/run/dbus/system_bus_socket": No such file or directory
[root@Zeratul:~]# mkdir /var/run/dbus
[root@Zeratul:~]# dbus-daemon --system --fork
[root@Zeratul:~]# export DBUS_SYSTEM_BUS_ADDRESS=unix:path=/var/run/dbus/system_
bus_socket
[root@Zeratul:~]# 
[root@Zeratul:~]# 
[root@Zeratul:~]# bluetoothd --nodetach --debug &
[root@Zeratul:~]# bluetoothd[500]: Bluetooth daemon 5.54
bluetoothd[500]: src/main.c:parse_config() parsing /etc/bluetooth/main.conf
bluetoothd[500]: src/main.c:parse_config() Key file does not have key 'DiscoverableTimeout'
bluetoothd[500]: src/main.c:parse_config() Key file does not have key 'AlwaysPairable'
bluetoothd[500]: src/main.c:parse_config() Key file does not have key 'PairableTimeout'
bluetoothd[500]: src/main.c:parse_config() Key file does not have key 'Privacy'
bluetoothd[500]: src/main.c:parse_config() Key file does not have key 'JustWorksRepairing'
bluetoothd[500]: src/main.c:parse_config() Key file does not have key 'Name'
bluetoothd[500]: src/main.c:parse_config() Key file does not have key 'Class'
bluetoothd[500]: src/main.c:parse_config() Key file does not have key 'DeviceID'
bluetoothd[500]: src/main.c:parse_config() Key file does not have key 'ReverseServiceDiscovery'
bluetoothd[500]: src/main.c:parse_config() Key file does not have key 'Cache'
bluetoothd[500]: src/main.c:parse_config() Key file does not have key 'KeySize'
bluetoothd[500]: src/main.c:parse_config() Key file does not have key 'ExchangeMTU'
[... 140 lines cut ...]
none

### Project release version

master

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Ubuntu 18.04.6 LTS

### Shell

sh

### Additional context

[dmesg_log.txt](https://github.com/user-attachments/files/21049077/dmesg_log.txt)

FrankWT, 2025-07-11:
@mantriyogesh The bluetoothctl test for Bluetooth still doesn't work. We tested it with btmgmt. The phone can connect to the device's Bluetooth, but it will disconnect very soon. What could be the problem?

The following is the log of btmgmt when the mobile phone is connected to Bluetooth
[mgmt]#                    hci0 57:F3:78:6C:17:30 type LE Random connected eir_len 0
[mgmt]#                    hci0 57:F3:78:6C:17:30 type LE Random disconnected with reason 2

mantriyogesh, 2025-07-11:
Actually, the ESP-Hosted role is only to transport the bluetooth messages reliably.
Sorry to state this, but, we as the ESP-hosted team are not Bluetooth experts as such.
 
You can check the details using `sudo btmon`.
We also run into issues sometimes, with iPhone, due to some very strict device-level management.

Can you please verify the GATT server case on Linux and use nrfConnect as gatt client on an Android device?

If the Android device works, you might check what the problem is with the iPhone later.

mantriyogesh, 2025-07-11:
I had tested this, and it worked fine for Android, btw.