espressif/esp-hosted #623: ESP32C5 SDIO/SPI and UART Pinout

giuliobenetti, 2025-07-04:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

Add to the manual ESP32C5 SDIO/SPI and UART Pinout.

Is it simply the same as ESP32C6?

### Use cases

I'm going to swap C6 with C5 because of Dual band and #620 issue

mantriyogesh, 2025-07-08:
@Shreyas0-7, please add the doc or point the pinout to the public beta c5 branch.

Shreyas0-7, 2025-07-08:
@mantriyogesh Yes will do
@giuliobenetti you can find documentation for c5 here [C5 ng beta](https://github.com/espressif/esp-hosted/tree/feat/esp32c5_ng_beta_support/esp_hosted_ng)