espressif/esp-hosted #624: Extremely low speed on SDIO on Linux

angelo-zehnder, 2025-07-07:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Everything is configured correctly, the driver is loading and firmware on the esp32 module is configured correctly.
```
[  100.482706] esp32_sdio: esp_probe: ESP network device detected
[  100.491379] esp32_sdio: get_firmware_data: Rx Pre ====== 0
[  100.497098] esp32_sdio: get_firmware_data: Rx Pos ======  0
[  100.501400] esp32_sdio: get_firmware_data: Tx Pre ======  0
[  100.508341] esp32_sdio: get_firmware_data: Tx Pos ======  10
[  100.525053] esp32_sdio: probe of mmc0:0001:2 failed with error -22
[  100.532911] esp32_sdio: process_esp_bootup_event: Received ESP bootup event
[  100.538536] esp32_sdio: process_event_esp_bootup: Bootup Event tag: 3
[  100.546433] esp32_sdio: esp_validate_chipset: Chipset=ESP32 ID=00 detected over SDIO
[  100.552949] esp32_sdio: process_event_esp_bootup: Bootup Event tag: 0
[  100.559247] esp32_sdio: process_event_esp_bootup: Bootup Event tag: 1
[  100.565730] esp32_sdio: process_fw_data: ESP chipset's last reset cause:
[  100.572321] esp32_sdio: print_reset_reason: POWERON_RESET
[  100.577748] esp32_sdio: check_esp_version: ESP-Hosted Version: NG-1.0.4.0.2
[  100.588141] esp32_sdio: esp_reg_notifier: Driver init is ongoing
[  100.982437] esp32_sdio: init_bt: ESP Bluetooth init
[  100.996587] esp32_sdio: print_capabilities: Capabilities: 0x1d. Features supported are:
[  101.003428] esp32_sdio: print_capabilities:   * WLAN on SDIO
[  101.008815] esp32_sdio: print_capabilities:   * BT/BLE
[  101.014151] esp32_sdio: print_capabilities:     - HCI over SDIO
[  101.019937] esp32_sdio: print_capabilities:     - BT/BLE dual mode
```

### Actual behavior (suspected bug)

The problem is that the device presents a very low speed, the bus is configured at 25mhz.
```
cat /sys/kernel/debug/mmc0/ios 
clock:          25000000 Hz
actual clock:   25000000 Hz
vdd:            21 (3.3 ~ 3.4 V)
bus mode:       2 (push-pull)
chip select:    0 (don't care)
power mode:     2 (on)
bus width:      0 (1 bits)
timing spec:    0 (legacy)
signal voltage: 0 (3.30 V)
driver type:    0 (driver type B)
```
Basically, the speed is around 2mbits, no errors on the dmesg, everything seems to work fine.
```
curl http://mirror.bakertelekom.fr/Ubuntu/25.04/ubuntu-25.04-netboot-amd64.tar.gz --output ubuntu-25.04-netboot-amd64.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100 91.7M  100 91.7M    0     0   170k      0  0:09:11  0:09:11 --:--:--  172k
```

Any suggestion?!

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

The behavior is always present, the steps to reproduce is simply to load the driver and making it working.

### Project release version

NG-1.0.4.0.2

### System architecture

other (details in Additional context)

### Operating system

Linux

### Operating system version

Yocto distro based on Linux 5.15.145

### Shell

sh

angelo-zehnder, 2025-07-08:
@SohKamYung-Espressif Bus was 1 bit for testing. Enabling 4 bit mode is exactly the same.

@mantriyogesh My bus is running at 25mhz, you have to agree with me it could not be the signal integrity: at those speed a small problem in signal integrity will make the bus going crazy and not working at all.

I'm saying that everything is looking right from the external pov, but the product is extremely slow. What options I have to debug in a way I can understand what's going on?

SohKamYung-Espressif, 2025-07-08:
@angelo-zehnder One other thing I can think of is your SDIO hardware DMA is not enabled and data transfer is being done by software.

Can you check your CPU load (using `top`, for example) when doing data transfers and see if the esp-hosted driver is using up a lot of CPU cycles (maybe 100%)?

You may also want to connect a logic analyser to the SDIO pins to see if it is running at 25 MHz.

angelo-zehnder, 2025-07-08:
@SohKamYung-Espressif The problem was that I compiled the firmware with a too high loglevel, therefore the firmware was spending time print debug messages. Once I moved to a default loglevel problem solved by itself.
Thank you for your support!