espressif/esp-hosted #626: Driver loaded only after mmc controller unbinding

angelo-zehnder, 2025-07-08:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

esp32_sdio driver will be loaded at boot.

### Actual behavior (suspected bug)

```
# modprobe esp32_sdio resetpin=48
root@ccproa1r:~# echo "58005000.mmc" > /sys/bus/amba/drivers/mmci-pl18x/unbind
root@ccproa1r:~# echo "58005000.mmc" > /sys/bus/amba/drivers/mmci-pl18x/bind
[  102.390023] mmci-pl18x 58005000.mmc: allocated mmc-pwrseq
[  102.413120] mmci-pl18x 58005000.mmc: mmc0: PL180 manf 53 rev2 at 0x58005000 irq 50,0 (pio)
root@ccproa1r:~# [  102.504793] mmc0: queuing unknown CIS tuple 0x01 [d9 01 ff] (3 bytes)
[  102.517462] mmc0: queuing unknown CIS tuple 0x1a [01 01 00 02 07] (5 bytes)
[  102.526253] mmc0: queuing unknown CIS tuple 0x1b [c1 41 30 30 ff ff ff ff] (8 bytes)
[  102.539575] mmc0: new SDIO card at address 0001
[  102.563636] esp32_sdio: esp_probe: ESP network device detected
[  102.581018] esp32_sdio: get_firmware_data: Rx Pre ====== 0
[  102.591899] esp32_sdio: get_firmware_data: Rx Pos ======  0
[  102.596643] esp32_sdio: get_firmware_data: Tx Pre ======  0
[  102.601645] esp32_sdio: get_firmware_data: Tx Pos ======  10
[  102.623154] esp32_sdio: probe of mmc0:0001:2 failed with error -22
[  102.638230] esp32_sdio: process_esp_bootup_event: Received ESP bootup event
[  102.644122] esp32_sdio: process_event_esp_bootup: Bootup Event tag: 3
[  102.650219] esp32_sdio: esp_validate_chipset: Chipset=ESP32 ID=00 detected over SDIO
[  102.658204] esp32_sdio: process_event_esp_bootup: Bootup Event tag: 0
[  102.664522] esp32_sdio: process_event_esp_bootup: Bootup Event tag: 1
[  102.670812] esp32_sdio: process_fw_data: ESP chipset's last reset cause:
[  102.677568] esp32_sdio: print_reset_reason: POWERON_RESET
[  102.683016] esp32_sdio: check_esp_version: ESP-Hosted Version: NG-1.0.4.0.2
[  102.693552] esp32_sdio: esp_reg_notifier: Driver init is ongoing
[  103.022813] esp32_sdio: init_bt: ESP Bluetooth init
[  103.032677] esp32_sdio: print_capabilities: Capabilities: 0x1d. Features supported are:
[  103.039452] esp32_sdio: print_capabilities:   * WLAN on SDIO
[  103.045321] esp32_sdio: print_capabilities:   * BT/BLE
[  103.050201] esp32_sdio: print_capabilities:     - HCI over SDIO
[  103.056153] esp32_sdio: print_capabilities:     - BT/BLE dual mode

```

Driver is not loaded at boot, modprobe does nothing to trigger an actual driver loading I have to unbind/bind the mmc controller.

```

&sdmmc1 {
	pinctrl-names = "default", "opendrain", "sleep";
	pinctrl-0 = <&sdmmc1_b4_pu_pins_a &sdmmc1_clk_pins_a>;
	pinctrl-1 = <&sdmmc1_b4_od_pins_a &sdmmc1_clk_pins_a>;
	pinctrl-2 = <&sdmmc1_b4_sleep_pins_a>;
	non-removable;
	cap-sdio-irq;
	bus-width = <4>;
	mmc-pwrseq = <&wifi_pwrseq>;
	vmmc-supply = <&scmi_vdd_sd>;
	vqmmc-supply = <&scmi_vdd_sd>;
	status = "okay";

	esp_hosted: wifi@0 {
		compatible = "espressif,esp_sdio";
		reg = <0>;
	};
};
```
DTS is also correctly configured. 

Any suggestion?

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

As described above, the problem is always reproducible.

### Project release version

NG-1.0.4.0.2 (master)

### System architecture

other (details in Additional context)

### Operating system

Linux

### Operating system version

Custom yoctobuild on STM32MP 133A

### Shell

sh

angelo-zehnder, 2025-07-09:
@mantriyogesh As per attached logs, the driver is not loading manually. When loaded (at boot or manually) it's not doing anything until I unbind/bind the mmc controller. At that point the SDIO/MMC bus is re scanned and the driver correctly starts to work.

I'm honestly puzzled by this!

mantriyogesh, 2025-07-10:
Is this just similar to Raspberry pi specific sdio option,`poll_once=off` documented  in https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/setup.md#22-sdio-configuration

[image]

You can also have a look at https://github.com/espressif/esp-hosted/blob/5f5daf930fc20f48787d62d8b3de6b5a9d81942f/esp_hosted_ng/esp/esp_driver/network_adapter/main/Kconfig.projbuild#L94-L100 which can be enabled.

One more config that comes to my mind, related to this, is [`broken-cd;`](https://wiki.st.com/stm32mpu/wiki/SDMMC_device_tree_configuration)

[image]

angelo-zehnder, 2025-07-14:
Not being familiar with the SDIO discovery in the Linux Kernel, I had a look at the source code of the kernel to understand.

Basically, once the MMC bus is attached, the kernel starts a discovery on the bus looking at the peripherals it has declared in the dts.
Therefore, in case of dts like this:

```
&sdmmc1 {
	pinctrl-names = "default", "opendrain", "sleep";
	pinctrl-0 = <&sdmmc1_b4_pu_pins_a &sdmmc1_clk_pins_a>;
	pinctrl-1 = <&sdmmc1_b4_od_pins_a &sdmmc1_clk_pins_a>;
	pinctrl-2 = <&sdmmc1_b4_sleep_pins_a>;
	non-removable;
	cap-sdio-irq;
	bus-width = <4>;
	vmmc-supply = <&scmi_vdd_sd>;
	vqmmc-supply = <&scmi_vdd_sd>;
	status = "okay";

	esp_hosted: wifi@0 {
		compatible = "espressif,esp_sdio";
		reg = <0>;
	};
};
```

The kernel expects to find only one peripheral on the bus. The discovery has completed correctly if you can see this message on the console:
```
mmc0: new SDIO card at address 0001
```

The problem is that the esp32 firmware must be ready while discovering on the bus, not later at the kernel loading time. To do that, I simply added a wifi power sequence to the dts and attached to the mmc bus:

```
	wifi_pwrseq: wifi-pwrseq {
		compatible = "mmc-pwrseq-simple";
		reset-gpios = <&gpiod 0 GPIO_ACTIVE_LOW>;
		power-off-delay-us = <200>;
		post-power-on-delay-ms = <1000>;
	};
```
The post power on delay of one second is needed because using a shorter time the firmware is not ready and the kernel is not able to correctly discover the device.

Once the kernel discovered the device, it needs to load the driver. Usually udev looks for an entry in the Hwdb, but there is no such a entry for ID 6666:2222. I solved simply adding a udev rule to do that:

```
KERNELS=="mmc*", SUBSYSTEM=="sdio", ATTR{vendor}=="0x6666", ATTR{device}=="0x2222", ACTION=="add", RUN+="/sbin/modprobe esp32_sdio"
```

So, in the end I think this is the correct approach kernel wise. No hacks involved. I suspect also that the `resetpin` parameter and the reset sequence is completely useless kernel wise and should be removed.

closed 2025-07-14 by angelo-zehnder