espressif/esp-hosted #627: can support esp32c3 usb in linux?

lwj850533527, 2025-07-10:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

support esp32c3 usb in linux

### Actual behavior (suspected bug)

can not support esp32c3 usb in linux

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

support esp32c3 usb in linux

### Project release version

1.0.1.0.3

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

5.1

### Shell

ZSH

mantriyogesh, 2025-07-10:
@kapilkedawat PTAL

lwj850533527, 2025-07-10:
If we want to support USB drivers, is there any USB driver code available for reference in esp32s3
Can we achieve it ourselves

kapilkedawat, 2025-07-10:
Hi @lwj850533527, what exactly is your usecase? USB throughput over WiFi will be very limited after removing the transport overhead. Does SDIO/SPI doesn't suit your design?  

Adding USB driver will be significant work for linux.

lwj850533527, 2025-07-10:
we want to do cross pcb board communication，SDIO/SPI unable to meet the requirements

lwj850533527, 2025-07-11:
Has ESP32S3 ever used  USB driver? Is this a feasible solution？
If we want to add usb driver， will it involve modifications to non open source code
@kapilkedawat

closed 2025-11-21 by lwj850533527