espressif/esp-hosted #630: Occasionally dead loop occurs "More segment"

alexsvn, 2025-07-16:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

rarely

### Expected behavior

normal

### Actual behavior (suspected bug)

Software based on commit 513bdfa93f5dda86d828767e5e55def718443911
hosted_fg software
Occasionally dead loop occurs (found once), log as follows
I (2347088) NETWORK_ADAPTER: More segment: 0 curr seq: 68 header seq: 69
I (2347188) NETWORK_ADAPTER: More segment: 0 curr seq: 68 header seq: 69
I (2347288) NETWORK_ADAPTER: More segment: 0 curr seq: 68 header seq: 69

How to restore the status

```c
void process_serial_rx_pkt(uint8_t *buf)
   ...
    while (r.valid)
    {
        ESP_LOGI(TAG,"More segment: %u curr seq: %u header seq: %u\n",
            header->flags & MORE_FRAGMENT, r.cur_seq_no, header->seq_num);
        vTaskDelay(10);
    }
    ...
```

### Error logs or terminal output

```shell
I (2347088) NETWORK_ADAPTER: More segment: 0 curr seq: 68 header seq: 69
I (2347188) NETWORK_ADAPTER: More segment: 0 curr seq: 68 header seq: 69
I (2347288) NETWORK_ADAPTER: More segment: 0 curr seq: 68 header seq: 69
```

### Steps to reproduce the behavior

Step:
Execute test

### Project release version

513bdfa93f5dda86d828767e5e55def718443911

### System architecture

other (details in Additional context)

### Operating system

Linux

### Operating system version

3.18.48

### Shell

sh

mantriyogesh, 2025-11-03:
I still think there is necessity to first verify if the transport is working fine.

Some things to note:
1. Always ensure the slave firmware and host code used for the same variant.
For example, you cannot use ESP-Hosted-FG firmware with ESP-Hosted-MCU host code.
2. [isolated transport test](https://github.com/espressif/esp-hosted-mcu/blob/main/docs/troubleshooting.md#2-raw-throughput-testing) in tx and rx independently tested is very important.
3. If SDIO is being used, would highly suggest to take camera picture and send us how they are connected. Hardware requirements are stated ([here](https://github.com/espressif/esp-hosted-mcu/blob/main/docs/sdio.md#3-hardware-considerations))
4. Please state the host platform used and slave used, Is Host ESP32-P4 or something else? Is host ESP32-C6 or different?

The original issue of 
```
I (217088) co-pro-main: More segment: 0 curr seq: 0 header seq: 0
I (217188) co-pro-main: More segment: 0 curr seq: 0 header seq: 0
I (217288) co-pro-main: More segment: 0 curr seq: 0 header seq: 0
```

is something we confirm, but we tried to replicate this issue, and not able to reproduce easily, so not yet fixed.

```
W (33363) H_SDIO_DRV: invalid ret or len_from_slave: 0 0
```

is different, overall now my focus has changed due to these kind of issues, to first cross check the transport validity..

nkp123, 2025-11-06:
Thanks for your reply,

About your notes:

1. I am testing ESP-Hosted-MCU variant, both on host and the slave.

2. In general the hardware setup is the same as in my other issue, about TCP/UDP performance (https://github.com/espressif/esp-hosted-mcu/issues/112#issuecomment-3412565790). As I am facing stability issues let's put that issue on hold.
In general:
- My setup is 4 bit SDIO @ 20MHz on dedicated, custom PCB, with all signals pulled up with 51kOhm resistors.
- I am getting ~53 Mbps of raw SDIO transport bandwidth from host to coprocessor.
- I am getting ~63 Mbps of raw SDIO transport bandwidth from coprocessor to host.

3. Here is the schematics:
[image]
The PCB is 4 layer with one whole ground plane. Traces are length-matched, however I did not do proper impedance matching.

4. The host is ESP32P4NRW32, coprocessor is ESP32-WROOM-32E-N16R2.

As a workaround disabling SDIO streaming mode seems to work, however, as I understand, streaming mode is more prone to packet drops on the line, is that right?

I will make additional tests if I am dropping packets and test again different SDIO settings, particularly reducing clk speed and switching to 1 bit mode. I will also try to test signal quality on the SDIO lines directly via oscilloscope.

Thanks,
Norbert