espressif/esp-hosted #631: Support for ESP32-C3-MINI-1-N4X for NG variant

mayurpatel-karcher, 2025-07-16:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

Hello guys,

It's not an issue but just a query.

OLD  : **ESP32-C3-MINI-1-N4**
NEW : **ESP32-C3-MINI-1-N4X**

We are currently using ESP32-C3-MINI-1-N4 with your **esp-hosted project with NG variant** and its working file. (we also worked together on other issue where AP mode was not working correctly and now that issue has been also solved). 

Now we got the notification from espressif that there is a new variant ESP32-C3-MINI-1-N4X (with "X" in the end) and our HW guys confirmed that its not much effort to integrate it. So now we want to be sure if SW is the same case. 

So, **Will the SW work 100% same on new variant as it is working now OR any changes are required on SW side also?**
Happy to see a response asap. Thank you.

mayurpatel-karcher, 2025-07-16:
@mantriyogesh can you answer this?

mantriyogesh, 2025-07-16:
@kapilkedawat PTAL.