espressif/esp-hosted #633: esp is not detected unless sd-controller is unbound and rebound.

libschmudie-so, 2025-07-16:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I expected it to load the esp32_sdio module, probe the mmc device and create the wlan0 interface

### Actual behavior (suspected bug)

Instead, it loads the esp32_sdio and fails to probe the mmc device

### Error logs or terminal output

```shell
[ 1436.019793] esp32_sdio mmc2:0001:1: probe with driver esp32_sdio failed with error -110
[ 1436.019952] esp32_sdio mmc2:0001:2: probe with driver esp32_sdio failed with error -110
```

### Steps to reproduce the behavior

1. Make sure the esp32 is booted.
2. Load the esp32_sdio module

### Project release version

24c86389142110706fb71f7df63979f0112c7580

### System architecture

ARM 32-bit (Raspberry Pi 32-bit)

### Operating system

Linux

### Operating system version

Alpine 3.22

### Shell

sh

libschmudie-so, 2025-07-17:
Hi @mantriyogesh 

I do not believe it is related to signal integrity.

All SDIO lines are length matched within 150 mils
Ground plane is underneath soc, esp32 and sdio lines
22k Pull-Ups are on all SDIO Data and CMD lines
eFuse in ESP32 is burned

Also, if I do this:

```
 echo 4021000.mmc > /sys/bus/platform/drivers/sunxi-mmc/unbind
 echo 4021000.mmc > /sys/bus/platform/drivers/sunxi-mmc/bind
```

The ESP is detected and works fine:

```
[   17.294012] esp32_sdio: loading out-of-tree module taints kernel.
[   17.504768] esp32_sdio: esp_probe: ESP network device detected
[   17.505294] esp32_sdio: get_firmware_data: Rx Pre ====== 0
[   17.505355] esp32_sdio: get_firmware_data: Rx Pos ======  0
[   17.505472] esp32_sdio: get_firmware_data: Tx Pre ======  0
[   17.505491] esp32_sdio: get_firmware_data: Tx Pos ======  10
[   17.505980] esp32_sdio mmc2:0001:2: probe with driver esp32_sdio failed with error -22
[   17.507131] esp32_sdio: process_esp_bootup_event: Received ESP bootup event
[   17.507172] esp32_sdio: process_event_esp_bootup: Bootup Event tag: 3
[   17.507190] esp32_sdio: esp_validate_chipset: Chipset=ESP32 ID=00 detected over SDIO
[   17.507204] esp32_sdio: process_event_esp_bootup: Bootup Event tag: 0
[   17.507217] esp32_sdio: process_event_esp_bootup: Bootup Event tag: 1
[   17.507230] esp32_sdio: process_fw_data: ESP chipset's last reset cause:
[   17.507258] esp32_sdio: print_reset_reason: POWERON_RESET
[   17.507281] esp32_sdio: check_esp_version: ESP-Hosted Version: NG-1.0.4.0.1
[   17.508072] esp32_sdio: esp_reg_notifier: Driver init is ongoing
[   17.792414] esp32_sdio: init_bt: ESP Bluetooth init
[   17.794700] esp32_sdio: print_capabilities: Capabilities: 0x1d. Features supported are:
[   17.794740] esp32_sdio: print_capabilities: 	 * WLAN on SDIO
[   17.794752] esp32_sdio: print_capabilities: 	 * BT/BLE
[   17.794763] esp32_sdio: print_capabilities: 	   - HCI over SDIO
[   17.794775] esp32_sdio: print_capabilities: 	   - BT/BLE dual mode
```

mantriyogesh, 2025-07-17:
duplicate #https://github.com/espressif/esp-hosted/issues/626

libschmudie-so, 2025-07-17:
If I add the reset pin to a pwrseq in the device tree the GPIO is not
available anymore to the ESP32 Kernel Module. In the docs it is stated that
the resetpin is mandatory. If I do not provide the reset pin to the kernel
module will it still work and also will OTA work?

libschmudie-so, 2025-07-17:
> Ideally it should work for sdio. Although SDIO , we strongly suggest using
> resetpin, that way we know the slave is always fresh start and always works
> without manually accessing the slave and slave logs.

But if I have to use the pwrseq for it to even work then I cannot give your
module the resetpin as it will fail

mantriyogesh, 2025-07-17:
> Since the kernel is resetting the esp on boot anyways it will be fine

Yes.

> I will try it this way and reopen this if I have any further issues

sure, do let us know! Thanks.

angelo-zehnder, 2025-07-24:
Don't use the `resetpin` module parameter. In my opinion, it is the wrong way to solve the problem and shall be removed.

Simply attache the reset pin on the mmc pwrseq and let the kernel resets the esp32 just before the discovery, this will ensure a consistent behavior.

Then let the kernel loading the driver through a udev rule.

mantriyogesh, 2025-07-25:
Thanks for the suggestion. Using `mmc pwrseq` and udev is a clean and consistent approach.

The `resetpin` parameter offers a more generic option, especially for cases where such integration isn’t available or may not be straightforward for all users. Our goal is to provide a simple, consistent, and flexible mechanism.

angelo-zehnder, 2025-07-25:
Don't get me wrong but I have a couple drivers in the mainline linux kernel and can express some concerns.

The driver quality purely from the point of view of compliance with Linux standards is pretty low.

Kernel abandoned modules parameters 15 years ago at least when the dts became the standard way to express those things.
Your driver lacks even the smallest hint in this way.

Trust me if I say the `resetpin` parameter is the wrong way to solve an initialization problem which is present in a lot of drivers and it has already been solved in a standard way.

Indeed your documentation is lacking a solid way to make the driver working, it's full of hacks and scripts and you cannot even explain why it's not loading or it's not working correctly. IMHO you should really urge to work toward some integration within the linux kernel mainline. That way you could receive a ton of valid feedback and offer something of a sufficient quality.

mantriyogesh, 2025-07-25:
This does look good plan IMO. But let me have a discussion within the teams once and come back..

nichonaugle, 2025-07-28:
Agree with this issue, I experience the same issue. -110 error goes away after rebinding mmc

mantriyogesh, 2025-08-05:
Sorry, I took some time to come back to this, @nichonaugle.

> [!Note]
>
> This post requires a detailed understanding and also involves FG, NG, and MCU repo understanding.

**For now, our thinking is to sort out the DTS things first, as it affects every user, and holds higher severity**

Let me tell you about the current things on our plates. Considering that Esp-Hosted-FG and Esp-Hosted-MCU are very similar, with only changes in their RPC handling, we are trying to combine the firmware repos to make the host repo separate. Why? Because we even face duplication of effort to maintain the same transport across the 3 variants. Having said that, if things go smoothly, we will further change the NG firmware to use this 'firmware' repo, which would make the Linux code for FG and NG alone sit in the ESP-hosted repo. Furthermore, the esp-hosted transport level repo can actually be unified to just leverage higher-level separation in FG and NG at build time. This is not yet final, but we're just planning the thought process.

Current priorities:

1. Componentize the slave-side code, which should also be modular so that only the things needed can be used as distinct components.
Something like this:

```
┌─────────────────────────────────────────────────────────┐
│                  Application Layer                      │
└─────────────────────────────────────────────────────────┘
┌─────────────────────────────────────────────────────────┐
│                Extension Layer                          │
│  ┌─────────────────────────────────────────────────────┐│
│  │        ehrcp_ext_user_defined_rpc                   ││
│  │        (renamed from ehrcp_ext_rpc)                 ││
│  └─────────────────────────────────────────────────────┘│
└─────────────────────────────────────────────────────────┘
┌─────────────────────────────────────────────────────────┐
│                 Component Layer                         │
│  ┌─────────────────────┐ ┌─────────────────────────────┐│
│  │ ehrcp_base_rpc_     │ │ ehrcp_protobuf_linux        ││  
│  │ linux               │ │ (protobuf handling)         ││
│  │ (core RPC logic)    │ │                             ││
│  └─────────────────────┘ └─────────────────────────────┘│
└─────────────────────────────────────────────────────────┘
┌─────────────────────────────────────────────────────────┐
│                Core Component (Minimal)                 │
│  ┌─────────────────────────────────────────────────────┐│
│  │              esp_hosted_slave                       ││
│  │  ┌───────────┐ ┌─────────────┐ ┌─────────────────┐  ││
│  │  │Transport  │ │protocomm/   │ │Event System     │  ││
│  │  │(SPI/SDIO) │ │pserial_task │ │RPC Registration │  ││
│  │  └───────────┘ └─────────────┘ └─────────────────┘  ││
│  └─────────────────────────────────────────────────────┘│
└─────────────────────────────────────────────────────────┘
```

Although this doesn't capture all the components, it should give an idea of how the components would be structured.

The component layer shows Linux RPC right now, but it's the same level where NG RPC or message passing would lie. It's also the same layer where the MCU RPC would lie. The end application is nothing but a simple example for NG, FG, or MCU. It would just use the needed RPC, depending on the component it needs.

> [!Note]
>
> Component names are just representative right now, and would be changed consistently.

What would this achieve: It would make the slave repo for FG & MCU unified and in separate code and repo from the host.

2. Integrate the NG firmware into this repo. Again, only the components needed should be modular and can be integrated for the NG example.

All this is still in progress, and some parts of the code, which are variant-specific, would again be moved as separate components with clearly defined APIs for those components.

But all this is only slave code, yet I discussed it.

Once moved out, we would be left with only Linux repos for FG and NG in the main repo. How to separate them is an easy decision to make, but still maintaining two separate transports makes them out of sync at times, especially with ad-hoc bug fixes.

I really would like to know your inputs and thought process for the above.