espressif/esp-hosted #634: ESP32C6 Over SPI, VERY SLOW

nichonaugle, 2025-07-19:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

My current setup has an ESP32C6 connected over SPI to an RV1106G3. The device tree has been properly setup and I am able to communicate between the Linux host and esp client. However the bit rate is EXTREMELY slow. I am using the NG release 1.0.4.0.0 for both the Linux host and esp client. When I set "raw_tp_mode=1" while loading the kernel module, the Linux host just freezes. However after setting up dhcp and wpa supplicant, I am able to download files, granted very slowly. Below is attached the bit rate I am getting which realistically hovers around 0.5Mbps (very very slow). I can confirm that all pins are connected properly and are using very short traces away from power to avoid EMI. Proper grounding has also been taken into consideration. I have tried running the 4 SPI modes and setting frequencies between 2MHz and 26MHZ for the esp and linux host, all yielding the same result.

[image]

### Actual behavior (suspected bug)

I have tried downloading files, uploading files, pinging from esp to a server, and from server to esp with iperf and have never been able to get above ~400-500kbps reliably. Here are some more startup logs: 

[image]

Note: My esp32c6 is a xiao board so the GPIO 14 is set high to use external antenna. I have tried using onboard chip antenna as well and the same results occured
[image]

Results after connecting to network with wpa_supplicant: 
[image]

Results after dns completes setup: 
[image]

Ping results: 
[image]

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

Reproducibility is quite complex and I believe this may be a hardware limitation or slight configuration issue. If there are other pointers that can be given, i.e. using a different chipset or modified menuconfig I would be more than happy to try it out.

these are the commands I am running in order as listed:

modprobe bluetooth
modprobe cfg80211
insmod /lib/modules/$(uname -r)/extra/esp32_spi.ko resetpin=56 raw_tp_mode="0"
ifconfig wlan0 up
echo $'network={\n\tssid="SSID"\n\tpsk="PASSWORD"\n}' > /etc/wpa_supplicant.conf
wpa_supplicant -d -Dnl80211 -iwlan0 -c /etc/wpa_supplicant.conf -B
udhcpc -i wlan0 -v

 "After this the network will be online and ping 8.8.8.8 will be functional"

### Project release version

ng release 1.0.4.0.0

### System architecture

other (details in Additional context)

### Operating system

Linux

### Operating system version

24.04.2 LTS (Noble Numbat)

### Shell

Bash

nichonaugle, 2025-07-19:
@mantriyogesh

nichonaugle, 2025-07-19:
Thank you for your fast reply. 

1) Regarding iperf3, When I try a higher bitrate the packet loss is very high, with the true bitrate staying around 500Kbps. Attached is the image, I do not believe this is the issue.

[image]

2) Porting guide was followed, and other references were followed to configure the SPI dts (two other issues describe parts of their setup, 
reference A: https://forums.luckfox.com/viewtopic.php?t=1124   
reference B: https://github.com/espressif/esp-hosted/issues/500) 

3) I just updated both the Host and the slave to 1.0.4.0.2 (latest from master) and the same result occurred. What you said regarding the raw_tp testing makes me wonder why the linux host is freezing when that command is run... Initially I notice a large 912.5 kbits/sec then it drops to zero indefinitely, while the host stays in a frozen state. I am connected in via UART serial console.

[image]

[image]

The logs from ```tail -F /var/log/messages``` are useless as loading the kernel module freezes the entire OS. 

@mantriyogesh 
**Can you provide the list of packages needed on the linux host to successfully run esp-hosted?** Given that I am running a custom buildroot OS with busybox, it is fairly lightweight and may be missing packages. I was unable to find any reference to this in the documentation.

mantriyogesh, 2025-07-19:
Can you take a close camera picture of your setup?

mantriyogesh, 2025-07-20:
### ESP-Hosted NG – Debug Plan (Step-by-Step)

A **top-down approach would be hard to debug** here. I recommend checking the **hardware first**, then going **layer by layer in software**.

Can you please confirm that the following interrupt lines are **working stably**?

* `DATA_READY`
* `HANDSHAKE`
* `RESET`

> [!NOTE]
> As we’re entering debug mode, be aware that **debug prints slow down the system** and may cause race/timing issues. These prints are **not suitable for production builds**.

## Power and Voltage

### **Host Power Adapter Rating**

* Use a stable and sufficient power source for the host device as per the input power rating specified in the SoC documentation
* Power ESP directly externally from your laptop, instead of powering from Host's USB ports.

### **Power & Voltage Checks**

Before proceeding with software debug, ensure stable GPIO voltage levels:

* Measure voltage at each GPIO (MISO, MOSI, CS, CLK, DATA\_READY, HANDSHAKE, RESET) — should measure **3.3V** (unless using a different logic level system).
* Confirm the ESP32 and host ground (GND) are properly connected.
* Possible voltage dips on breadboards should be assessed. Possibly short the GPIOs directly using jumper cables from ESP to host with <=10 cm length, all equal.
* ⚠️ Ensure **GPIOs are not reused** for other functions or multiplexed peripherals in either ESP32 or host device configuration. Reused pins may lead to conflicts and unexpected behavior.

## spidev
You are already through it, otherwise the capabilities wouldn't have received at host.

## GPIOs

### **SPI lines:** MISO, MOSI, CS, CLK

* Basic validation confirmed via `print_capabilities` on host. This confirms **MOSI, CS, CLK** are functioning.
* **MISO:** Not explicitly tested yet. Needs validation using the return path or raw throughput on the opposite side.

### Other 3 GPIOs (Handshake, Data Ready, and ResetSlave)

#### ISR for Handshake and Data Ready

Add debug traces (or GPIO toggles) inside IRQ handlers to verify whether the interrupts are **triggering and handled correctly**.

1. **Handshake IRQ Handler**

   [esp\_spi.c#L64](https://github.com/espressif/esp-hosted/blob/ba39e6392f4f1c2e9ace3c5fa463bab34a42cb31/esp_hosted_ng/host/spi/[esp_spi.c#L64](https://github.com/espressif/esp-hosted/blob/ba39e6392f4f1c2e9ace3c5fa463bab34a42cb31/esp_hosted_ng/host/spi/esp_spi.c#L64))
2. **Data Ready IRQ Handler**

   [esp\_spi.c#L73](https://github.com/espressif/esp-hosted/blob/ba39e6392f4f1c2e9ace3c5fa463bab34a42cb31/esp_hosted_ng/host/spi/[esp_spi.c#L73](https://github.com/espressif/esp-hosted/blob/ba39e6392f4f1c2e9ace3c5fa463bab34a42cb31/esp_hosted_ng/host/spi/esp_spi.c#L73))

#### **Reset Pin (Host → ESP32)**

Confirm: ESP32 resets reliably when the module is inserted with the `resetpin` parameter:

```bash
insmod esp32_spi.ko resetpin=<slave_reset_pin>
```

[SPI_design_note.pdf](https://github.com/user-attachments/files/21333019/SPI_design_note.pdf)

mantriyogesh, 2025-07-20:
Maybe, if you add https://github.com/espressif/esp-hosted/blob/ba39e6392f4f1c2e9ace3c5fa463bab34a42cb31/esp_hosted_fg/esp/esp_driver/network_adapter/main/spi_slave_api.c#L726 to your `NG` slave side code, it might work. but that's just a guess. Anyway, let's proceed step by step in https://github.com/espressif/esp-hosted/issues/634#issuecomment-3093200922

nichonaugle, 2025-07-20:
@mantriyogesh Thank you for your detailed and prompt reply, it is very greatly appreciated. I went to work checking off each point above, and here are my results...

All of the GPIOs measure 3.3V except CLK and MISO. I do not have a scope so I presume they are transmitting and my multimeter is unable to catch the edges fast enough.

I attempted to the fix using ``` register_hs_disable_pin(GPIO_CS); ``` and it appears to not have made a difference. I did add the missing methods in the same file and compiled it successfully. However, as I was looking through the host source code I decided to pass 2 instead of 1 to the following command: 

```insmod /lib/modules/$(uname -r)/extra/esp32_spi.ko resetpin=56 raw_tp_mode="2"```

Now the system does not crash and seems to be running at ~11500 kbits/sec on both client and host:

Linux Host
[image]

ESP32C6 Client
[image]

What is the reasoning for this and how do I get it to produce this throughput for real networking scenarios? I am unsure of whether this is an ESP or linux issue at this point. Seems like a promising step in the right direction!

Looking forward to your reply.

nichonaugle, 2025-07-20:
Upon further inspection, it seems that this is used for testing ESP to HOST raw throughput. This points to an issue with the linux host. Are there any missing packages or dependencies that may be causing issues with communication from the host?

[image]

mantriyogesh, 2025-07-21:
Actually, the documentation for raw throughput, as you may know, is at [`Raw throughput test`](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/Raw_TP_Testing.md).

The `rpi_init.sh` should be modified according to your porting, as mentioned in [`Porting of  rpi_init.sh`](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md#22-rpi_initsh).

You are right; it would translate internally to the module parameters.

https://github.com/espressif/esp-hosted/blob/ba39e6392f4f1c2e9ace3c5fa463bab34a42cb31/esp_hosted_ng/host/rpi_init.sh#L151-L157

and

https://github.com/espressif/esp-hosted/blob/ba39e6392f4f1c2e9ace3c5fa463bab34a42cb31/esp_hosted_ng/host/rpi_init.sh#L72

Did you manage to test the raw throughput on both sides individually? May I know if the other way traffic passes or fails? Could you provide textual logs as attachments from both sides?

nichonaugle, 2025-07-21:
The raw throughput tests from HOST -> ESP are seen in my previous comment here: https://github.com/espressif/esp-hosted/issues/634#issuecomment-3092500106

Where the host crashed shortly after starting. ❌

My most recent comment shows the throughput from ESP -> HOST, and the functionality shows ~11500 kbits/sec of throughput✅

I'm am currently exploring more logging on the host side with potential modifications to the DTS

mantriyogesh, 2025-07-21:
Noted.

Did you get a chance to cross-check if https://github.com/espressif/esp-hosted/issues/634#issuecomment-3093200922 -> `GPIOs` section, to check if the Handshake, reset pin, and data ready work as intended?

mantriyogesh, 2025-07-21:
-- Internal --

@Shreyas0-7, can we please cross-check if the raw throughout test passes in host->slave directions?

To verify on Raspberry pi: 
> I just updated both the Host and the slave to 1.0.4.0.2 (latest from master) and the same result occurred. What you said regarding the raw_tp testing makes me wonder **why the linux host is [...]

nichonaugle, 2025-07-21:
I will update you on the results from the GPIO handshake and data ready once I complete some more testing, I know the reset GPIO works properly as the ESP automatically restarts when the kernel module is first loaded. I will also take a look at using an older version to see if that will resolve the issue.

mantriyogesh, 2025-07-21:
Irrespective of the older or newer version, `GPIO handshake and data ready` are mandatory. This is the base design.

Please check [SPI_design_note.pdf](https://github.com/user-attachments/files/21333019/SPI_design_note.pdf) shared earlier.

nichonaugle, 2025-07-21:
[image]

Using gpioinfo, the kernel module is able to set the respective GPIO's as inputs for both HANDSHAKE and DATA READY.

Shreyas0-7, 2025-07-21:
Hey @mantriyogesh have tested raw throughput from host tp slave for c5sdio with rpi.  Here are the results

[raw_tp.log](https://github.com/user-attachments/files/21342877/raw_tp.log)

nichonaugle, 2025-07-21:
I have added logging at the locations described and below are the results. The IRQ and Data Ready are being triggered frequently 

[image]

I also included the interrupt list and it validates that the pins are receiving multiple events to the right interrupt pins: 

[image]

Shreyas0-7, 2025-07-22:
Here are logs for c6 spi raw throughput 
[raw_tp_spi_c6.log](https://github.com/user-attachments/files/21365493/raw_tp_spi_c6.log)

mantriyogesh, 2025-07-29:
@nichonaugle Oh! I think you already have this thread, and it is the same issue, in my opinion.

nichonaugle, 2025-07-29:
@mantriyogesh The issue is still happening. I noticed there were others facing the same issue and I still working to diagnose the root problem. This definitely seems like an OS specific issue as I have the same problem for both SPI and SDIO, however I am unsure of whether it is a timing related problem or a driver/dts related problem. Here is the log where it freezes: 

[image]

I am also attaching the modified kernel driver with logs to show where it is failing. The main two functions with added logging are raw_tp_tx_process in esp_stats-MODIFIED.txt and write_packet in esp_sdio-MODIFIED.txt

[esp_sdio-MODIFIED.txt](https://github.com/user-attachments/files/21490815/esp_sdio-MODIFIED.txt)
[esp_stats-MODIFIED.txt](https://github.com/user-attachments/files/21490816/esp_stats-MODIFIED.txt)