espressif/esp-hosted #635: "sdio_buffer_alloc" assert fail

YeChaoLi, 2025-07-20:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

It should normally run like the demo.
\

### Actual behavior (suspected bug)

the sdio_buffer_alloc will call mempool_alloc, and sometimes hit assert that indicating it's a null ptr return.
I have check my internal ram has still 10% extra
I use esp32p4+esp32c6 

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

I use this in my product code...  It contains tinyUSB, cjson, lvgl, and other applications...

### Project release version

espressif/esp_wifi_remote: ^0.14.3

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

ubuntu24

### Shell

ZSH

SohKamYung-Espressif, 2025-07-21:
Hello @YeChaoLi ,

If you do not need high network throughput, you can reduce the amount of memory ESP-Hosted uses as buffers in SDIO. See the SDIO [Performance and Memory Usage](https://github.com/espressif/esp-hosted-mcu/blob/main/docs/sdio.md#9-performance-and-memory-usage) section of the documentation.

You can also take a look at https://github.com/espressif/esp-hosted/issues/597 which is similar to your issue.

SohKamYung-Espressif, 2025-07-21:
@YeChaoLi 

Please transfer this issue to https://github.com/espressif/esp-hosted-mcu/issues , which covers esp32p4+esp32c6 .