espressif/esp-hosted #637: ESP32C3 esp_hosted_ng in raspberry pi5

derekwen168, 2025-07-21:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

1. I am using rpi_init.sh spi to initialize ESP-Hosted-NG on a Raspberry Pi 5. Since wlan0 is occupied by the Pi 5's internal Wi-Fi, I need to change the wireless interface to wlan1 (assigned to the ESP32).
How can I modify the script to use wlan1 instead of wlan0?
2.The following error occurred when executing rpi_init.sh spi.
Before running the script, I had already applied the patch from https://github.com/espressif/esp-hosted/issues/587.

### Actual behavior (suspected bug)

derek@raspberrypi:~/esp-hosted/esp_hosted_ng/host $ sudo bash rpi_init.sh spi
Building for spi protocol
rm: cannot remove 'spidev_disabler.dtbo': No such file or directory
rm -rf *.o */*.o *.ko *.mod.c *.symvers *.order .*.cmd .tmp_versions
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/6.12.20+rpt-rpi-2712/build M=/home/derek/esp-hosted/esp_hosted_ng/host clean
make[1]: Entering directory '/usr/src/linux-headers-6.12.20+rpt-rpi-2712'
make[1]: Leaving directory '/usr/src/linux-headers-6.12.20+rpt-rpi-2712'
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/6.12.20+rpt-rpi-2712/build M=/home/derek/esp-hosted/esp_hosted_ng/host modules
make[1]: Entering directory '/usr/src/linux-headers-6.12.20+rpt-rpi-2712'
  CC [M]  /home/derek/esp-hosted/esp_hosted_ng/host/spi/esp_spi.o
  CC [M]  /home/derek/esp-hosted/esp_hosted_ng/host/esp_bt.o
  CC [M]  /home/derek/esp-hosted/esp_hosted_ng/host/main.o
  CC [M]  /home/derek/esp-hosted/esp_hosted_ng/host/esp_cmd.o
  CC [M]  /home/derek/esp-hosted/esp_hosted_ng/host/esp_utils.o
  CC [M]  /home/derek/esp-hosted/esp_hosted_ng/host/esp_cfg80211.o
  CC [M]  /home/derek/esp-hosted/esp_hosted_ng/host/esp_stats.o
  CC [M]  /home/derek/esp-hosted/esp_hosted_ng/host/esp_debugfs.o
  CC [M]  /home/derek/esp-hosted/esp_hosted_ng/host/esp_log.o
  LD [M]  /home/derek/esp-hosted/esp_hosted_ng/host/esp32_spi.o
  MODPOST /home/derek/esp-hosted/esp_hosted_ng/host/Module.symvers
  CC [M]  /home/derek/esp-hosted/esp_hosted_ng/host/esp32_spi.mod.o
  CC [M]  /home/derek/esp-hosted/esp_hosted_ng/host/.module-common.o
  LD [M]  /home/derek/esp-hosted/esp_hosted_ng/host/esp32_spi.ko
make[1]: Leaving directory '/usr/src/linux-headers-6.12.20+rpt-rpi-2712'
insmod: ERROR: could not insert module esp32_spi.ko: Device or resource busy

### Error logs or terminal output

```shell
N/A
```

### Steps to reproduce the behavior

N/A

### Project release version

ESP-Hosted-NG-release_v1.0.4.0.0

### System architecture

ARM 64-bit (Apple M1/M2, Raspberry Pi 4/5)

### Operating system

Linux

### Operating system version

Linux raspberrypi 6.12.20+rpt-rpi-2712

### Shell

sh

mantriyogesh, 2025-07-21:
please attach the slave and host side logs intextual form.

esp: log from flashing/bootup
host: dmesg from system startup

also please state the gpios used in connections

derekwen168, 2025-07-22:
After running bash rpi_init.sh spi ap_support and encountering an error, I executed dmesg | tail -n 50.
result..

derek@raspberrypi:~/esp-hosted/esp_hosted_ng/host $ dmesg | tail -n 50
[17086.447517] NET: Registered PF_BLUETOOTH protocol family
[17086.447520] Bluetooth: HCI device and connection manager initialized
[17086.447531] Bluetooth: HCI socket layer initialized
[17086.447535] Bluetooth: L2CAP socket layer initialized
[17086.447541] Bluetooth: SCO socket layer initialized
[17090.725261] esp32_spi: loading out-of-tree module taints kernel.
[17090.928281] esp32_spi: spi_dev_init: Using SPI MODE 2
[17090.928452] esp32_spi: spi_dev_init: Config - SPI GPIOs: Handshake[593] Dataready[598]
[17090.928455] esp32_spi: spi_dev_init: Config - SPI clock[10MHz] bus[0] cs[0] mode[2]
[17090.928521] esp32_spi: spi_dev_init: Failed to obtain GPIO for Data ready pin, err:-16
[17091.340258] ------------[ cut here ]------------
[17091.340265] Trying to free already-free IRQ 187
[17091.340274] WARNING: CPU: 0 PID: 60819 at kernel/irq/manage.c:1900 free_irq+0x174/0x2a0
[17091.340283] Modules linked in: esp32_spi(O+) bluetooth ecdh_generic ecc md5 aes_ce_ccm sha512_generic sha512_arm64 nls_utf8 cifs cifs_arc4 nls_ucs2_utils netfs cifs_md4 snd_seq_dummy snd_hrtimer snd_seq snd_seq_device nft_chain_nat xt_MASQUERADE nf_nat xt_conntrack nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nft_compat nf_tables nfnetlink binfmt_misc brcmfmac_wcc spidev aes_ce_blk aes_ce_cipher ghash_ce gf128mul libaes sha2_ce sha256_arm64 sha1_ce joydev brcmfmac rpi_hevc_dec brcmutil v4l2_mem2mem pisp_be videobuf2_dma_contig cfg80211 sha1_generic videobuf2_memops videobuf2_v4l2 cp210x raspberrypi_hwmon videodev usbserial rfkill videobuf2_common i2c_brcmstb mc gpio_keys spi_bcm2835 pwm_fan rp1_pio raspberrypi_gpiomem rp1_adc spi_dw_mmio rp1_mailbox i2c_designware_platform rp1 spi_dw i2c_designware_core nvmem_rmem i2c_dev ledtrig_pattern fuse dm_mod ip_tables x_tables ipv6 vc4 snd_soc_hdmi_codec snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd drm_display_helper drm_dma_helper cec v3d drm_shmem_helper
[17091.340356]  drm_kms_helper gpu_sched drm drm_panel_orientation_quirks backlight uio_pdrv_genirq uio
[17091.340364] CPU: 0 UID: 0 PID: 60819 Comm: insmod Tainted: G           O       6.12.20+rpt-rpi-2712 #1  Debian 1:6.12.20-1+rpt1~bpo12+1
[17091.340369] Tainted: [O]=OOT_MODULE
[17091.340370] Hardware name: Raspberry Pi 5 Model B Rev 1.0 (DT)
[17091.340372] pstate: 604000c9 (nZCv daIF +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[17091.340374] pc : free_irq+0x174/0x2a0
[17091.340376] lr : free_irq+0x174/0x2a0
[17091.340378] sp : ffffc000837c3930
[17091.340379] x29: ffffc000837c3930 x28: ffffd06fcf5e0000 x27: ffffd06fcfe4f0f8
[17091.340383] x26: ffffc000837c3c10 x25: 00000000000000bb x24: ffff80000275d8ac
[17091.340386] x23: ffff80000275d960 x22: 0000000000000000 x21: ffff800003913800
[17091.340390] x20: ffff80000275d800 x19: ffffd06f5434d980 x18: ffffffffffffffff
[17091.340393] x17: 0000000000000000 x16: ffffd06fce526820 x15: ffffc000837c3530
[17091.340396] x14: 0000000000000000 x13: 3738312051524920 x12: 656572662d796461
[17091.340399] x11: 65726c6120656572 x10: ffffd06fcfc68388 x9 : ffffd06fce51c1d0
[17091.340402] x8 : 00000000ffffefff x7 : ffffd06fcfc68388 x6 : 80000000fffff000
[17091.340405] x5 : ffff80001ff033c8 x4 : 0000000000000000 x3 : ffffaf9050700000
[17091.340408] x2 : 0000000000000000 x1 : 0000000000000000 x0 : ffff80000672c600
[17091.340411] Call trace:
[17091.340412]  free_irq+0x174/0x2a0
[17091.340415]  spi_exit+0x1ac/0x270 [esp32_spi]
[17091.340426]  esp_init_interface_layer+0x3d0/0x670 [esp32_spi]
[17091.340432]  esp_init+0x154/0x3ff8 [esp32_spi]
[17091.340438]  do_one_initcall+0x60/0x2a0
[17091.340442]  do_init_module+0x60/0x230
[17091.340446]  load_module+0x1ff0/0x20d8
[17091.340448]  init_module_from_file+0x8c/0xd8
[17091.340450]  __arm64_sys_finit_module+0x150/0x330
[17091.340452]  invoke_syscall+0x50/0x120
[17091.340456]  el0_svc_common.constprop.0+0x48/0xf0
[17091.340459]  do_el0_svc+0x24/0x38
[17091.340462]  el0_svc+0x30/0xd0
[17091.340467]  el0t_64_sync_handler+0x100/0x130
[17091.340470]  el0t_64_sync+0x190/0x198
[17091.340473] ---[ end trace 0000000000000000 ]---
[17091.764223] esp32_spi: spi_init: Failed Init SPI device
derek@raspberrypi:~/esp-hosted/esp_hosted_ng/host $

mantriyogesh, 2025-07-23:
worth checking:

https://forums.raspberrypi.com/viewtopic.php?t=359302

-- internal --
@Shreyas0-7, I don't have Rpi5 right now with me. Would you please cross-check which GPIO numbers should be used for Raspberry Pi 5?

Shreyas0-7, 2025-07-23:
Hey @derekwen168 can you please try with the following patch
please apply this patch in `/local_path/esp-hosted/esp_hosted_ng/host/` directory
unzip the patch
`git apply rpi5_spi.path`
[rpi5_spi.zip](https://github.com/user-attachments/files/21384085/rpi5_spi.zip)

derekwen168, 2025-07-23:
Hi,
The execution finally succeeded, but I noticed that a new wlanX interface was created.
derek@raspberrypi:~/esp-hosted/esp_hosted_ng/host $ sudo bash rpi_init.sh spi ap_support
Enabling AP support
Building for spi protocol
rm: cannot remove 'spidev_disabler.dtbo': No such file or directory
bluetooth module inserted
Setting CONFIG_AP_SUPPORT to y
rm -rf *.o */*.o *.ko *.mod.c *.symvers *.order .*.cmd .tmp_versions
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/6.12.20+rpt-rpi-2712/build M=/home/derek/esp-hosted/esp_hosted_ng/host clean
make[1]: Entering directory '/usr/src/linux-headers-6.12.20+rpt-rpi-2712'
make[1]: Leaving directory '/usr/src/linux-headers-6.12.20+rpt-rpi-2712'
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/6.12.20+rpt-rpi-2712/build M=/home/derek/esp-hosted/esp_hosted_ng/host modules
make[1]: Entering directory '/usr/src/linux-headers-6.12.20+rpt-rpi-2712'
  CC [M]  /home/derek/esp-hosted/esp_hosted_ng/host/spi/esp_spi.o
  CC [M]  /home/derek/esp-hosted/esp_hosted_ng/host/esp_bt.o
  CC [M]  /home/derek/esp-hosted/esp_hosted_ng/host/main.o
  CC [M]  /home/derek/esp-hosted/esp_hosted_ng/host/esp_cmd.o
  CC [M]  /home/derek/esp-hosted/esp_hosted_ng/host/esp_utils.o
  CC [M]  /home/derek/esp-hosted/esp_hosted_ng/host/esp_cfg80211.o
  CC [M]  /home/derek/esp-hosted/esp_hosted_ng/host/esp_stats.o
  CC [M]  /home/derek/esp-hosted/esp_hosted_ng/host/esp_debugfs.o
  CC [M]  /home/derek/esp-hosted/esp_hosted_ng/host/esp_log.o
  LD [M]  /home/derek/esp-hosted/esp_hosted_ng/host/esp32_spi.o
  MODPOST /home/derek/esp-hosted/esp_hosted_ng/host/Module.symvers
  CC [M]  /home/derek/esp-hosted/esp_hosted_ng/host/esp32_spi.mod.o
  CC [M]  /home/derek/esp-hosted/esp_hosted_ng/host/.module-common.o
  LD [M]  /home/derek/esp-hosted/esp_hosted_ng/host/esp32_spi.ko
make[1]: Leaving directory '/usr/src/linux-headers-6.12.20+rpt-rpi-2712'
esp32 module inserted 
ESP32 host init successfully completed

mantriyogesh, 2025-07-23:
Glad that it worked fine for you. [rpi5_spi.zip](https://github.com/user-attachments/files/21384085/rpi5_spi.zip) was sufficient, or were there any more changes you needed, if I may know?

Please try to use exact compatible versions to avoid some race cases across version updates.

If you wish to use the release binary, please use the commit or tag attached to that release at the host as well.

mantriyogesh, 2025-07-23:
> Currently, I have only enabled SPI on the Raspberry Pi 5 using dtparam=spi=on and dtoverlay=disable-bt; no other settings have been configured.

Thanks for confirmation..

> Additionally, since the Raspberry Pi 5 already has a built-in wlan0, will executing rpi_init.sh automatically create a new interface as wlan1?

Yes.
https://github.com/espressif/esp-hosted/tree/master/esp_hosted_ng#31-user-guide

[image]

mantriyogesh, 2025-07-28:
Can you send full logs at ESP and host for finding some clue if something is off from the standard way?

Please attach text logs -
 - ESP full log from bootup. 
 - Linux dmesg log from bootup

derekwen168, 2025-07-29:
@mantriyogesh @Shreyas0-7 
Can I just ignore this error message?

mantriyogesh, 2025-07-29:
@derekwen168 Thanks for the logs. It’s a good sign that the internet is working fine.

[   54.125285] esp32_spi: process_rx_buf: offset_rcv[0] != exp[12], drop
[   56.125540] Bluetooth: hci1: command 0x1003 tx timeout
[   56.125540] Bluetooth: hci1: Opcode 0x1003 failed: -110
[   59.357693] esp32_spi: wait_and_decode_cmd_resp: Command[0xF] timed out
[   59.357714] esp32_spi: cmd_get_tx_power: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22

These messages indicate something isn’t quite right. Do they appear only once during events like a C3 reboot (and then stop), or are they recurring during normal network operation?

If these messages continue to appear even during regular operation, they will need to be addressed.

Also, do you observe these messages during the [RAW throughput test](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/Raw_TP_Testing.md)? This test is an important step to validate the transport performance in both directions.

mantriyogesh, 2025-07-29:
@derekwen168 
1. Can you please attach fiile, esp_hosted_ng/esp/esp_driver/network_adapter/sdkconfig

2. Also, please ensure that both the ESP and host sides are built from the same Git commit. If you’re unsure, it’s best to flash both from the latest master to avoid any mismatch.

Do you happen to have a logic analyzer with you?

@Shreyas0-7 
Please share the enabling debug logs for host and slave, if we can get hold of the issue using the logs.

derekwen168, 2025-07-29:
@mantriyogesh 
Okay, I'll send it again, thank you.

[sdkconfig.zip](https://github.com/user-attachments/files/21482666/sdkconfig.zip)

Shreyas0-7, 2025-07-29:
Hey @derekwen168 please find the highlighted option 

[image]

mantriyogesh, 2025-07-31:
The issue you observe as per the log is in SPI host RX case (need to check the slave log at that point, if that was also wrong)

For a valid buffer, we populate the offset ([here](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/esp/esp_driver/network_adapter/main/spi_slave_api.c#L648)

For dummy buffer, we keep it null ([here](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/esp/esp_driver/network_adapter/main/spi_slave_api.c#L341-L343)

But it shouldn't matter, as we would be ignoring that host RX packet ([here](https://github.com/espressif/esp-hosted/blob/fa92b8068a2109ec2d0c6527fab5611cfec0e372/esp_hosted_ng/host/spi/esp_spi.c#L226)

mantriyogesh, 2025-07-31:
@Shreyas0-7 PTAL in slave log in https://github.com/espressif/esp-hosted/issues/637#issuecomment-3132288177.

If in case you have a full log, it would be good to attach, @derekwen168 . especially when this error started:

```
E (642650) FW_MAIN: Sending data failed=257
```

If you don't have it right now, it's okay. At least this hints at (some) problem.

mantriyogesh, 2025-07-31:
This log,
 
```
E (642650) FW_MAIN: Sending data failed=257
```

It looks like a false positive, @derekwen168. Although, we should not print this log this frequently.

We still might need to check the SPI level Tx and Rx logs.

Shreyas0-7, 2025-08-05:
Hey attaching patch which will help us to debug issue more efficiently
please apply the patch in `/LOCAL_PATH/esp_hosted/`
please unzip the patch
[spi.zip](https://github.com/user-attachments/files/21596648/spi.zip)
also attaching logs which we encountered for bootup for you to compare

[esp_debug.log](https://github.com/user-attachments/files/21596640/esp_debug.log)
[host_debug.log](https://github.com/user-attachments/files/21596639/host_debug.log)

derekwen168, 2025-08-05:
@mantriyogesh @Shreyas0-7 
Both sides have now been updated to version 1.0.5.0.0. Please refer to the attached log files for details.

[esp_log.txt](https://github.com/user-attachments/files/21600313/esp_log.txt)
[host_log.txt](https://github.com/user-attachments/files/21600314/host_log.txt)

derekwen168, 2025-08-06:
@mantriyogesh 
Sure, I'll wait for your updated patch.

Shreyas0-7, 2025-08-07:
Hey @derekwen168 
please use this patch
[spi.zip](https://github.com/user-attachments/files/21654197/spi.zip)

mantriyogesh, 2025-08-07:
@derekwen168 With attached logs,
1. SPI rx and tx are well-synchronised at both places (no issues seen from logs)
2. Failed to see:
```
[ 288.882866] esp32_spi: process_rx_buf: offset_rcv[0] != exp[12], drop
```
3. Failed to see:
```
E (642650) FW_MAIN: Sending data failed=257
```

[a] Can you please retry reproducing the issue and share the logs for that from bootup for both sides?
[b] Please expect a considerable throughput drop, as a lot of logging is involved.

derekwen168, 2025-08-07:
@mantriyogesh 
I'll reflash version 1.0.5.0 without applying the spi.patch to see if the issue still occurs.

mantriyogesh, 2025-08-08:
> spi.patch file removed the original log message: esp32_spi: process_rx_buf: offset_rcv[0] != exp[12], drop.

No. I do not see this.  Check the image below. Please check your local changes, if any.

[image]

I attached the patch for your reference with the line number. There is no such line deleted in this patch.

```diff
     1	diff --git a/esp_hosted_ng/esp/esp_driver/network_adapter/main/spi_slave_api.c b/esp_hosted_ng/esp/esp_driver/network_adapter/main/spi_slave_api.c
     2	index c6a31899..3d4953fd 100644
     3	--- a/esp_hosted_ng/esp/esp_driver/network_adapter/main/spi_slave_api.c
     4	+++ b/esp_hosted_ng/esp/esp_driver/network_adapter/main/spi_slave_api.c
     5	@@ -341,6 +341,7 @@ static uint8_t * get_next_tx_buffer(uint32_t *len)
     6	     header->if_type = 0xF;
     7	     header->if_num = 0xF;
     8	     header->len = 0;
     9	+    ESP_LOGI("SPI TX", "--- NEW TX ZERO packet ---");
    10
    11	     if (len) {
    12	         *len = 0;
    13	@@ -370,6 +371,7 @@ static int process_spi_rx(interface_buffer_handle_t *buf_handle)
    14	     offset = le16toh(header->offset);
    15
    16	     if (!len || (len > RX_BUF_SIZE)) {
    17	+    ESP_LOGI("SPI RX", "--- New RX packet with zero len received---");
    18	         return -1;
    19	     }
    20
    21	@@ -393,7 +395,8 @@ static int process_spi_rx(interface_buffer_handle_t *buf_handle)
    22	     buf_handle->free_buf_handle = esp_spi_read_done;
    23	     buf_handle->payload_len = le16toh(header->len) + offset;
    24	     buf_handle->priv_buffer_handle = buf_handle->payload;
    25	-
    26	+    ESP_LOGI("SPI RX", "--- New RX packet  len %d---",buf_handle->payload_len);
    27	+    ESP_LOG_BUFFER_HEXDUMP("SPI RX", buf_handle->payload, MIN(buf_handle->payload_len, 32), ESP_LOG_INFO);
    28	     if (header->if_type == ESP_INTERNAL_IF) {
    29	         ret = xQueueSend(spi_rx_queue[PRIO_Q_HIGH], buf_handle, portMAX_DELAY);
    30	     } else if (header->if_type == ESP_HCI_IF) {
    31	@@ -657,6 +660,9 @@ static int32_t esp_spi_write(interface_handle_t *handle, interface_buffer_handle
    32	                                                 offset + buf_handle->payload_len));
    33	 #endif
    34
    35	+    ESP_LOGI("SPI TX", "ESP -> Host len %d",buf_handle->payload_len);
    36	+    ESP_LOG_BUFFER_HEXDUMP("SPI TX", tx_buf_handle.payload, MIN(buf_handle->payload_len, 32), ESP_LOG_INFO);
    37	+
    38	     if (header->if_type == ESP_INTERNAL_IF) {
    39	         ret = xQueueSend(spi_tx_queue[PRIO_Q_HIGH], &tx_buf_handle, portMAX_DELAY);
    40	     } else if (header->if_type == ESP_HCI_IF) {
    41	diff --git a/esp_hosted_ng/host/spi/esp_spi.c b/esp_hosted_ng/host/spi/esp_spi.c
    42	index 51544a56..5317ad5b 100644
    43	--- a/esp_hosted_ng/host/spi/esp_spi.c
    44	+++ b/esp_hosted_ng/host/spi/esp_spi.c
    45	@@ -223,9 +223,15 @@ static int process_rx_buf(struct sk_buff *skb)
    46	 	header = (struct esp_payload_header *) skb->data;
    47
    48	 	if (header->if_type >= ESP_MAX_IF) {
    49	+	esp_info("--- New RX packet---");
    50	+	esp_info("--- ZERO packet received ---");
    51	 		return -EINVAL;
    52	 	}
    53	-
    54	+	len = le16_to_cpu(header->len);
    55	+	len += sizeof(struct esp_payload_header);
    56	+	esp_info("--- New RX packet--- %d",len);
    57	+    esp_hex_dump("RX packet: ", skb->data, min((u32)len, 32u));
    58	+	esp_info("raw header->offset: 0x%04x\n", header->offset);
    59	 	offset = le16_to_cpu(header->offset);
    60
    61	 	/* Validate received SKB. Check len and offset fields */
    62	@@ -235,14 +241,14 @@ static int process_rx_buf(struct sk_buff *skb)
    63	 		return -EINVAL;
    64	 	}
    65
    66	-	len = le16_to_cpu(header->len);
    67	 	if (!len) {
    68	+		esp_err("len is 0, drop\n");
    69	 		return -EINVAL;
    70	 	}
    71
    72	-	len += sizeof(struct esp_payload_header);
    73
    74	 	if (len > SPI_BUF_SIZE) {
    75	+		esp_err("len %d > SPI_BUF_SIZE %d, drop\n", len, SPI_BUF_SIZE);
    76	 		return -EINVAL;
    77	 	}
    78
    79	@@ -269,6 +275,7 @@ static int process_rx_buf(struct sk_buff *skb)
    80	 	return 0;
    81	 }
    82
    83	+
    84	 static void esp_spi_work(struct work_struct *work)
    85	 {
    86	 	struct spi_transfer trans;
    87	@@ -323,9 +330,11 @@ static void esp_spi_work(struct work_struct *work)
    88	 			/* Configure TX buffer if available */
    89
    90	 			if (tx_skb) {
    91	+				esp_info("--- New TX packet %d---", tx_skb->len);
    92	+				esp_hex_dump("TX packet: ", tx_skb->data, 32);
    93	 				trans.tx_buf = tx_skb->data;
    94	-				esp_hex_dump_verbose("tx: ", trans.tx_buf, 32);
    95	 			} else {
    96	+				esp_info("--- ZERO packet sent ---");
    97	 				tx_skb = esp_alloc_skb(SPI_BUF_SIZE);
    98	 				trans.tx_buf = skb_put(tx_skb, SPI_BUF_SIZE);
    99	 				memset((void *)trans.tx_buf, 0, SPI_BUF_SIZE);
   100
```

derekwen168, 2025-08-08:
@mantriyogesh 
So what should I do at this point?
Both sides are using version 1.0.5.0 — when the spi.patch is applied, the message esp32_spi: process_rx_buf: offset_rcv[0] != exp[12], drop does not appear;
but if the patch is not applied, the message shows up.

mantriyogesh, 2025-08-11:
@Shreyas0-7 Have we tested this on Raspberry Pi 5 or 4?

Can we exactly replicate the use scenario for **wired** set up with **spi**, with ESP32-**C3** and Raspberry-Pi-**5** similar to images and connections mentioned as per https://github.com/espressif/esp-hosted/issues/637#issuecomment-3099827919?

derekwen168, 2025-08-16:
@mantriyogesh 
Is there any update?

mantriyogesh, 2025-08-16:
@Shreyas0-7, can you please confirm if this issue is observed on our side?

derekwen168, 2025-08-19:
@Shreyas0-7 
Can you test it using AP mode?
Thanks.

Shreyas0-7, 2025-08-19:
@derekwen168
Sure give me some time, will try AP mode too

derekwen168, 2025-08-21:
@Shreyas0-7 
I’ll re-flash the latest version again. If I encounter any issues, I’ll let you know. Thanks.

derekwen168, 2025-08-21:
@Shreyas0-7 
I’m flashing the ESP32-C3 in a Windows 11 environment — could this have any impact?

derekwen168, 2025-08-21:
@Shreyas0-7 
Could you please test by connecting a phone or another device to this AP, and then try accessing the internet?

derekwen168, 2025-08-21:
@Shreyas0-7 
From the log you provided, it looks like it’s only from the startup. Could I trouble you to actually transfer some data in AP mode?

Shreyas0-7, 2025-08-28:
Sure @derekwen168 will check for data transfer for AP. Also I noticed `FW_MAIN: Sending data failed=257` which is caused by ESP_ERR_NO_MEM returned by `esp_wifi_internal_tx`. Can you please print available heap memory at the time you receive this error on slave(esp side).

derekwen168, 2025-09-08:
@Shreyas0-7 
Which directory should I run this patch in?

Shreyas0-7, 2025-09-08:
Hey @derekwen168 please apply this patch in `/LOCAL_PATH/esp-hosted/` directory.
After applying the patch
`cd esp-hosted/esp_hosted_ng/esp/esp_driver/network_adapter/`
`idf.py menuconfig`
search for`ESP_SPI_DEASSERT_HS_ON_CS` and enable it
`idf.py build`

derekwen168, 2025-09-08:
@Shreyas0-7 
Because the Pi5’s built-in Wi-Fi is currently used as a client and the ESP32-C3 is acting as an AP, disabling NetworkManager would cause the Pi5’s built-in Wi-Fi to stop working.

Shreyas0-7, 2025-09-08:
@derekwen168 you can use this command to exclude interface from Network Manager
`nmcli device set wlan0 managed no`
Substitue your interface in place of wlan0

derekwen168, 2025-09-09:
Hi @Shreyas0-7 
I’m currently using the Pi5’s built-in Wi-Fi to connect to an Internet-enabled Wi-Fi router.
The ESP32 is set up as an AP for the phone to connect to.
I’ve configured iptables between the Pi5 and the ESP32, so when the phone connects to the ESP32 AP, it ultimately accesses the Internet through the Wi-Fi router.

derekwen168, 2025-09-09:
Hi @Shreyas0-7 
Currently, we haven’t seen any process_rx_buf drop issues, but will “Sending data failed=257” cause data transmission to be interrupted?
Also, if the issue is due to the TX buffer being exhausted, is there a way to resolve it?

Shreyas0-7, 2025-09-09:
@derekwen168 Glad to hear that, 
Yes, it can cause problems in data transmission as the packet will be dropped on esp firmware.
You can increase `ESP_WIFI_DYNAMIC_TX_BUFFER_NUM` in idf.py menuconfig, but it may reduce the memory available for the application.

derekwen168, 2025-09-09:
Hi @Shreyas0-7 
Alright, it seems that, so far, this issue cannot be resolved.