espressif/esp-hosted #642: ESP32 Hosted NG sometimes hangs on unload

icvidal, 2025-08-04:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

rarely

### Expected behavior

On `rmmod esp_spi` the module should cleanly unload in a short time. 

### Actual behavior (suspected bug)

Sometimes, `rmmod` hangs while running `flush_workqueue` as part of `spi_exit`. 
The kernel prints messages like the following:
```
Aug  1 22:07:52 dreamie user.err kernel: [134174.742830] INFO: task rmmod:12470 blocked for more than 122 seconds.
```

Some extra debug statements point to flushing `spi_workqueue`

### Error logs or terminal output

```shell
/sys/kernel/config/usb_gadget # dmesg
[134297.617143]       Not tainted 5.10.21 #1
[134297.617150] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[134297.617157] task:rmmod           state:D stack:    0 pid:12470 ppid: 12456 flags:0x00100000
[134297.617175] Stack : 80db0000 00000000 80f70000 800a7860 80dc8380 813b5260 0000024e 00000000
[134297.617213]         00000004 a91b9318 80dbcfb4 82b43100 7fffffff 83bcbe08 00000002 80db0000
[134297.617250]         c0010000 c0010000 84e55800 80b0493c 80dd0000 80db0000 80f66474 00000006
[134297.617298]         83bcbe0c 80b08b30 80f684b4 00000400 0000024d 00000000 2cc4bd09 7fffffff
[134297.617345]         c2000018 000030b6 00000000 00000000 00000000 00000000 84e55800 a91b9318
[134297.617379]         ...
[134297.617391] Call Trace:
[134297.617410] [<80b04130>] __schedule+0x438/0xbbc
[134297.617424] [<80b0493c>] schedule+0x88/0x13c
[134297.617437] [<80b08b30>] schedule_timeout+0x200/0x3e8
[134297.617449] [<80b06038>] wait_for_completion+0xa4/0x11c
[134297.617461] [<8006307c>] flush_workqueue+0x124/0x5a4
[134297.617483] [<c0004db8>] spi_exit+0x400/0x424 [esp32_hosted_ng]
[134297.617502] [<c0005808>] esp_deinit_interface_layer+0x48/0x70 [esp32_hosted_ng]
[134297.617520] [<c001093c>] esp_exit+0x9c/0x10c [esp32_hosted_ng]
[134297.617539] [<800f5534>] sys_delete_module+0x1c4/0x2a8
[134297.617552] [<8002d138>] syscall_common+0x34/0x58
[134297.617562] 
[134420.491385] INFO: task rmmod:12470 blocked for more than 368 seconds.
[134420.491402]       Not tainted 5.10.21 #1
[134420.491407] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[134420.491414] task:rmmod           state:D stack:    0 pid:12470 ppid: 12456 flags:0x00100000
[134420.491429] Stack : 80db0000 00000000 80f70000 800a7860 80dc8380 813b5260 0000024e 00000000
[134420.491462]         00000004 a91b9318 80dbcfb4 82b43100 7fffffff 83bcbe08 00000002 80db0000
[134420.491515]         c0010000 c0010000 84e55800 80b0493c 80dd0000 80db0000 80f66474 00000006
[134420.491553]         83bcbe0c 80b08b30 80f684b4 00000400 0000024d 00000000 2cc4bd09 7fffffff
[134420.491586]         c2000018 000030b6 00000000 00000000 00000000 00000000 84e55800 a91b9318
[134420.491617]         ...
[134420.491629] Call Trace:
[134420.491646] [<80b04130>] __schedule+0x438/0xbbc
[134420.491658] [<80b0493c>] schedule+0x88/0x13c
[134420.491668] [<80b08b30>] schedule_timeout+0x200/0x3e8
[134420.491679] [<80b06038>] wait_for_completion+0xa4/0x11c
[134420.491690] [<8006307c>] flush_workqueue+0x124/0x5a4
[134420.491711] [<c0004db8>] spi_exit+0x400/0x424 [esp32_hosted_ng]
[134420.491728] [<c0005808>] esp_deinit_interface_layer+0x48/0x70 [esp32_hosted_ng]
[134420.491745] [<c001093c>] esp_exit+0x9c/0x10c [esp32_hosted_ng]
[134420.491760] [<800f5534>] sys_delete_module+0x1c4/0x2a8
[134420.491771] [<8002d138>] syscall_common+0x34/0x58
[... 269 lines cut ...]

### Project release version

latest

### System architecture

other (details in Additional context)

### Operating system

Linux

### Operating system version

5.10

### Shell

ZSH

### Additional context

We have around 20 devices continually running to reproduce this. After a few days, one or two random devices will reproduce it.
Note: we are suffering from some WiFi drops. When this happens, we execute a procedure where we remove the module, reload and reconnect. It is possible that both issues are connected.

mantriyogesh, 2025-08-05:
> Aug  1 22:03:42 gadget user.info kernel: [133925.268809] ESP: -> wait_event_interruptible_timeout
> Aug  2 00:03:47 gadget daemon.info ntpd[1103]: Deleting interface #3 wlan0, 192.168.107.12#123, interface stats: received=216, sent=222, dropped=0, active_time=74488 secs
> Aug  2 00:03:47 gadget daemon.info ntpd[1103]: 162.159.200.1 local addr 192.168.107.12 -> <null>
> Aug  2 00:03:47 gadget daemon.info ntpd[1103]: 195.95.153.43 local addr 192.168.107.12 -> <null>
> Aug  2 00:03:47 gadget daemon.info ntpd[1103]: 178.215.228.24 local addr 192.168.107.12 -> <null>
> Aug  1 22:03:47 gadget user.err kernel: [133930.274131] esp32_hosted_ng: wait_and_decode_cmd_resp: Command[0x12] timed out

OR even the very first one,
> Aug  1 22:03:42 gadget user.err kernel: [133925.234371] esp32_hosted_ng: wait_and_decode_cmd_resp: Command[0x6] timed out

Why does such a time-out occur in the system? Does the slave receive these events and respond back?
Is the response received but not decodable, because incorrect (bit-shifted buffer possibility?)

Any such 'cmd' is going to trigger esp_send_packet(), which then uses spi_workqueue to send and wait for a response.
it's important not to do 'close_data_path()' very early.

All these are just review comments (even to self) while browsing the code for this specific case.

@Shreyas0-7 is it possible to reproduce this issue reliably at our end?

@icvidal, sorry to bother, but would it be possible to try the latest master once (both sides same commit), if this issue is being reproduced?

mantriyogesh, 2025-08-07:
@icvidal ?

Shreyas0-7, 2025-08-07:
Hey @icvidal please find the patch which will help you debug spi transport
[spi.zip](https://github.com/user-attachments/files/21664410/spi.zip)

mantriyogesh, 2025-08-12:
Thanks for confirming the issue. If possible, could you send us the original file in an open domain, such as https://elixir.bootlin.com/linux/v6.15/source/drivers/spi/spi.c, and the patch you applied for your SoC? This would help us or others working on the same MPU to verify and reference for other SoCs.