espressif/esp-hosted #650: 使用esp_hosted_fg，把esp32做sdio从机进行连接时出现读取错误

chenzhihui374, 2025-08-20:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

正常读写通信

### Actual behavior (suspected bug)

读取从机数据失败

### Error logs or terminal output

```shell
[  327.073398] jzmmc_v1.2 jzmmc_v1.2.0: card inserted, state=0
[  328.338089] mmc0: queuing unknown CIS tuple 0x01 (3 bytes)
[  328.346389] mmc0: queuing unknown CIS tuple 0x1a (5 bytes)
[  328.349805] mmc0: queuing unknown CIS tuple 0x1b (8 bytes)
[  328.424731] mmc0: queuing unknown CIS tuple 0x80 (1 bytes)
[  328.424807] mmc0: queuing unknown CIS tuple 0x81 (1 bytes)
[  328.424870] mmc0: queuing unknown CIS tuple 0x82 (1 bytes)
[  328.425897] mmc0: queuing unknown CIS tuple 0x80 (1 bytes)
[  328.425961] mmc0: queuing unknown CIS tuple 0x81 (1 bytes)
[  328.426022] mmc0: queuing unknown CIS tuple 0x82 (1 bytes)
[  328.426117] mmc0: queuing unknown CIS tuple 0x91 (3 bytes)
[  328.426151] mmc0: new SDIO card at address 0001
[  328.428898] esp32_sdio: esp_probe: ESP network device detected
[  328.428971] esp32_sdio: init_sdio_func: sdio_enable_func ret is : 0
[  328.431048] esp32_sdio: esp_probe: Using sdio clock[0 MHz]
[  328.923405] jzmmc_v1.2 jzmmc_v1.2.0: timeout 500ms op:53 r sz:512 state:0 STAT:0x1B000140 DMALEN:0x00000000 blks:0/1 clk:enable clk_gate:enable
[  329.423392] jzmmc_v1.2 jzmmc_v1.2.0: timeout 1000ms op:53 r sz:512 state:0 STAT:0x1B001141 DMALEN:0x00000000 blks:0/1 clk:enable clk_gate:enable
[  329.423432] jzmmc_v1.2 jzmmc_v1.2.0: polling status(0x00001000) error, op=53, status=0x1B001141
[  329.423493] esp32_sdio: init_context: Read: PACKET_LEN reg 1073041504 - Err[-5]
[  329.423505] esp32_sdio: esp_probe: ESP network device init context ret : -5
[  329.423679] esp_sdio: probe of mmc0:0001:1 failed with error -5
[  329.424108] esp_sdio: probe of mmc0:0001:2 failed with error -22
```

### Steps to reproduce the behavior

插入sdio设备就出错

### Project release version

最新

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Ubuntu 18.04

### Shell

sh

mantriyogesh, 2025-08-21:
### Log Analysis

* ESP32 is detected as an SDIO card (`mmc0:0001`).
* Probe shows **`Using sdio clock[0 MHz]`**, which is invalid.
* All subsequent CMD53 transfers time out → driver fails with **`-5` (`-EIO`, I/O error)** [errno-base.h](https://elixir.bootlin.com/linux/v6.16.1/source/include/uapi/asm-generic/errno-base.h#L9).
* The extra line:

  ```
  esp_sdio: probe of mmc0:0001:2 failed with error -22
  ```

  is expected and can be ignored (slot 2 not used in esp-hosted). Slot 1 fails due to legitimate issues.

### Next Step

Please check why the host controller isn’t providing a valid SDIO clock (0 Hz). This may be due to host driver behavior or device tree configuration.

Additionally, once the host SDIO issue is corrected, you can check the actual clock released by your base Linux SDIO driver using

```bash
cat /sys/class/mmc_host/mmc0/mmc0:0001/ios
```
Then, re-try to use the esp-hosted kernel module.

chenzhihui374, 2025-08-21:
您的邮件陈志辉已收到，谢谢来信！