espressif/esp-hosted #655: Port esp-hosted_ng to support spi Linux host kernel version 3.10.14 on chipset esp32c3

fycai000, 2025-08-25:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

Hello .
   As the up-to-date esp-hosted_ng project is not compatible for lower version kernel such as 3.10.14,  So I try to port the host spi linux drivers that it can cross-compile through already, Data is transmitted normally between esp32C3 and linux host by spi.
   But when I test wpa_supplicant tools to connect a station , the wifi authorization step has been failed.

   Could you help me to find out the problems?
  

[image]

[image]

[image]

mantriyogesh, 2025-08-25:
@kapilkedawat ptal..

fycai000, 2025-08-27:
Hi @Shreyas0-7 
When testing the embedded idf demo "softap_sta" in the project of hosted-ng ,it also cannot connect wifi, Whether it is normal or not ?
Thanks.

fy@cai:~/esp/esp_hosted_ng/esp/esp_driver/esp-idf/examples/wifi/softap_sta$ get_idf
Checking "python3" ...
Python 3.10.0
"python3" has been detected
Activating ESP-IDF 5.4
* Checking python version ... 3.10.0
* Checking python dependencies ... OK
* Deactivating the current ESP-IDF environment (if any) ... OK
* Establishing a new ESP-IDF environment ... OK
* Identifying shell ... bash
* Detecting outdated tools in system ... Found tools that are not used by active ESP-IDF version.
For removing old versions of riscv32-esp-elf-gdb, xtensa-esp32-elf, xtensa-esp32s2-elf, riscv32-esp-elf, xtensa-esp32s3-elf, openocd-esp32, xtensa-esp-elf-gdb, esp32ulp-elf use command 'python 
/home/fy/esp/esp_hosted_ng/esp/esp_driver/esp-idf/tools/idf_tools.py uninstall'
To free up even more space, remove installation packages of those tools.
Use option python /home/fy/esp/esp_hosted_ng/esp/esp_driver/esp-idf/tools/idf_tools.py uninstall --remove-archives.
* Shell completion ... Autocompletion code generated

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build
fy@cai:~/esp/esp_hosted_ng/esp/esp_driver/esp-idf/examples/wifi/softap_sta$ idf.py -p /dev/ttyUSB0 flash monitor flash
WARNING: Command "flash" is found in the list of commands more than once. Only first occurrence will be executed.
Executing action: flash
Running ninja in directory /home/fy/esp/esp_hosted_ng/esp/esp_driver/esp-idf/examples/wifi/softap_sta/build
Executing "ninja flash"...
[1/1] cd /home/fy/esp/esp_hosted_ng/es...ap_sta/build/bootloader/bootloader.bin
Bootloader binary size 0x5150 bytes. 0x2eb0 bytes (36%) free.
[8/10] Generating binary image from built executable
esptool.py v4.9.0
Creating esp32c3 image...
Merged 1 ELF section
Successfully created esp32c3 image.
Generated /home/fy/esp/esp_hosted_ng/esp/esp_driver/esp-idf/examples/wifi/softap_sta/build/softap_sta.bin
[9/10] cd /home/fy/esp/esp_hosted_ng/e...s/wifi/softap_sta/build/softap_sta.bin
softap_sta.bin binary size 0xc2190 bytes. Smallest app partition is 0x100000 bytes. 0x3de70 bytes (24%) free.
[9/10] cd /home/fy/esp/esp_hosted_ng/e...nents/esptool_py/run_serial_tool.cmake

esptool.py --chip esp32c3 -p /dev/ttyUSB0 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x0 bootloader/bootloader.bin 0x10000 softap_sta.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.9.0
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32-C3 (QFN32) (revision v0.3)
Features: WiFi, BLE, Embedded Flash 4MB (XMC)
Crystal is 40MHz
MAC: 70:04:1d:0d:1f:20
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x000d2fff...
Flash will be erased from 0x00008000 to 0x00008fff...
SHA digest in image updated
Compressed 20816 bytes to 13047...
Writing at 0x00000000... (100 %)
Wrote 20816 bytes (13047 compressed) at 0x00000000 in 0.7 seconds (effective 240.0 kbit/s)...
Hash of data verified.
Compressed 795024 bytes to 458172...
Writing at 0x000cc4c0... (100 %)
Wrote 795024 bytes (458172 compressed) at 0x00010000 in 15.1 seconds (effective 420.4 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 103...
Writing at 0x00008000... (100 %)
[... 106 lines cut ...]
I (3375) wifi:ap channel adjust o:1,1 n:6,2
I (3375) wifi:new:<6,0>, old:<1,1>, ap:<6,2>, sta:<6,0>, prof:1, snd_ch_cfg:0x0
I (3375) wifi:state: init -> auth (0xb0)
I (4385) wifi:state: auth -> init (0x200)
I (4395) wifi:new:<6,0>, old:<6,0>, ap:<6,2>, sta:<6,0>, prof:1, snd_ch_cfg:0x0
I (4395) wifi:new:<6,2>, old:<6,0>, ap:<6,2>, sta:<6,0>, prof:1, snd_ch_cfg:0x0
I (4395) wifi:state: init -> auth (0xb0)
I (5395) wifi:state: auth -> init (0x200)
I (5405) wifi:new:<6,0>, old:<6,2>, ap:<6,2>, sta:<6,0>, prof:1, snd_ch_cfg:0x0
I (5405) wifi:new:<6,2>, old:<6,0>, ap:<6,2>, sta:<6,0>, prof:1, snd_ch_cfg:0x0
I (5405) wifi:state: init -> auth (0xb0)
I (6415) wifi:state: auth -> init (0x200)
I (6425) wifi:new:<6,0>, old:<6,2>, ap:<6,2>, sta:<6,0>, prof:1, snd_ch_cfg:0x0
I (6425) wifi:new:<6,2>, old:<6,0>, ap:<6,2>, sta:<6,0>, prof:1, snd_ch_cfg:0x0
I (6425) wifi:state: init -> auth (0xb0)
I (7425) wifi:state: auth -> init (0x200)
I (7435) wifi:new:<6,0>, old:<6,2>, ap:<6,2>, sta:<6,0>, prof:1, snd_ch_cfg:0x0
I (7435) wifi:new:<6,2>, old:<6,0>, ap:<6,2>, sta:<6,0>, prof:1, snd_ch_cfg:0x0
I (7435) wifi:state: init -> auth (0xb0)
I (8445) wifi:state: auth -> init (0x200)
I (8455) wifi:new:<6,0>, old:<6,2>, ap:<6,2>, sta:<6,0>, prof:1, snd_ch_cfg:0x0
I (8455) wifi:new:<6,2>, old:<6,0>, ap:<6,2>, sta:<6,0>, prof:1, snd_ch_cfg:0x0
I (8455) wifi:state: init -> auth (0xb0)
I (9455) wifi:state: auth -> init (0x200)
I (9465) wifi:new:<6,0>, old:<6,2>, ap:<6,2>, sta:<6,0>, prof:1, snd_ch_cfg:0x0