espressif/esp-hosted #657: crash all of a sudden P4+C5

savenlid, 2025-09-02:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

work as it usually does

### Actual behavior (suspected bug)

Crash.

I can understand that hosted is maybe unhappy, but would be better if it did not crash and instead explained the reason behind it being unhappy.

this way we boot loop.

One weakness in my design is that C5-EN pin dont have a pull-up, its depends on being driven from C4.
If c4-hosted part never drives it , it will float I assume.
But regardless I would prefer not to crash.

### Error logs or terminal output

```shell
I (4313) H_API: ESP-Hosted starting. Hosted_Tasks: prio:23, stack: 5120 RPC_task_stack: 5120
I (4321) H_API: ** add_esp_wifi_remote_channels **
I (4325) transport: Add ESP-Hosted channel IF[1]: S[0] Tx[0x4000c0ac] Rx[0x4001ca02]
--- 0x4000c0ac: transport_drv_sta_tx at C:/workspace/batteries-chargers/esp32/ch1_p4/managed_components/espressif__esp_hosted/host/drivers/transport/transport_drv.c:245
--- 0x4001ca02: esp_wifi_remote_channel_rx at C:/workspace/batteries-chargers/esp32/ch1_p4/managed_components/espressif__esp_wifi_remote/esp_wifi_remote_net.c:19
I (4333) transport: Add ESP-Hosted channel IF[2]: S[0] Tx[0x4000bff0] Rx[0x4001ca02]
--- 0x4000bff0: transport_drv_ap_tx at C:/workspace/batteries-chargers/esp32/ch1_p4/managed_components/espressif__esp_hosted/host/drivers/transport/transport_drv.c:275
--- 0x4001ca02: esp_wifi_remote_channel_rx at C:/workspace/batteries-chargers/esp32/ch1_p4/managed_components/espressif__esp_wifi_remote/esp_wifi_remote_net.c:19
sdio_mempool_create free:33651528 min-free:33651528 lfb-def:33030144 lfb-8bit:33030144

assert failed: xTaskCreateStaticPinnedToCore freertos_tasks_c_additions.h:299 (xPortcheckValidStackMem(puxStackBuffer))
Core  0 register dump:
MEPC    : 0x4ff00d2c  RA      : 0x4ff0c716  SP      : 0x4ff3c9a0  GP      : 0x4ff19680
--- Stack dump detected
--- 0x4ff00d2c: panic_abort at C:/Espressif/frameworks/esp-idf-v5.5-2/components/esp_system/panic.c:483
--- 0x4ff0c716: __ubsan_include at C:/Espressif/frameworks/esp-idf-v5.5-2/components/esp_system/ubsan.c:311
TP      : 0x00000000  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x00000050  S1      : 0x00000001  A0      : 0x4ff3c9dc  A1      : 0x4ff20a41
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x4ff45000
A6      : 0x0000000c  A7      : 0x76757473  S2      : 0x00000009  S3      : 0x4ff3cb27
S4      : 0x4ff20a40  S5      : 0x00000800  S6      : 0x00000000  S7      : 0x00000001
S8      : 0x00110020  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00001880  MTVEC   : 0x4ff00003  MCAUSE  : 0x00000002  MTVAL   : 0x00000000
--- 0x4ff00003: _vector_table at ??:?
MHARTID : 0x00000000

--- Warning: Failed to run gdb_panic_server.py script: Command '['riscv32-esp-elf-gdb', '--batch', '-n', 'C:\\workspace\\batteries-chargers\\esp32\\ch1_p4\\build\\acchg1.elf', '-ex', 'add-symbol-file C:\\workspace\\batteries-chargers\\esp32\\ch1_p4\\build\\bootloader\\bootloader.elf', '-ex', 'target remote | "C:\\Espressif\\python_env\\idf5.5_py3.11_env\\Scripts\\python.exe" -m esp_idf_panic_decoder --target esp32p4 "C:\\Users\\toonsav\\AppData\\Local\\Temp\\tmpmng7eagm"', '-ex', 'bt']' returned non-zero exit status 1.
--- Warning: b'add symbol table from file "C:workspacebatteries-chargersesp32ch1_p4buildbootloaderbootloader.elf"\r\nC:workspacebatteries-chargersesp32ch1_p4buildbootloaderbootloader.elf: No such file or directory.\r\nerror starting child process \'C:\\Espressif\\python_env\\idf5.5_py3.11_env\\Scripts\\python.exe -m esp_idf_panic_decoder --target esp32p4 C:\\Users\\toonsav\\AppData\\Local\\Temp\\tmpmng7eagm\': CreateProcess: No such file or directory\r\nNo stack.\r\n'

Stack memory:
4ff3c9a0: 0x00000804 0x0000000c 0x40125490 0x4ff169ba 0x4ff2033c 0x40125490 0x4ff20320 0x400f2dfc
4ff3c9c0: 0x4ff2034c 0x4ff3c9d4 0x4ff20350 0x400f2ed8 0x4ff20a40 0x00393932 0x4ff29fd0 0x65737361
4ff3c9e0: 0x66207472 0x656c6961 0x78203a64 0x6b736154 0x61657243 0x74536574 0x63697461 0x6e6e6950
4ff3ca00: 0x6f546465 0x65726f43 0x65726620 0x6f747265 0x61745f73 0x5f736b73 0x64615f63 0x69746964
4ff3ca20: 0x2e736e6f 0x39323a68 0x78282039 0x74726f50 0x63656863 0x6c61566b 0x74536469 0x4d6b6361
4ff3ca40: 0x70286d65 0x74537875 0x426b6361 0x65666675 0x00292972 0x000e5020 0x4ff00402 0x4ff3cb14
4ff3ca60: 0x00010000 0x000e5020 0x4ff00402 0x4ff3cb24 0x00000804 0x00000168 0x4ff29fd0 0x4ff1474a
4ff3ca80: 0x00000804 0x00000168 0x4ff29c50 0x4ff1478a 0x00000000 0x00000000 0x4ff29fbc 0x4ff0134c
4ff3caa0: 0x00000804 0x00000000 0x00000000 0x4f000000 0x00010000 0x00000804 0x40113c84 0x4ff0e666
4ff3cac0: 0x7fffffff 0x30100508 0x4ff3aa54 0x4ff11c96 0x0000000c 0x00000000 0x4ff43b8c 0x4ff0111e
4ff3cae0: 0x4ff3cb28 0x4ff3cb2c 0x30100508 0x4ff0deb0 0x00010000 0x000e5020 0x4ff00402 0x0004c790
4ff3cb00: 0x00010000 0x000e5020 0x4ff00402 0x00160020 0x0000000c 0x00300000 0x00000000 0x4ff0e7bc
4ff3cb20: 0x454c4449 0x4ff3aa54 0x30100508 0x00000800 0x00010000 0x00300000 0x00000000 0x4ff11dec
4ff3cb40: 0x0000000c 0x00300000 0x00000000 0x400e0f3a 0x0000000c 0x02625a00 0x00000000 0x40001cec
4ff3cb60: 0x0000000c 0x02625a00 0x00000000 0x4ff00766 0x5012d000 0x00000010 0x00000000 0x01f3cb9c
4ff3cb80: 0x5f0207e9 0x4ff00402 0x000000ee 0x64000012 0x0000c700 0x01000000 0x0000000c 0x0000000c
4ff3cba0: 0x400f0020 0x40000020 0xffff0000 0x4ff2e04c 0x595b8e34 0x4ff3cd50 0x4ff3cc10 0x00110000
4ff3cbc0: 0xce1d627f 0x000f0000 0x00110020 0x00160020 0x400f0020 0x40000020 0x0004c790 0x000e5020
4ff3cbe0: 0x00007900 0x4fc1da4c 0x00000000 0x4ff40000 0x00007900 0x4fc1da4c 0x0000ffff 0xffffffff

4ff3cc20: 0x4ff00402 0x000000ee 0x64000012 0x0000c700 0x01000000 0x400f0020 0x0004c790 0x30100000
4ff3cc40: 0x00000388 0x4ff00000 0x000034d0 0x40000020 0x000e5020 0x4ff034d0 0x000159a0 0x4ff18e80
--- 0x4ff00402: call_start_cpu0 at C:/Espressif/frameworks/esp-idf-v5.5-2/components/esp_system/port/cpu_start.c:357
--- 0x30100000: clk_ll_mpll_disable at C:/Espressif/frameworks/esp-idf-v5.5-2/components/hal/esp32p4/include/hal/clk_tree_ll.h:165
--- (inlined by) rtc_clk_mpll_disable at C:/Espressif/frameworks/esp-idf-v5.5-2/components/esp_hw_support/port/esp32p4/rtc_clk.c:586
--- 0x4ff00000: _vector_table at ??:?
--- 0x40000020: esp_app_format_init_elf_sha256 at C:/Espressif/frameworks/esp-idf-v5.5-2/components/esp_app_format/esp_app_desc.c:88
--- 0x4ff034d0: uart_rx_intr_handler_default at C:/Espressif/frameworks/esp-idf-v5.5-2/components/esp_driver_uart/src/uart.c:1279
4ff3cc60: 0x00008268 0x50108080 0x00000020 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff3cc80: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff3cca0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00110020 0x0015c7b8 0x0015cb48
4ff3ccc0: 0x00160020 0x00245048 0x0025a9f0 0x00262c60 0x00000000 0x00000000 0x00000000 0x00000000
4ff3cce0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00152cb0 0x595b8e34 0x55dfc5d6
4ff3cd00: 0x5e589db6 0x5f7f0dae 0xce1d627f 0x0c26e7a7 0x3934c00a 0x4de364d4 0x00000000 0x00010000
4ff3cd20: 0xffffffff 0xffffffff 0xffffffff 0x4ff40000 0x00000000 0x00000024 0x00000000 0x4ff29f56
--- 0x4ff29f56: _regi2c_ctrl_ll_master_enable_clock at C:/Espressif/frameworks/esp-idf-v5.5-2/components/hal/esp32p4/include/hal/regi2c_ctrl_ll.h:27
--- (inlined by) bootloader_hardware_init at C:/Espressif/frameworks/esp-idf-v5.5-2/components/bootloader_support/src/esp32p4/bootloader_esp32p4.c:98
--- (inlined by) bootloader_init at C:/Espressif/frameworks/esp-idf-v5.5-2/components/bootloader_support/src/esp32p4/bootloader_esp32p4.c:132
4ff3cd40: 0xf7f63be9 0xcafe963e 0x00000000 0x00000000 0x0010d000 0x00002000 0x00000000 0x00000000
4ff3cd60: 0x00000000 0x00000000 0x00110000 0x00400000 0x00510000 0x00400000 0x00000000 0x00000000
4ff3cd80: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
```

### Steps to reproduce the behavior

just boot.

it came suddenly.
verified on a seconds pcba to be the same.
surely I have changed something I assume, but not related to hosted-wifi.

### Project release version

idf 5.5, latest hosted versions  on both P4 and C5

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

windows 11

### Shell

ZSH

savenlid, 2025-09-02:
If I disable my PSRAM in the P4 ( 32MB ) it works again.

It may or may not have a relation to esp-hosted-mcu.
Regardless its a problem to not be able to use the 32MB , expensive memory :-)  

I (491) H_API: ** add_esp_wifi_remote_channels **
I (496) transport: Add ESP-Hosted channel IF[1]: S[0] Tx[0x4000ba1e] Rx[0x4001c414]
--- 0x4000ba1e: transport_drv_sta_tx at C:/workspace/batteries-chargers/esp32/ch1_p4/managed_components/espressif__esp_hosted/host/drivers/transport/transport_drv.c:245
--- 0x4001c414: esp_wifi_remote_channel_rx at C:/workspace/batteries-chargers/esp32/ch1_p4/managed_components/espressif__esp_wifi_remote/esp_wifi_remote_net.c:19
I (503) transport: Add ESP-Hosted channel IF[2]: S[0] Tx[0x4000b962] Rx[0x4001c414]
--- 0x4000b962: transport_drv_ap_tx at C:/workspace/batteries-chargers/esp32/ch1_p4/managed_components/espressif__esp_hosted/host/drivers/transport/transport_drv.c:275
--- 0x4001c414: esp_wifi_remote_channel_rx at C:/workspace/batteries-chargers/esp32/ch1_p4/managed_components/espressif__esp_wifi_remote/esp_wifi_remote_net.c:19
sdio_mempool_create free:100924 min-free:100924 lfb-def:61440 lfb-8bit:61440

I (519) H_SDIO_DRV: sdio_data_to_rx_buf_task started
I (519) main_task: Started on CPU0
I (529) main_task: Calling app_main()
I (529) Main: Chip revision: 103
I (529) Main: Major chip version: 1
I (539) Main: Minor chip version: 3
I (599) NVS: Secure NVS partition initialized successfully
I (629) FILE_SYSTEMS: Partition size (storage): total: 4194304, used: 1781760
I (629) FILE_SYSTEMS: Partition size (storage2): total: 16777216, used: 8192

savenlid, 2025-09-03:
I can send a download link to the complete project on  your email. 

If that's OK?

But without PSRAM enabled this worked for months. 

I will provide more details once I get out of the car.

But in general everything should be latest greatest on both p4 and c5.

goaarvoh, 2026-08-30:
Root cause for this assert, with a fix — I hit the identical failure on an ESP32-P4 (rev v1.3) and traced it. It matches @savenlid's observation that disabling PSRAM makes it go away.

## Cause

`host/port/esp/freertos/src/port_esp_hosted_host_os.c`, in `hosted_thread_create()`:

```c
StaticTask_t *task_buffer = heap_caps_calloc(1, sizeof(StaticTask_t), MALLOC_CAP_INTERNAL);   // correct
...
StackType_t *thread_stack = heap_caps_malloc(tstack_size, MALLOC_CAP_SPIRAM);                 // <-- the bug
...
*thread_handle = xTaskCreateStatic(..., tstack_size, sr_arg, tprio, thread_stack, task_buffer);
```

The task control block is allocated from internal RAM, but the **task stack** is allocated from PSRAM. FreeRTOS on ESP-IDF requires a task stack to live in internal RAM and enforces it in `xTaskCreateStaticPinnedToCore` via `xPortcheckValidStackMem()` — hence the assert.

**Why it only appears with PSRAM enabled:** with `CONFIG_SPIRAM` off there is no SPIRAM pool, the allocation returns `NULL`, and the function takes its existing early-out path. So the bug is invisible without PSRAM, which is exactly the "works when I disable PSRAM" behaviour reported above.

## Fix

```diff
-	StackType_t *thread_stack = heap_caps_malloc(tstack_size, MALLOC_CAP_SPIRAM);
+	StackType_t *thread_stack = heap_caps_malloc(tstack_size,
+			MALLOC_CAP_INTERNAL | MALLOC_CAP_8BIT);
```

## ⚠ The fix alone is not sufficient — second failure

Moving the stacks into internal RAM (~10 KB, for `Hosted_Tasks: stack 5120` + `RPC_task_stack: 5120`) then leaves the SDIO transport mempool short, and the boot fails differently:

```
E HS_MP: mempool create failed: no mem
assert failed: sdio_mempool_create sdio_drv.c:258 (buf_mp_g)
```

That second half needs no patch — it is already an option in this component:

```
CONFIG_ESP_HOSTED_MEMPOOL_PREFER_SPIRAM=y
```

The two changes move different things in opposite directions, each to where it is allowed to be: transport **buffers** may live in PSRAM on the ESP32-P4 because GDMA reaches it through cache; **stacks** may not. Note `hosted_malloc_align()` in the same file already prefers SPIRAM with an internal fallback and is correct as-is — it should not be changed.

## A red herring, so nobody else spends time on it

`CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY` looks like the culprit and is not. Setting it to `n` changes nothing here, because this allocation names `MALLOC_CAP_SPIRAM` explicitly rather than leaving placement to the config. Tested on hardware before the real cause was found.

## Verified

- esp_hosted **2.12.12**, esp_wifi_remote 1.6.4, ESP-IDF **v5.4**, target esp32p4 (rev v1.3), slave esp32c6 over SDIO
- `CONFIG_SPIRAM=y`, `CONFIG_SPIRAM_MODE_HEX=y`, `CONFIG_SPIRAM_SPEED_200M=y`, 32 MB PSRAM
- Before: boot loop inside ESP-IDF init, before `app_main` — 13 boots in a 25 s serial capture, 12 of them `SW_CPU_RESET`
- After patch + `MEMPOOL_PREFER_SPIRAM=y`: **20/20 consecutive boot cycles clean, 0 failures**, hosted initialising normally with PSRAM at 200 MHz

Happy to open a PR with the one-line change if that is useful.