espressif/esp-hosted #658: OpenWrt-22.03 ESP-Hosted-NG (SDIO) with ESP32-C5-DevKitC-1 - Host driver reports wlan0 MAC as `00:00:00:00:00:00`

san-av, 2025-09-03:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

# Issue: Host driver reports wlan0 MAC as `00:00:00:00:00:00`

## Summary
Host driver loads and wlan0 interface is created. WiFi AP scanning works, but connection fails because the interface MAC is always `00:00:00:00:00:00`.

## Setup
- **Host:** NUC980 IoT board, OpenWrt 22.03, Linux 5.10.140  
- **Slave:** ESP32-C5-DevKitC-1 V1.2 (SDIO 4-bit mode)  
- **Firmware:** ESP-Hosted-NG (`esp32c5_ng_beta_support` branch, NG-1.0.4.0.4)  
- **Host driver:** Built as OpenWrt module from same repo/branch  
- **Hardware wiring:** 75 kΩ pull-ups on CMD/DAT0–DAT3, 3–4 cm cladded wires, custom adapter PCB with ground plane  

## Logs (host)
```text
root@OpenWrt:/# insmod esp32_sdio_c5 clockspeed=50                                                                   
[  909.749741] esp32_sdio_c5: esp_probe: ESP network device detected                                    
[  909.756491] esp32_sdio_c5: get_firmware_data: Rx Pre ====== 0                                        
[  909.763816] esp32_sdio_c5: get_firmware_data: Rx Pos ======  0
[  909.770949] esp32_sdio_c5: get_firmware_data: Tx Pre ======  0
[  909.776766] esp32_sdio_c5: get_firmware_data: Tx Pos ======  10
[  909.785566] esp32_sdio_c5: process_esp_bootup_event: Received ESP bootup event
[  909.793991] esp32_sdio_c5: process_event_esp_bootup: Bootup Event tag: 3
[  909.802699] esp32_sdio_c5: esp_validate_chipset: Chipset=ESP32-C5 ID=17 detected over SDIO
[  909.811124] esp32_sdio_c5: process_event_esp_bootup: Bootup Event tag: 0
[  909.817799] esp32_sdio_c5: process_event_esp_bootup: Bootup Event tag: 1
[  909.824641] esp32_sdio_c5: process_fw_data: ESP chipset's last reset cause:
[  909.831699] esp32_sdio_c5: print_reset_reason: POWERON_RESET
[  909.837341] esp32_sdio_c5: check_esp_version: ESP-Hosted Version: NG-1.0.4.0.4
[  909.848941] esp32_sdio_c5: probe of mmc1:0001:2 failed with error -22
[  909.857824] esp32_sdio_c5: esp_reg_notifier: Driver init is ongoing
root@OpenWrt:/# [  910.619483] esp32_sdio_c5: init_bt: ESP Bluetooth init
[  910.625441] esp32_sdio_c5: print_capabilities: Capabilities: 0xd. Features supported are:
[  910.639316] esp32_sdio_c5: print_capabilities:        * WLAN on SDIO
[  910.645674] esp32_sdio_c5: print_capabilities:        * BT/BLE
[  910.651274] esp32_sdio_c5: print_capabilities:          - HCI over SDIO
[  910.657424] esp32_sdio_c5: print_capabilities:          - BLE only
````

## Logs (slave)

ESP32 firmware reports correct MAC:

```text
ESP-ROM:esp32c5-eco2-20250121
Build:Jan 21 2025
rst:0x1 (POWERON),boot:0x38 (SPI_FAST_FLASH_BOOT)
SPI mode:DIO, clock div:1
load:0x408556b0,len:0x1754
load:0x4084bba0,len:0xd9c
load:0x4084e5a0,len:0x3090
entry 0x4084bbaa
I (23) boot: ESP-IDF v6.0-dev-205-g27d68f57e-dirty 2nd stage bootloader
I (24) boot: compile time Jul 17 2025 14:01:21
I (24) boot: chip revision: v1.0
I (25) boot: efuse block revision: v0.2
I (29) boot.esp32c5: SPI Speed      : 80MHz
I (33) boot.esp32c5: SPI Mode       : DIO
I (37) boot.esp32c5: SPI Flash Size : 4MB
I (40) boot: Enabling RNG early entropy source...
I (45) boot: Partition Table:
I (47) boot: ## Label            Usage          Type ST Offset   Length
I (54) boot:  0 nvs              WiFi data        01 02 00009000 00004000
[... 137 lines cut ...]
I (919576) FW_MAIN: Get Tx power command
I (986207) FW_MAIN: Get Tx power command
I (986267) FW_MAIN: Get Tx power command
```

After the host driver is loaded, it brings up the wlan0 interface.
The available WiFi Access points can be scanned but whenever an attempt is made to join a specific network (wifi access point), the host driver flags the following message:

```text
[ 1564.205849] esp32_sdio_c5: esp_cfg80211_change_iface: current iface type=0 new iface type=0
[ 1564.214616] esp32_sdio_c5: esp_cfg80211_change_iface: 415 operating in same mode
[ 1564.370141] esp32_sdio_c5: esp_set_mac_address: 416 00:00:00:00:00:00
```

## Expected behavior

The host driver should retrieve the ESP32’s real MAC (`30:ed:a0:e4:4c:28`) instead of using `00:00:00:00:00:00`.

## Notes

* Scanning works, confirming basic host–slave communication over SDIO.
* Issue seems limited to MAC retrieval / `esp_set_mac_address()`.
* Tested with `clockspeed=50`. No timeout issues.
* Kernel: `5.10.140` on OpenWrt 22.03.
---

kapilkedawat, 2025-09-03:
@Shreyas0-7 PTAL

san-av, 2025-09-03:
Hi @Shreyas0-7, 
Shall I use the same ESP32-C5 SDIO binaries as in [esp32c5_ng_beta_support](https://github.com/espressif/esp-hosted/tree/feat/esp32c5_ng_beta_support) branch or build new ones from the master branch?

Shreyas0-7, 2025-09-03:
Hey @san-av you will have to use new binaries and new host code from master branch

san-av, 2025-09-03:
Thanks @Shreyas0-7,
I am trying to build the esp_hosted_ng binaries from the master branch; but as shown below, the **esp32c5** target is **not** listed:

san@raspberrypi2:~/esp-hosted-master/esp_hosted_ng/esp/esp_driver/network_adapter $ idf.py --list-targets
esp32
esp32s2
esp32c3
esp32s3
esp32c2
esp32c6
esp32h2
esp32p4
san@raspberrypi2:~/esp-hosted-master/esp_hosted_ng/esp/esp_driver/network_adapter $

Shreyas0-7, 2025-09-03:
Hey @san-av did you setup IDF again for master?
https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/setup.md#13-esp-comprehensive-guide

mantriyogesh, 2025-09-04:
@kapilkedawat ++

kapilkedawat, 2025-09-04:
@san-av somehow your os is trying to set the interface address to 00:00:00:00:00:00. Could you please set host log level to debug and share full host logs(along with wpa_supplicant logs as well).

For unblocking, you can comment out this line. https://github.com/espressif/esp-hosted/blob/5633e76aa294a6844bfe1a30e4fc679153ed8d10/esp_hosted_ng/host/main.c#L471

san-av, 2025-09-06:
Thank you **@kapilkedawat** for your guidance. I followed your suggestions and built and tested the esp32_sdio_c5 host driver at a **50 MHz** clock speed on our NUC980 IoT host device. The driver now successfully assigns the correct wlan0 MAC address in STA mode and is able to connect to an access point and reach the WAN.

I then performed **iperf tests** on the DUT under under 802.11**n** and 802.11**ac** modes (various channels, bandwidths, and TX power levels). The measured iperf throughput in all cases was around **13.5 Mbits/sec**. The corresponding full host logs (including wpa_supplicant) and slave device logs are included in the zipped files attached here. Could you please review the logs and share your thoughts on why the iperf throughput is low.

Additionally, I carried out a **raw throughput** test over SDIO, which produced the following results:

Host → ESP-C5 : 41.5 Mbit/s

ESP-C5 → Host : 13.2 Mbit/s

san-av, 2025-09-09:
Hi **@Shreyas0-7**, I re-checked the hardware connections — no issues were found.

[image]

[image]

As shown in the pictures above, the hardware setup consists of three distinct PCBs:

1. **NUC980 IoT Devkit** – host device (bottom)
2. **ESP32-C5-DevKitC-1** – slave device (topmost)
3. **Custom adapter PCB** – placed in between, with **75 KΩ pull-ups on all lines except CLK**, connecting the NUC980 IoT Devkit to the ESP32-C5-DevKitC-1

I have applied mac_addr.patch, but this time I did not comment out the following line (471). As a result, the host driver still fails to set the MAC address:

[esp-hosted/esp_hosted_ng/host/main.c](https://github.com/espressif/esp-hosted/blob/5633e76aa294a6844bfe1a30e4fc679153ed8d10/esp_hosted_ng/host/main.c#L471)  
 .ndo_set_mac_address = esp_set_mac_address,

Please refer to the attached test log file.
[NUC980-ESP-Hoste-NG-Logs_2025.09.09.md](https://github.com/user-attachments/files/22243705/NUC980-ESP-Hoste-NG-Logs_2025.09.09.md)

Could the relatively low raw throughput (reported earlier) over SDIO be related to the current stacked hardware arrangement (NUC980 devkit → custom adapter PCB → ESP32-C5 devkit)? In our production design, the ESP32-C5 chip will be integrated directly on the custom NUC980-based board. We expect this integration to provide better signal integrity and improved performance, and would greatly appreciate your feedback before we move forward with production next week.

Shreyas0-7, 2025-09-10:
Hey @san-av 
Command `cat /sys/kernel/debug/mmc1/ios` is not showing actual clock frequency being used can you please check, what is the actual clock frequency being used.
Also it seems like network_manager/wpa_supplicant is trying to set mac addr as 00.00.00.00.00.00. If possible can you please provide debug logs for the to help us identify the issue.

7h3cyb3rm0nk, 2026-10-08:
## Same issue on Nuvoton NUC980 IoT + ESP32-C5

### Setup

| Item | Value |
|---|---|
| Host | Nuvoton NUC980 IoT board (OpenWrt) |
| Slave | ESP32-C5 |
| Firmware | `release/ng-1.0.6` |
| Host driver | `release/ng-1.0.6` |

### Observations

#### 1. `phy0` MAC is zero from driver bootup

```text
root@OpenWrt:/# cat /sys/class/ieee80211/phy0/macaddress
00:00:00:00:00:00
```

`iwinfo` and `ip addr` before connecting:
```
root@OpenWrt:/# iwinfo
wlan0     ESSID: unknown
          Access Point: 3C:DC:75:80:73:AC
          Mode: Client  Channel: unknown (unknown)
          Center Channel 1: unknown 2: unknown
          Tx-Power: 20 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11bgnac
          Hardware: 0092:6666 0000:0000 [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: no  PHY name: phy0
```

```
root@OpenWrt:/# ip addr show wlan0
4: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN qlen 1000
    link/ether 3c:dc:75:80:73:ac brd ff:ff:ff:ff:ff:ff
    
```

`wlan0` has a valid MAC (`3c:dc:75:80:73:ac`), but `phy0` reports all zeros.

#### 2. Manual connect to an open network works

```
root@OpenWrt:/# ip link set wlan0 up
root@OpenWrt:/# iw wlan0 connect 'Fbi honeypot'
root@OpenWrt:/# [ 2062.043441] esp32_sdio_c5: cmd_auth_request: Authentication request: f2:70:fd:25:14:67 13 0 0 0
[ 2062.605558] esp32_sdio_c5: cmd_assoc_request: Association request: f2:70:fd:25:14:67 13 0
[ 2062.858699] esp32_sdio_c5: process_assoc_event: Connection status: 0
root@OpenWrt:/# iwinfo
wlan0     ESSID: "Fbi honeypot"
          Access Point: 3C:DC:75:80:73:AC
          Mode: Client  Channel: 13 (2.472 GHz)
          Center Channel 1: unknown 2: unknown
          Tx-Power: 20 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11bgnac
          Hardware: 0092:6666 0000:0000 [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: no  PHY name: phy0
```
```
root@OpenWrt:/# ip addr show wlan0
4: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UNKNOWN qlen 1000
    link/ether 3c:dc:75:80:73:ac brd ff:ff:ff:ff:ff:ff

```

#### 3. Via netifd (`wifi reload`), the MAC is set to zeros

`/etc/config/wireless`:
```
root@OpenWrt:/# cat /etc/config/wireless

config wifi-device 'radio0'
        option type 'mac80211'
        option path 'platform/ahb/b0019000.fmi/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
        option channel '36'
        option band '5g'
        option htmode 'VHT80'
        option disabled '0'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'wwan'
        option mode 'sta'
        option ssid 'Fbi honeypot'
        option encryption 'none'
```
```
root@OpenWrt:/# wifi reload
root@OpenWrt:/# wi[ 2447.349649] esp32_sdio_c5: esp_stop: Deinitializing interface wlan0 on ESP side
[ 2447.357849] esp32_sdio_c5: process_disconnect_event: Disconnect event for ssid Fbi honeypot [reason:3]
[ 2447.404691] esp32_sdio_c5: esp_reg_notifier: cfg80211 regulatory domain callback for 00, current=
[ 2448.158141] esp32_sdio_c5: esp_cfg80211_change_iface: current iface type=0 new iface type=0
[ 2448.166499] esp32_sdio_c5: esp_cfg80211_change_iface: 416 operating in same mode
[ 2448.297732] esp32_sdio_c5: esp_set_mac_address: 440 00:00:00:00:00:00
```

#### 4. `esp_set_mac_address()` is called by `ip link set ... address`

```
root@OpenWrt:/# ip link set wlan0 address 12:34:45:67:89:00
[ 2797.487691] esp32_sdio_c5: esp_set_mac_address: 440 12:34:45:67:89:00
root@OpenWrt:/# ip addr show wlan0
4: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc pfifo_fast state DOWN qlen 1000
    link/ether 12:34:45:67:89:00 brd ff:ff:ff:ff:ff:ff
```
This confirms the driver's `ndo_set_mac_address` path works with an arbitrary MAC, so the setter itself is functional.

### Analysis

netifd's `/lib/netifd/wireless/mac80211.sh` (`mac80211_prepare_vif`) sets the interface MAC for non-AP modes:

```sh
mac80211_prepare_vif() {
       json_select config

        json_get_vars ifname mode ssid wds powersave macaddr enable wpa_psk_file vlan_file

        [ -n "$ifname" ] || ifname="wlan${phy#phy}${if_idx:+-$if_idx}"
        if_idx=$((${if_idx:-0} + 1))

        set_default wds 0
        set_default powersave 0

        json_select ..

        [ -n "$macaddr" ] || {
          macaddr="$(mac80211_generate_mac $phy)"
          macidx="$(($macidx + 1))"
        }
        # ...[code omitted]...
        if [ "$mode" != "ap" ]; then
          # ALL ap functionality will be passed to hostapd
          # All interfaces must have unique mac addresses
          # which can either be explicitly set in the device
          # section, or automatically generated
          ip link set dev "$ifname" address "$macaddr"
        fi
        # ...[code omitted]...
}
```

If no `macaddr` is configured, netifd calls `mac80211_generate_mac $phy`, which derives the MAC from `phy0`'s `macaddress` attribute. That attribute is `00:00:00:00:00:00` here, so `ip link set dev wlan0 address <zeros>` triggers `esp_set_mac_address()` with zeros.

kapilkedawat, 2026-10-08:
Please open a new issue in https://github.com/espressif/esp-hosted-linux

closed 2026-10-08 by kapilkedawat