espressif/esp-hosted #664: Compatibility issues with Linux kernel 4.9.84 in ESP-Hosted-NG Release 1.0.4.0.0

yuMKY, 2025-09-22:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I am trying to build ESP-Hosted-NG Release 1.0.4.0.0 on an embedded platform running Linux kernel 4.9.84, and I’m running into multiple compilation errors because some of the kernel APIs have changed compared to newer versions.
Would it be possible to add conditional compilation guards (#if LINUX_VERSION_CODE >= ...) or provide compatibility wrappers so that the driver can build on older kernels (like 4.9.x)?

### Actual behavior (suspected bug)

Examples of errors:
1. fs API difference (kernel_read)
error: passing argument 2 of ‘kernel_read’ makes integer from pointer without a cast [-Werror=int-conversion]
while ((nread = kernel_read(file, ota_chunk, OTA_CHUNK_SIZE, &file->f_pos)) > 0) {
                                  ^
2. cfg80211_auth_request incompatibility
error: ‘struct cfg80211_auth_request’ has no member named ‘auth_data_len’; did you mean ‘sae_data_len’?
error: ‘struct cfg80211_auth_request’ has no member named ‘auth_data’; did you mean ‘auth_type’?

3. station_parameters incompatibility
error: ‘struct station_parameters’ has no member named ‘he_capa’; did you mean ‘ht_capa’?
error: ‘WLAN_EID_EXTENSION’ undeclared
error: ‘WLAN_EID_EXT_HE_CAPABILITY’ undeclared

4. ........

[image]
[image]

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

1. Copy the esp_hosted_ng/host folder into the drivers directory of the Linux 4.9.84 kernel source tree.
2. Modify the Makefile:
        CROSS_COMPILE := <your cross-compile toolchain>
        KERNEL := <path to your kernel source>
        ARCH := arm
3. make target=sdio

### Project release version

ESP-Hosted-NG Release 1.0.4.0.0

### System architecture

ARM 32-bit (Raspberry Pi 32-bit)

### Operating system

Linux

### Operating system version

Linux version 4.9.84

### Shell

Bash

mantriyogesh, 2025-09-22:
@Shreyas0-7 PTAL..

yuMKY, 2025-09-24:
Hello @Shreyas0-7 ，Based on yesterday’s patch package, there are still some errors. However, I have tried to fix them, and the compilation has passed. I need you to confirm whether my modifications are correct and that no other issues will arise.

errors:
1. host/esp_cmd.c:2340:2:错误：ignoring return valueof ‘strscpy', declared with attribute warn_unused_result [-Werror=unused-result]strscpy(cmd->country_code, priv->country_code, MAx_couNTRy_LEN);
2. host/esp_cfg80211.c:1076:25:错误：initializationof‘int （*)（struct wiphy *， struct net_device *, enum nl8o211_iftype, u32 *, struct vif_params *）’[或称 int (*)(struct wiphy *, struct net_device *, enum nl80211_iftype, unsigned int *, struct vif_params *)'} fromincompatible pointer type (int (*)(struct wiphy *, struct net_device *, enum nl80211_iftype, sstruct vif_params *)’ [-Werror=incompatible-pointer-types].change_virtual_intf = esp_cfg80211_change_iface,
3. host/esp_cfg80211.c:1174:3:错误：ignoring returnvalue of ‘strscpy', declared with attribute warn_unused_result [-Westrscpy(priv->country_code, request->alpha2, MAX_couNTRY_LEN);
4. host/esp_stats.c:108:3:错误：implicit declarationof function ‘timer_setup'; did you mean ‘ether_setup'? [-Werror=implicit-function-declaration]timer_setup(&log_raw_tp_stats_timer, log_raw_tp_stats_timer_cb, 0);

[image]

[image]

[image]

fix:
1. Added return value checking for the **strscpy** function.
2. Added an additional parameter to **esp_cfg80211_change_iface**.
3. Added return value checking for the **strscpy** function.
4. Check the kernel version: use **timer_setup** if it is ≥ 4.15.0; otherwise, use **setup_timer**, and modify the corresponding callback function accordingly.

[image]

[image]

[image]

[image]

[image]

yuMKY, 2025-09-25:
> [@yuMKY](https://github.com/yuMKY) can you please post the final change as [pull-request](https://github.com/espressif/esp-hosted/pulls) ?

@mantriyogesh Hello,

The driver now compiles successfully, but I am using a custom kernel and I am not sure whether these modifications are compatible with the standard Linux kernel. I also haven’t tried compiling it on any other kernel versions.

Currently, I am testing the driver’s communication with the ESP32C6 module, but I am encountering issues, as shown below:

**insmod esp32_sdio.ko resetpin=73 clockspeed=12

dmesg -c
>> [sdmmc_0] Err: #Cmd_52 (0x80022000)=>(E: 0x0008)(S: 0x000010FF)__(L:886)
esp32_sdio: probe of mmc0:0001:1 failed with error 8
>> [sdmmc_0] Err: #Cmd_52 (0x80042000)=>(E: 0x0008)(S: 0x000010FF)__(L:886)
esp32_sdio: probe of mmc0:0001:2 failed with error 8**

The resetpin=73 successfully resets the module, but the driver initialization fails at the probe stage, and there is no other output. The module is flashed with the sdio_only firmware under esp32c6 in ESP-Hosted-NG-release_v1.0.4.0.0.zip.

@Shreyas0-7 Could you please advise on how I should troubleshoot this issue？

mantriyogesh, 2025-09-26:
8 is [ENOEXEC](https://elixir.bootlin.com/linux/v4.5/source/include/uapi/asm-generic/errno-base.h#L11). I am not sure if you have correctly flashed esp and have sufficient rights to do insmod in Linux. Anyway, please help with full textual logs from boot up to understand the issue, at both sides esp (idf.py monitor) and Linux (dmesg.txt)

yuMKY, 2025-09-26:
> 8 is [ENOEXEC](https://elixir.bootlin.com/linux/v4.5/source/include/uapi/asm-generic/errno-base.h#L11). I am not sure if you have correctly flashed esp and have sufficient rights to do insmod in [...]

@mantriyogesh ，Thank you for your reply.Please find the attached logs for reference.

Today I flashed the "sdio+uart" firmware to the ESP32-C6 module.
**“esp32c6.log”** contains the boot log from the module (I am not very familiar with how to use idf.py monitor). When I insert the driver with “insmod esp32_sdio.ko resetpin=73 clockspeed=12,” the module only outputs the reboot information, and nothing further.

**“dmesg.log”** includes the kernel side logs. Some messages such as “esp32_sdio: esp_init: Starting ESP driver initialization” were added by me for debugging.

**Please let me know if any additional information is needed.**

[esp32c6.log](https://github.com/user-attachments/files/22551926/esp32c6.log)

[dmesg.log](https://github.com/user-attachments/files/22551927/dmesg.log)

[others.txt](https://github.com/user-attachments/files/22551928/others.txt)

yuMKY, 2025-09-26:
@mantriyogesh   I came across a similar issue reported earlier. Do you know how it was finally resolved?
[](https://github.com/espressif/esp-hosted/issues/293)

yuMKY, 2025-09-28:
@mantriyogesh  Thank you for your response. I tried removing resistor **R45**, and as a result, I was able to successfully initialize the driver and the wlan0 interface appeared. I’m not sure why removing R45 allows it to work correctly. However, after removing this resistor, the signal is fixed at a high level, and I can no longer use the resetpin to reset the ESP32-C6.

When the resistor is present, I can reset the ESP32-C6 via the resetpin, and the final signal level is high, indicating that the module is enabled. I don’t understand why R45 affects driver initialization. Could you provide some possible reasons? I need to use the resetpin to control ESP32-C6 for resets.

[image]

mantriyogesh, 2025-09-29:
I might need some additional context to get hold of the situation.

For starters, would it be possible to atatch the both side text based logs for esp and host, in both with or without case?
This would let us get to faster conclusion. Please also attach the sdkconfig generated along with the logs at esp (to understand what configs went inside the bin).

I am also confused if you have used sdio only or still using sdio_uart in this test (anyway logs would speak that).
I am suspecting if it is mere timing issue, to probe. in the sense, sdio slave driver was not yet ready by the host started sdio probing, but this is just suspect, in general i did not observe such trend on Linux systems perticularly.

Even with this register intact, if you just just the sdio only binaries, ideally should have worked, IMO. I might need you help to comment from your actual side how the observations you see in the system (with logs), to conclude exact what was problem..

yuMKY, 2025-09-30:
@mantriyogesh 
Sorry for the late reply, I’ve been quite busy these days.
Recently I tried changing the reset GPIO, but the result was the same. Then I switched to using the ESP32’s power pin to reset the ESP32 module, and now it can be reset successfully and the wlan0 node is created.

Regarding the sdkconfig file you mentioned, I’m not quite sure how to handle that. I haven’t compiled the ESP32 firmware myself. Instead, I used the precompiled firmware from GitHub:
[ESP-Hosted-NG-release_v1.0.4.0.0.zip](https://github.com/espressif/esp-hosted/releases/download/release%2Fng-v1.0.4.0.0/ESP-Hosted-NG-release_v1.0.4.0.0.zip)

Anyway, the issue is now resolved. I might run into more problems later, and I may continue to ask here or open a new issue. I hope you can keep providing your support. Thanks a lot!